thesofproject/sof #6026: [BUG] no multichannel sound through video on Intel Ice Lake laptop with integrated graphics

pfps, 2022-07-19:
**Describe the bug**

I'm not even sure that this is a bug, or that it is an SOF bug, but it looks to me as if my laptop should be able to play multi-channel sound through its video (HDMI/DP) outputs.  But attempts to do so result in two-channel sound or fail completely.  Is there something in SOF that needs to be changed to enable multi-channel sound through the video outputs?

**To Reproduce**
Play 5+1 audio routed through HDMI/DP outputs (through pipewire or directly using speaker-test).  The pipewire audio is remixed to two channels and speaker-test complains 'Channels count (6) not available for playbacks: Invalid argument'.  

Command to reproduce through pipewire:
```
speaker-test -c6 -t wav
[and then route output to Ice Lake ... HDMI / DisplayPort 2 Output]
[sound is remixed two-channel sound]
```
Command to play directly
```
[set output configuration to Off in pulse audio volume control]
speaker-test -c6 -t wav -D hw:0,4

speaker-test 1.2.7

Playback device is hw:0,4
Stream parameters are 48000Hz, S16_LE, 6 channels
WAV file(s)
Channels count (6) not available for playbacks: Invalid argument
Setting of hwparams failed: Invalid argument
```

**Reproduction Rate**
Happens all the time.

**Expected behavior**
Multichannel (instead of two-channel) sound is played by the speaker system connected to video display.

**Impact**
This is just an annoyance, but the setup should be fairly common.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.18.11-200.fc36.x86_64 (current Fedora 36 kernel)
    * SOF: 2:1:1-3964a (I guess)
2) Name of the topology file
    * Topology: unknown - how can I find this out?
3) Name of the platform(s) on which the bug is observed.
    * Platform: Lenovo Yoga C940 - Ice Lake CPU, integrated video, two Thunderbolt ports

**Screenshots or console output**

[Overall description - sof-logger does not appear to be installed in Fedora]

I have an Intel Ice Lake laptop without a separate video chip.  The integrated video output is sent to two Thunderbolt / USB-C ports, as is usual.

My question is whether SOF supports multi-channel audio through these video connections.  It appears that it is possible, as /proc/asound/card0/codec#2 shows several 8-channel capabilities, including

```
Node 0x03 [Audio Output] wcaps 0x6611: 8-Channels Digital
  Converter: stream=15, channel=0
  Digital: Enabled KAE
  Digital category: 0x0
  IEC Coding Type: 0x0
  PCM:
    rates [0x7f0]: 32000 44100 48000 88200 96000 176400 192000
    bits [0x1a]: 16 24 32
    formats [0x5]: PCM AC3
  Power states:  D0 D3 EPSS
  Power: setting=D0, actual=D0
Node 0x04 [Pin Complex] wcaps 0x40778d: 8-Channels Digital Amp-Out CP
  Amp-Out caps: ofs=0x00, nsteps=0x00, stepsize=0x00, mute=1
  Amp-Out vals:  [0x00 0x00]
  Pincap 0x0b000094: OUT Detect HBR HDMI DP
  Pin Default 0x18560010: [Jack] Digital Out at Int HDMI
    Conn = Digital, Color = Unknown
    DefAssociation = 0x1, Sequence = 0x0
  Pin-ctls: 0x00:
  Unsolicited: tag=00, enabled=0
  Power states:  D0 D3 EPSS
  Power: setting=D0, actual=D0
  Devices: 0
  Connection: 0
```

It appears that this shows up in amixer, including information about the sound system, as

```
numid=32,iface=PCM,name='ELD',device=4
  ; type=BYTES,access=r--v----,values=36
  : values=0x10,0x00,0x08,0x00,0x69,0x26,0x00,0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x63,0x18,0x51,0x28,0x42,0x65,0x79,0x6f,0x6e,0x64,0x20,0x54,0x56,0x09,0x07,0x07,0x15,0x17,0x50,0x00
numid=40,iface=PCM,name='Playback Channel Map',device=4
  ; type=INTEGER,access=rw---R--,values=8,min=0,max=36,step=0
  : values=3,4,0,0,0,0,0,0
  | container
    | chmap-variable=FL,FR
```

But I can't find a way to make all the channels available.  I've tried to use amixer to change the hardware control for channel but that doesn't change anything.

So where is the problem?  Is it at all possible to send 7+1 or 5+1 sound through the video output?  If so, is SOF or a topology file or ASLA somehow not exposing the capability?

Here are the sof messages in the system log, from the last boot to now:

```
Jul 18 09:37:24 idefix kernel: sof-audio-pci-intel-icl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
Jul 18 09:37:24 idefix kernel: sof-audio-pci-intel-icl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
Jul 18 09:37:24 idefix kernel: sof-audio-pci-intel-icl 0000:00:1f.3: enabling device (0000 -> 0002)
Jul 18 09:37:24 idefix kernel: sof-audio-pci-intel-icl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
Jul 18 09:37:24 idefix kernel: sof-audio-pci-intel-icl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
Jul 18 09:37:24 idefix kernel: sof-audio-pci-intel-icl 0000:00:1f.3: use msi interrupt mode
Jul 18 09:37:24 idefix kernel: sof-audio-pci-intel-icl 0000:00:1f.3: hda codecs found, mask 5
Jul 18 09:37:24 idefix kernel: sof-audio-pci-intel-icl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
Jul 18 09:37:24 idefix kernel: sof-audio-pci-intel-icl 0000:00:1f.3: DMICs detected in NHLT tables: 2
Jul 18 09:37:24 idefix kernel: sof-audio-pci-intel-icl 0000:00:1f.3: Firmware info: version 2:1:1-3964a
Jul 18 09:37:24 idefix kernel: sof-audio-pci-intel-icl 0000:00:1f.3: Firmware: ABI 3:21:0 Kernel ABI 3:19:1
Jul 18 09:37:24 idefix kernel: sof-audio-pci-intel-icl 0000:00:1f.3: warn: FW ABI is more recent than kernel
Jul 18 09:37:24 idefix kernel: sof-audio-pci-intel-icl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
Jul 18 09:37:24 idefix kernel: sof-audio-pci-intel-icl 0000:00:1f.3: Firmware info: version 2:1:1-3964a
Jul 18 09:37:24 idefix kernel: sof-audio-pci-intel-icl 0000:00:1f.3: Firmware: ABI 3:21:0 Kernel ABI 3:19:1
Jul 18 09:37:24 idefix kernel: sof-audio-pci-intel-icl 0000:00:1f.3: warn: FW ABI is more recent than kernel
Jul 18 09:37:24 idefix kernel: sof-audio-pci-intel-icl 0000:00:1f.3: Topology: ABI 3:21:0 Kernel ABI 3:19:1
Jul 18 09:37:24 idefix kernel: sof-audio-pci-intel-icl 0000:00:1f.3: warn: topology ABI is more recent than kernel
Jul 18 09:37:24 idefix kernel: input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input38
Jul 18 09:37:24 idefix kernel: input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input39
Jul 18 09:37:24 idefix kernel: input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input40
Jul 18 09:37:24 idefix kernel: input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input41
Jul 18 09:37:24 idefix kernel: input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input42
```

[systemlog.txt](https://github.com/thesofproject/sof/files/9140425/systemlog.txt)

lgirdwood, 2022-07-19:
> So where is the problem? Is it at all possible to send 7+1 or 5+1 sound through the video output? If so, is SOF or a topology file or ASLA somehow not exposing the capability?

I'm pretty sure we are missing an 8 channel topology here for this configuration and possibly some IP programming. 

The default fw build supports 8 channel pipelines when set in the topology, so this could be tried as a first step. However, if this does not work, we may need to set other IP bits for 8 channels in the FW or GFX driver (if using DP/HDMI audio). Fwiw, this is on the road map for @kv2019i and @jsarha. Best to wait for @kv2019i to comment next week.

pfps, 2022-07-19:
OK, I have a setup that can be used for testing (laptop, TV, 5.1 speaker system connected to TV via ARC).  

When I plug in the connection to the TV (and speaker system), I get ELD information in /proc/asound/card0/eld#2.3

```
monitor_present		1
eld_valid		1
monitor_name		Beyond TV
connection_type		DisplayPort
eld_version		[0x2] CEA-861D or below
edid_version		[0x3] CEA-861-B, C or D
manufacture_id		0x1863
product_id		0x2851
port_id			0x0
support_hdcp		0
support_ai		1
audio_sync_delay	0
speakers		[0x1] FL/FR
sad_count		2
sad0_coding_type	[0x1] LPCM
sad0_channels		2
sad0_rates		[0xe0] 32000 44100 48000
sad0_bits		[0xe0000] 16 20 24
sad1_coding_type	[0x2] AC-3
sad1_channels		6
sad1_rates		[0x4e0] 32000 44100 48000 96000
sad1_max_bitrate	640000
```

plbossart, 2022-07-19:
@pfps it's a known SOF limitation that's unlikely to be fixed for 'old' platforms like IceLake.

If you want to play multichannel audio, the best shortcut is to add this in /etc/modprobe.d/alsa-base.conf
````
options snd-intel-dspcfg dsp_driver=1
````

This will force the HDaudio legacy driver to be used. You will not have any limitation on HDMI, but you will not be able to use the local microphones. It's not ideal I know but that's the only quick solution we can offer.

pfps, 2022-07-19:
@plbossart With the legacy driver I don't get any sound from the HDMI-connected speaker system, even when connecting directly with all the alsamixer controls unmuted.

plbossart, 2022-07-19:
Not sure how to help you @pfps, we always recommend to try with the legacy driver. If that doesn't work then you are not on solid ground to start debugging further.

pfps, 2022-07-19:
I can get sound to the speakers with the SOF driver, just only stereo sound.  So something is working better here than with the legacy driver.  It is possible, I suppose, that the SOF driver has exposed a pipeline that goes through the Realtek chip, which can only handle stereo as far as I know, and then to the video stream.  But there are those exposed 8-channel maps and 8-channel amplifiers.

I'm willing to do some work on putting the pipeline together, but I don't think I have enough information to figure out what do to test out the possibilities.

plbossart, 2022-07-19:
there is no connection between the Realtek codec and the HDMI path. They are completely independent codecs.

pfps, 2022-07-19:
That's good to know.  So the S/PDIF 1 alsamixer control that mutes the HDMI-connected speakers is in the HDMI codec as is the PGA8.0 8 Master Playback Volume alsamixer control that changes its volume.  The pipeline to get to the HDMI-connected speakers appears to be going through hardware that has 8 channels it is just that something is limiting the sound to 2 channels.

plbossart, 2022-07-19:
no, the 'PGA8.0 8 Master Playback Volume' is an SOF control. It's internal to the Intel DSP and not to the HDMI codec. The entire DSP path is stereo.

pfps, 2022-07-19:
So the sound that's playing through the HDMI-connected speakers is not going through the Intel Icelake HDMI codec?  Even though the output device in pulse audio volume control is called HDMI / DisplayPort 2?

pfps, 2022-07-19:
Oh.

closed 2022-07-26 by kfrydryx

pfps, 2022-07-26:
I'm confused as to just what SOF is supposed to handle.   It appears to me that SOF is claiming the entire audio hardware on my Ice Lake laptop, which I take to include some DSP hardware, a RealTek ALC298 (or emulator), and Intel Icelake HDMI audio.  SOF is doing something with the DSP and the ALC298, but doesn't appear to be doing anything useful with the HDMI audio, even though the hardware does show up and ELD information is being processed.

Is it not possible to just open up an audio path that only goes through the HDMI audio hardware, which should allow for multi-channel output to the HDMI connections?  Is there some problem in SOF that restricts it to two-channel sound or to PCM encoding?  Is the problem that there is no information on how to open paths in the HDMI audio hardware?  

Is it possible to have some other driver control the HDMI audio and still have SOF control the DSP and thus have the microphone work?

kv2019i, 2022-07-26:
@pfps The DSP is used on all audio paths. HDMI/DP display audio is supported, but with current DSP topologies this is limited to stereo configuration. You can switch back to snd-hda-intel driver and disable DSP with following kernel option:

options snd_intel_dspcfg dsp_driver=1
  
On your laptop, the biggest thing you lose is support for the integrated digital microphone (on Linux desktop systems without a DMIC, SOF is not used by default).

pfps, 2022-07-26:
I haven't been able to get multi-channel sound through HDMI working on the hda-intel driver either.  I think the problem is that multi-channel sound to my speaker system needs AC3 encoding, or at least that's what I see in the eld information.

@kv2019i Do you mean that DSP is used on all audio paths in SOF or overall?  It looks like the former because you say that the hda-intel driver disables the DSP.  So why can't SOF set up an audio path that bypasses the DSP and goes straight to the HDMI codec.

In the end, it would be nice for SOF to support the all the major capabilities of recent Intel laptops unless they are blocked by some unsolvable issue (like there being no way to get documentation).

kv2019i, 2022-07-26:
@pfps It should definitely work with snd-hda-intel.  Probably best to first test with some app that can send encoded AC3 to ALSA (or try first with a HDMI receiver that can handle PCM multichannel, and only then move to AC3 encoded).

For the other question, we currently have a limitation that either all endpoints are handled via DSP or none are. This forces us to route audio streams to HDMI also via DSP, although this is not really ideal. It does work for for some configurations (the typical audio capable HDMI/DP display that is limited to stereo), but more complex HDMI receiver configurations cannot be handled with the current SOF FW and topologies. This is a known limitation and is on our backlog.

pfps, 2022-07-26:
@kv2019i That's somewhat encouraging.  Is there any way I can help?