thesofproject/sof #6034: [BUG] Volume DSP Panic on IPC response timeout, DSP Panic notification

wszypelt, 2022-07-21:
**Describe the bug**

ERROR:Exception DspPanicError occurred at 'C:\\work\\tools\\cavs_scripts-py\\utilities\\ipc_driver.py':171 DSP Panic on IPC response timeout! 0x00000005 0x0DEAD006 0x00004000
ERROR:Exception DspPanicError occurred at 'C:\\work\\tools\\cavs_scripts-py\\utilities\\ipc_driver.py':171 DSP Panic notification received: SOF_IPC_PANIC_EXCEPTION 0x00000005 0x0DEAD006 0x00004000

Bug occurs with a playback iteration greater than 1

Bug was found in:
TestVolumeSimple8000Hz16b16b1ch
TestVolumeSimple8000Hz24b32b4ch
TestVolumeSimple48000Hz32b32b8ch
TestVolumeChangeInMiddle8000Hz32b32b4ch
TestVolumeChangeInMiddle48000Hz32b32b4ch
TestVolumeMultipleChanges8000Hz24b32b2ch
TestVolumeMultipleChanges48000Hz24b32b2ch
TestVolumeMute8000Hz32b32b8ch
TestVolumeMute48000Hz32b32b8ch
TestVolumeInitialRampSimple8000Hz24b32b4ch
TestVolumeInitialRampSimple48000Hz32b32b2ch
TestDiffVolumeChangeInChannels8000Hz24b32b4ch
TestDiffVolumeChangeInChannels48000Hz32b32b4ch

**Topology**
```
TestVolumeSimple

                +--------------------------------+
    +------+    | +---+  +- -+  +---+  +-------+ |
    | Host |----->|Buf|->|Vol|->|Buf|->|SSP Dai|---+
    +------+    | +---+  +-- +  +---+  +-------+ | |
                +--------------------------------+ |
                                                   |
                +------------------+               |
    +------+    | +---+  +-------+ |               |
    | Host |<-----|Buf|<-|SSP Dai|<----------------+
    +------+    | +---+  +-------+ |
                +------------------+

```
Description:
Simultaneous playback and capture on external loopback (same SSP) with signal amplitude verification.
Playback pipeline contains volume component.
Volume gain is explicitly set to 1 before playback starts.

**To Reproduce**
Run tests with diagnostic driver 
with argument: --playback_iterations=2

03_00_TestVolumeSimple8000Hz16b16b1ch

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF BRANCH: main HASH: ea72e3e97e10acb27d02c3692b006c4f66fabfe4
2) Name of the platform(s) on which the bug is observed.
    * Platform: TGL RVP

the last working and not working commit:  ebc222dab56868ea3cb0d5abec5057990eae6bb9

**Logs**
[DSPPanicVolume.zip](https://github.com/thesofproject/sof/files/9157675/DSPPanicVolume.zip)

lgirdwood, 2022-07-21:
@ranj063 module adapter related ?

kfrydryx, 2022-08-02:
Still reproducing. Bisect points to module adapter - first bad commit f7663de1f41308b80bac4618b5835fcda1ff8fac.

lgirdwood, 2022-08-02:
@wszypelt @kfrydryx is this only reproducing with the loopback ? What happens if gain is NOT set prior to test. The DEAD code is an assert().
@lyakh can you take a look at this, thanks !

wszypelt, 2022-08-02:
@lgirdwood 
I do not know why. does not provide information about IPC4 regression tests.
During these tests, he failed the SDW test
### TestSdwTransferLoopback.test_16_00_loopback [m0-sdw_0_pdi_2_to_sdw_1_pdi_3-48000Hz_24in32bit_2ch] ###
I report the matter to the appropriate team so that they will try to fix it as soon as possible.

lyakh, 2022-08-15:
@wszypelt can this still be reproduced? But this test isn't in the standard CI, so we don't see it in PR testing?

wszypelt, 2022-08-22:
@lyakh @kfrydryx @lgirdwood 
Issue still reproducing, with playback iteration > 1. In CI we test all tests with playback_iteration = 1. Therefore, it does not show this issue. We perform stress tests every two weeks.

lgirdwood, 2022-08-24:
> @lyakh @kfrydryx @lgirdwood
> Issue still reproducing, with playback iteration > 1. In CI we test all tests with playback_iteration = 1. Therefore, it does not show this issue. We perform stress tests every two weeks.

Ok, so the Linux CI will test multiple times, but I dont think it uses the loopback mode above. @fredoh9 can you confirm if we have a similar loopback test in the Jenkins CI ?

PR #6230 (2022-09-06): module_adapter: Modify reset API
Modify the definition of the reset API in the module adapter interface
to make sure that it should reset the module state back to
MODULE_INITIALIZED and free all memory that was allocated during the
prepare() callback. With this change, stopping and restarting streams
will always be guaranteed to invoke the module's prepare() callback.
Also, fix the passthrough and cadence codec implementation to follow the
new definition.

Fixes #6034
closed 2022-09-06 by lgirdwood