thesofproject/sof #6075: [BUG] Unable to install hw params when recording with hw:0,4 on Dell laptops

miRoox, 2022-08-04:
**Describe the bug**
On the ADL and TGL Dell laptops, arecord with hw:0,4 will report error
```
arecord: set_params:1432: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: (85333 85334)
PERIOD_SIZE: 4096
PERIOD_BYTES: 16384
PERIODS: 4
BUFFER_TIME: (341333 341334)
BUFFER_SIZE: 16384
BUFFER_BYTES: 65536
TICK_TIME: 0
```

The bisect result shows it is introduced by e4a53594403d40926b349423045b0f13c252b220.

**To Reproduce**
```
arecord   -Dhw:0,4 -r 48000 -c 2 -f S16_LE -d 3 /dev/null -v -q
```

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/655b48ffac48
    * SOF: https://github.com/thesofproject/sof/commit/e4a53594403d40926b349423045b0f13c252b220
2) Name of the topology file (corresponding to the following platforms)
    * Topology: sof-adl-rt711-l0-rt1316-l12-rt714-l3.tplg, sof-tgl-rt715-rt711-rt1308-mono.tplg, sof-tgl-rt711-rt1316-rt714.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: ADLP_SKU0B00_SDCA, TGLU_SKU0A3E_SDW, TGLU_SKU0A32_SDCA

**Screenshots or console output**

**dmesg**
```
[  205.003048] kernel: snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[  205.003774] kernel: snd_sof:sof_ipc3_pcm_hw_params: sof-audio-pci-intel-tgl 0000:00:1f.3: stream_tag 2
[  205.003783] kernel: snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  205.004266] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60010000 (msg/reply size: 108/20): -22
[  205.004268] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: HW params ipc failed for stream 2
[  205.004269] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[  205.004291] kernel:  Microphone: ASoC: __soc_pcm_hw_params() failed (-22)
[  205.004297] kernel: sdw_deprepare_stream: subdevice #0-Capture: inconsistent state state 1
[  205.004308] kernel: snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  205.004492] kernel:  Microphone: ASoC: dpcm_fe_dai_hw_params failed (-22)
[  205.004499] kernel: snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 4 dir 1
```

plbossart, 2022-08-04:
I would rephrase the wording, 'unable to install' hints at a bad configuration when the real issue is an IPC error.

````
[  242.139062] kernel: snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  242.139579] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60010000 (msg/reply size: 108/20): -22
[  242.139584] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: HW params ipc failed for stream 2
[  242.139587] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[  242.139637] kernel:  Microphone: ASoC: __soc_pcm_hw_params() failed (-22)
[  242.139649] kernel: sdw_deprepare_stream: subdevice #0-Capture: inconsistent state state 1
[  242.139675] kernel: snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  242.139882] kernel:  Microphone: ASoC: dpcm_fe_dai_hw_params failed (-22)
````

It's likely that there are multiple hidden issues nested, the iir message below looks very suspicious, there is nothing specific related to those capture streams, only the source is different (RT715 v. DMIC).

````
[    19391192.198213] (    19391192.000000) c0 dma-trace          ../audio/eq_iir/eq_iir.c:921  ERROR eq_iir_prepare(), No proc func
[    19391216.937796] (          24.739582) c0 dma-trace          ....../pipeline-params.c:350  ERROR pipeline_prepare(): ret = -22, dev->comp.id = 18
[    19391239.489878] (          22.552082) c0 dma-trace            src/ipc/ipc3/handler.c:313  ERROR ipc: pipe 5 comp 18 prepare failed -22
````

miRoox, 2022-08-08:
Fixed by #6094

closed 2022-08-08 by miRoox