thesofproject/sof #6077: [FEATURE] Implement DRAIN for compress audio pipeline

dbaluta, 2022-08-04:
**Is your feature request related to a problem? Please describe.**

Implement compress API drain

**Describe the solution you'd like**

Compress API provides `SNDRV_PCM_TRIGGER_DRAIN` and SOF also offers command `SOF_IPC_STREAM_TRIG_DRAIN` but we need to implement the actual functionality both in Linux kernel driver and SOF FW

Linux kernel:

* forwards `SOF_IPC_STREAM_TRIG_DRAIN` to FW and waits (?) for FW to confirm that drain is done

SOF firmware:

* module adapter (?) implements drain. 

Edit: @LaurentiuM1234 @dbaluta @plbossart I'm trying to find initial ideas on how to implement DRAIN for compress audio.

dbaluta, 2022-08-04:
Main questions here:

1) How kernel knows that DRAIN has finished?
2) Do we need a new IPC from FW to Linux kernel driver to announce that drain has finished?

lgirdwood, 2022-08-04:
@dbaluta fwiw, IPC4 does have a pipeline state change IPC to indicate the drain has stopped, but we need to add this to ipc3 too.

dbaluta, 2022-08-04:
@lgirdwood do you mean End Of Stream state?

```

/* Ipc4 pipeline message <------> ipc3 pipeline message
 * RUNNING     <-------> TRIGGER START
 * INIT + PAUSED  <-------> PIPELINE COMPLETE
 * INIT + RESET <-------> PIPELINE COMPLETE
 * PAUSED      <-------> TRIGER_PAUSE
 * RESET       <-------> TRIGER_STOP + RESET
 * EOS(end of stream) <-------> NOT SUPPORT NOW                                                                                                                                                                    
 *
 *   IPC4 pipeline state machine
 *
 *                      INIT
 *                       |    \
 *                       |   __\|
 *                       |
 *                       |     RESET
 *                       |     _   _
 *                       |     /| |\
 *                       |    /    /\
 *                      \|/ |/_   /  \
 *        RUNNING <--> PAUSE _   /    \
 *            /  \      /|\ |\  /      \
 *           /    \      |    \/        \
 *          /      \     |    /\         \
 *         /        \    |   /  \         \
 *       |/_        _\|  |  /    \        _\|
 *     ERROR Stop       EOS       |______\ SAVE
 *                                      /
 */

```

lgirdwood, 2022-08-10:
> @lgirdwood do you mean End Of Stream state?
> 
> _src/ipc/ipc4/handler.c_
> 
> ```
> 
> /* Ipc4 pipeline message <------> ipc3 pipeline message
>  * RUNNING     <-------> TRIGGER START
>  * INIT + PAUSED  <-------> PIPELINE COMPLETE
>  * INIT + RESET <-------> PIPELINE COMPLETE
>  * PAUSED      <-------> TRIGER_PAUSE
>  * RESET       <-------> TRIGER_STOP + RESET
>  * EOS(end of stream) <-------> NOT SUPPORT NOW                                                                                                                                                                    
>  *
>  *   IPC4 pipeline state machine
>  *
>  *                      INIT
>  *                       |    \
>  *                       |   __\|
>  *                       |
>  *                       |     RESET
>  *                       |     _   _
>  *                       |     /| |\
>  *                       |    /    /\
>  *                      \|/ |/_   /  \
>  *        RUNNING <--> PAUSE _   /    \
>  *            /  \      /|\ |\  /      \
>  *           /    \      |    \/        \
>  *          /      \     |    /\         \
>  *         /        \    |   /  \         \
>  *       |/_        _\|  |  /    \        _\|
>  *     ERROR Stop       EOS       |______\ SAVE
>  *                                      /
>  */
> ```

Yes, a similar IPC3 mechanism could be used to indicate "I've finished playing the stream"