thesofproject/sof #6082: [BUG] TGL CODEC DSP Panic notification

apakuls, 2022-08-04:
**Describe the bug**

0x00000005 0x0DEAD006 0x00004000
ERROR:sof_core_dump.py:44, in decode_coredump() Error when decoding core dump: HTTPConnectionPool(host='10.102.14.242', port=80): Max retries exceeded with url: /coredump?platform_type=TGL (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x0000026CB28745C0>: Failed to establish a new connection: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond',))
ERROR:Error when decoding core dump: HTTPConnectionPool(host='10.102.14.242', port=80): Max retries exceeded with url: /coredump?platform_type=TGL (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x0000026CB2874F28>: Failed to establish a new connection: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond',))
ERROR:Exception DspPanicError occurred at 'C:\\work\\tools\\pyexecnetcache\\cavs_scripts-py\\utilities\\ipc_driver.py':171 DSP Panic on IPC response timeout! 0x00000005 0x0DEAD006 0x00004000
ERROR:Exception DspPanicError occurred at 'C:\\work\\tools\\pyexecnetcache\\cavs_scripts-py\\utilities\\ipc_driver.py':171 DSP Panic notification received: SOF_IPC_PANIC_EXCEPTION 0x00000005 0x0DEAD006 0x00004000

Bug was found in:
TestGenericProcessorCompMultiCorePlb48000Hz16b16b4ch
TestGenericProcessorCompMultiCorePlb48000Hz16b16b8ch
TestGenericProcessorPplMultiCorePlb48000Hz32b32b2ch

**Topology**
```
                +------------------------------------+
    +------+    | +---+  +-------+  +---+  +-------+ |
    | Host |----->|Buf|->|GenProc|->|Buf|->|SSP Dai|---+
    +------+    | +---+  +-------+  +---+  +-------+ | |
                +------------------------------------+ |
                                                       |
                +------------------+                   |
    +------+    | +---+  +-------+ |                   |
    | Host |<-----|Buf|<-|SSP Dai|<--------------------+
    +------+    | +---+  +-------+ |
                +------------------+

```
Description:
Simultaneous playback and capture on external loopback (same SSP) with signal amplitude verification.
Playback pipeline contains generic processor component on different core.

**To Reproduce**
Run tests with:
25_02_TestGenericProcessorCompMultiCorePlb48000Hz16b16b4ch

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash  
   * SOF BRANCH: main HASH: 2cfefc1103ef73713031a9f076dba953357f0229
2) Name of the platform(s) on which the bug is observed.
    * Platform: TGL RVP

The last working commit: dcf0577a7725413aad1a378d629b9cc6659550ef
Not working commit:  2cfefc1103ef73713031a9f076dba953357f0229

**Logs**
[CodecDSPPanic.zip](https://github.com/thesofproject/sof/files/9259413/CodecDSPPanic.zip)

lgirdwood, 2022-08-04:
@wszypelt is this with xtos or Zephyr ?

wszypelt, 2022-08-05:
@lgirdwood xtos

kv2019i, 2022-08-09:
@apakuls On TGL, we only support having components of a pipeline on the same core. Here it seems the playback pipeline has a single component on another core "Playback pipeline contains generic processor component on different core.". Can you confirm this is the case? If true, this can be closed as won't fix, @mwasko to confirm.

wszypelt, 2022-08-10:
@kv2019i 
- create topology: playback pipeline (core = 0) with generic processor component (core = 1) and capture pipeline
 (core = 0)connected to SSP

With that in mind, I am closing the issue as won't fix

closed 2022-08-10 by wszypelt