thesofproject/sof #6087: [BUG] alsatplg: Support SSP direction for each DAI

yongzhi1, 2022-08-04:
**Describe the bug**
SSP DAI direction based on individual dai_index is not supported:
https://github.com/alsa-project/alsa-utils/blob/master/topology/nhlt/intel/ssp/ssp-process.c#L531

**To Reproduce**
Sample changes https://github.com/yongzhi1/sof/tree/test-ssp1-playback.

Build above topology and check loopback capture on SSP0.

**Reproduction Rate** 100%

**Expected behavior**
We should be able to config SSP direction for each DAI.

juimonen, 2022-08-05:
@yongzhi1 could you try out:
https://github.com/juimonen/alsa-utils/tree/dai_index_fix

thanks!

yongzhi1, 2022-08-05:
Hi, thank you @juimonen for the quick fix, it works as expected now!!

closed 2022-08-05 by yongzhi1

yongzhi1, 2022-10-17:
Hi, @juimonen can you push https://github.com/juimonen/alsa-utils/tree/dai_index_fix for merging? thanks!!

juimonen, 2022-10-18:
@yongzhi1  https://github.com/alsa-project/alsa-utils/pull/164, has been there for a while...