thesofproject/sof #6090: [BUG][topology] How to get the values declaimed in PIPELINE_PCM_ADD in codec adapter

barry-waves, 2022-08-05:
In sof-tgl-max98357a-rt5682.m4 of adl-004-drop-stable branch, there is a SPK_MIC_PERIOD_US to define the schedule  period.

define(`ENDPOINT_NAME', `Speakers')
PIPELINE_PCM_ADD(
	ifdef(`2CH_2WAY',
	      ifdef(`WAVES', sof/pipe-waves-demux-eq-iir-playback.m4,
		    sof/pipe-demux-eq-iir-playback.m4),
	      ifdef(`WAVES', sof/pipe-waves-codec-demux-playback.m4,
		    ifdef(`DRC_EQ', sof/pipe-drc-eq-volume-demux-playback.m4,
			  sof/pipe-volume-demux-playback.m4))),
	1, 0, ifdef(`4CH_PASSTHROUGH', `4', `2'), s32le,
	SPK_MIC_PERIOD_US, 0, SPK_PLAYBACK_CORE,
	48000, 48000, 48000)
undefine(`ENDPOINT_NAME')')

In ./src/audio/codec_adapter/codec/waves.c, we expect the value of component->period_bytes associates with SPK_MIC_PERIOD_US. However, the value of component->period_bytes is always 192.

/* initializes MaxxEffect based on stream parameters */
static int waves_effect_init(struct comp_dev *dev)
{
    ...
    waves_codec->buffer_bytes = component->period_bytes;
   ...
}

Could you give us some advice to get correct value which associates with SPK_MIC_PERIOD_US? Thank you.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * SOF: adl-004-drop-stable branch
2) Name of the topology file
    * Topology: sof-tgl-max98357a-rt5682.m4 
3) Name of the platform(s) on which the bug is observed.
    * Platform: Intel ADL

aiChaoSONG, 2022-08-09:
I don't think you want  process_module->period_bytes (src/audio/module_adapter/module/waves.c), because it is calculated with
```
	mod->period_bytes = params->sample_container_bytes *
			   params->channels * params->rate / 1000;
```
that's why you always get 192.

can you try the dev->period (comp_dev), or dev->pipeline->period.

barry-waves, 2022-08-09:
Hi Mr. Song,
Thanks for your great help. I can get the correct value.

closed 2022-08-09 by barry-waves