thesofproject/sof #6091: [BUG] IO error and xrun on TGLH RVP NOCODEC ZEPHYR when testing multiple pause resume

XiaoyunWu6666, 2022-08-05:
**Describe the bug**
When multiple pause resume on TGLH RVP NOCODEC ZEPHYR, xrun happened

**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-tgl-h-nocodec.tplg MODEL=TGLH_RVP_NOCODEC_ZEPHYR ~/sof-test/test-case/multiple-pause-resume.sh -r 50

**Reproduction Rate**
Observe only once in CI so far

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: [655b48ffac48](https://github.com/thesofproject/linux/commit/655b48ffac48)
SOF Branch: main
SOF Commit: [8881a406af67](https://github.com/thesofproject/sof/commit/8881a406af67)
Zephyr Commit: [zephyr-v3.1.0-2058-g63a497280f22](https://github.com/zephyrproject-rtos/zephyr/commit/zephyr-v3.1.0-2058-g63a497280f22)
Topology: sof-tgl-h-nocodec.tplg
Platform: TGLH_RVP_NOCODEC_ZEPHYR

**Screenshots or console output**

marc-hb, 2022-08-05:
@XiaoyunWu6666 is a brand new regression? Did it start with
- #5976 

just like this other new regression?

- #6075

marc-hb, 2022-08-05:
`TGL: enable CONFIG_INCOHERENT correctly` was reverted in #6094

plbossart, 2022-08-05:
possible similar sighting in https://sof-ci.01.org/linuxpr/PR3796/build485/devicetest/?model=ADLP_RVP_SDW_ZEPHYR&testcase=multiple-pause-resume-5
xrun leading to IPC error.

lgirdwood, 2022-08-05:
Pause is always a P2.

plbossart, 2022-08-05:
I disagree @lgirdwood, If we end-up with a fatal IPC error it's a P1. Xruns are tolerable since they are not fatal.

lgirdwood, 2022-08-05:
> I disagree @lgirdwood, If we end-up with a fatal IPC error it's a P1. Xruns are tolerable since they are not fatal.

It's P2 as it's real world usage is negligible and I'd rather focus on issues that will hit the majority of users (and we have some). Sure, become a P1 is it causes IPC timeout or crash.

plbossart, 2022-08-05:
https://sof-ci.01.org/linuxpr/PR3796/build485/devicetest/?model=ADLP_RVP_SDW_ZEPHYR&testcase=multiple-pause-resume-5 is an IPC error and crash @lgirdwood 
it's like a kernel oops, it doesn't matter how we got there, it shouldn't happen. Rather, it shall not happen.

lgirdwood, 2022-08-05:
> https://sof-ci.01.org/linuxpr/PR3796/build485/devicetest/?model=ADLP_RVP_SDW_ZEPHYR&testcase=multiple-pause-resume-5 is an IPC error and crash @lgirdwood it's like a kernel oops, it doesn't matter how we got there, it shouldn't happen. Rather, it shall not happen.

This looks like the same issue/use case and it now qualifies as a P1 since it's a FW crash/fatal. i.e no longer a survivable IPC error reply as reported initially but now a fatal IPC timeout.

XiaoyunWu6666, 2022-08-08:
I ran a stress multiple pause resume test on ADLP TGLH TGLU zephyr platforms (NOCODEC and SDW) to see if there are any clues

lgirdwood, 2022-08-10:
@marcinszkudlinski fyi - could be related to issues you have been looking at. @lyakh can you take a look at this for TGL ?

lyakh, 2022-08-16:
Doesn't this look like a #5850 duplicate?

lyakh, 2022-08-18:
I found that daily test result Nr. 14447 and I see the xrun messages as quoted in this report, but I don't see IPC timeouts there?

lyakh, 2022-08-18:
ah, I see that @lgirdwood changed the original report title and added an "IPC timeout" there. Interestingly, I don't see an IPC timeout in *that* daily result, but I found a similar xrun in daily result Nr. 14634 on ADLP RVP nocodec XTOS and there I do see an IPC timeout, so that one is more similar to #5850 although the latter one is on a Chrome device and it's the HDA DMA that's hitting an xrun there, unlike the former, where it's DW DMA.

lyakh, 2022-08-19:
> possible similar sighting in https://sof-ci.01.org/linuxpr/PR3796/build485/devicetest/?model=ADLP_RVP_SDW_ZEPHYR&testcase=multiple-pause-resume-5 xrun leading to IPC error.
> 
> It's a SoundWire v. nocodec configuration though, it could be a separate issue.

indeed, I'd say this one is more like #5850

lyakh, 2022-08-25:
Having studied 5 daily and 1 PR test failures with IPC timeouts I came to the conclusion, that we are looking at 2 failure types:
```
topology	| DW DMA		| HDA DMA	| IPC timeout	| RTOS
-----------------------------------------------------------------------------------
nocodec		| 2 xruns in the end	| Ok		| on SSP	| both
HDA / SDW	| Ok			| many xruns	| on DMIC	| Zephyr
```
I tried to reproduce the problem with the multiple-pause-resume test on an UP-Xtreme TGL board with both topologies (nocodec was modified to only include SSP0) - no success so far.

plbossart, 2022-08-26:
@lyakh I find the table above confusing, if you have a timeout on DMIC, isn't that using the DW DMA? how can we have xruns on the HDA DMA here?

lyakh, 2022-08-26:
> @lyakh I find the table above confusing, if you have a timeout on DMIC, isn't that using the DW DMA? how can we have xruns on the HDA DMA here?

@plbossart I meant, that there are no xruns on DW DMA in those cases. HDA DMA xruns - no idea if they're related, e.g.
```
[     8358500.761613] (        3635.572754) c0 hda-dma            ..../intel/hda/hda-dma.c:948  ERROR hda_dma_link_check_xrun(): underrun detected
[     8360501.699034] (        2000.937378) c0 hda-dma            ..../intel/hda/hda-dma.c:948  ERROR hda_dma_link_check_xrun(): underrun detected
```
see https://sof-ci.01.org/linuxpr/PR3796/build485/devicetest/?model=ADLP_RVP_SDW_ZEPHYR&testcase=multiple-pause-resume-5

plbossart, 2022-08-26:
@lyakh The HDAudio host DMA would be used, there's really no notion of xrun at that level. There's a very large DDR buffer as destination.The xruns only make sense at the link level, no?

lyakh, 2022-09-06:
The most recent occurrence of this bug, known to me was on ADLP_RVP_NOCODEC in daily test result 14634 of August 14 2022. Let's wait for another week until a full month with no reproduction, then we can close this.

kfrydryx, 2022-09-13:
No longer reproducing

closed 2022-09-13 by kfrydryx