thesofproject/sof #6095: [BUG] SOF freezes when running SRC pipeline conversion 44.1khz -> 48khz

dbaluta, 2022-08-05:
**Describe the bug**

SOF freezes when playing file with following parameters: 44.1Khz, 32bits, 2 channels.

**To Reproduce**

Use `sof-imx8mp-src-wm8960.m4` and do:

`aplay -Dhw:0,0 audio41k2c32b.wav`.

aplay freezes and no sound can be heard.

**Reproduction Rate**

always

**Expected behavior**

aplay should play the wav file with accurate sound.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: latest
    * SOF: stable-v2.2 (anything newer reproduces the issue)
2) Name of the topology file
    * Topology: sof-imx8mp-src-wm8960.m4
3) Name of the platform(s) on which the bug is observed.
    * Platform: imx8, imx8x, imx8m

Root cause for this bug is:

```
commit 3aff4598f7ebd79d8cfb0398d85a85bbf4ba37e1
Author: Seppo Ingalsuo <seppo.ingalsuo@linux.intel.com>
Date:   Fri Apr 1 21:22:03 2022 +0300

    Audio: SRC: Increase length of internal buffer
    
    The too small size of this buffer caused a new 11025 to 8000 Hz
    conversion pipeline to freeze due to impossible condition in
    source/internal/sink buffers.
    
    Signed-off-by: Seppo Ingalsuo <seppo.ingalsuo@linux.intel.com>
```

singalsu, 2022-08-08:
Thanks for report @dbaluta ! There seems to be another freeze possibility that that the larger internal buffer triggered.

Have you tried if 11.025 to 8 kHz freezes now?

singalsu, 2022-08-08:
"Only FW compiled with Xtensa toolchain see this issues. GCC builds are ok." --> this is a very good pointer for starting debugging, it sounds like a bug in xcc version

singalsu, 2022-08-09:
In my test with UP2 with modified sof-apl-pcm512x.tplg to add support for s32_le (#6109) I don't get freeze with HiFi3 xt-xcc build with 8881a406af67536b9ac6bfbb53d1c712aacdae81 that is just before @dbaluta 's revert.

I think I need to test with Zephyr since I assume imx8 used it, apply to my test topology similar pipeline buffers parameters as on imx8. Also imx8 seems to be HiFi4, and there's a brand new src_hifi4.c also to inspect.

@RanderWang Did you observe the bug with a HiFi4 platform?

singalsu, 2022-08-09:
Zephyr build is not triggering this issue for me. Has anyone replicated this issue with HiFi3?

dbaluta, 2022-08-10:
> In my test with UP2 with modified sof-apl-pcm512x.tplg to add support for s32_le (#6109) I don't get freeze with HiFi3 xtos xt-xcc build with [8881a40](https://github.com/thesofproject/sof/commit/8881a406af67536b9ac6bfbb53d1c712aacdae81) that is just before @dbaluta 's revert.
> 
> I think I need to test with Zephyr since I assume imx8 used it, apply to my test topology similar pipeline buffers parameters as on imx8. Also imx8 seems to be HiFi4, and there's a brand new src_hifi4.c also to inspect.

I have this replicated with XTOS not Zephyr. Also, have not tried  `11.025 to 8 kHz` as our base is 48Khz and we convert from X to 48Khz always.

I might find some time at the end of the week to test this scenario. 

Also, as you noticed i.MX used HIFI4.

RanderWang, 2022-08-18:
I can reproduce it with Linux kernel + cSOF + cavs-src-gain-mixin.tplg

RanderWang, 2022-08-18:
two issues: 
(1) need to revert 
````
    Revert "Audio: SRC: Fix incorrect and insufficient buffer size check"
    
    This reverts commit 40b6c80f18b4e4f173a0b6ebc2376d4b9bfc2e7a.
````
a error happen 
     src/audio/src/src.c:709  INFO src_params(), source_rate = 44100, sink_rate = 48000
     src/audio/src/src.c:711  INFO src_params(), sourceb->channels = 2, sinkb->channels = 2, dev->frames = 48
     src/audio/src/src.c:720  INFO src_params(), blk_in = 0, blk_out = 0
....../pipeline-params.c:340  INFO pipe prepare
     src/audio/src/src.c:967  INFO src_prepare()
     src/audio/src/src.c:944  ERROR Source size 720 is less than required 744
     src/audio/src/src.c:990  ERROR src_prepare(): source or sink buffer size is insufficient
....../pipeline-params.c:347  ERROR pipeline_prepare(): ret = -105, dev->comp.id = 262144
  src/ipc/ipc4/handler.c:142  ERROR ipc: pipe 1 comp 0 prepare failed -105

(2) with step (1) we will find fw stuck

singalsu, 2022-08-18:
> I can reproduce it with Linux kernel + cSOF + cavs-src-gain-mixin.tplg

This was with xt-xcc? Can you try if gcc built firmware has this issue?

singalsu, 2023-02-14:
@dbaluta Can this bug be closed now after your recent fix?

dbaluta, 2023-02-14:
@singalsu Let me test this today and get back to you.

dbaluta, 2023-02-14:
i should have close this much earlier, it works not because of my recent fix but because we have reverted

```
    Revert "Audio: SRC: Fix incorrect and insufficient buffer size check"
    
    This reverts commit 40b6c80f18b4e4f173a0b6ebc2376d4b9bfc2e7a.
```

closed 2023-02-14 by dbaluta