thesofproject/sof #6100: [BUG] PCM probe does not work with multicore topologies

kv2019i, 2022-08-05:
**Describe the bug**
The pcm probe client uses notifier_register() to register to buffers (to probe). The notifier interface registers the notifier just for the calling CPU and if the buffer is attached to a pipeline running on a different core, no notifications are received and thus no data is probed.

**To Reproduce**
Use a multicore pipeline like cavs-nocodec. Start playback with pipeline assigned to non-zero core, attach probe points to buffers of this pipeline.

**Reproduction Rate**
100%

**Expected behavior**
Probe interface works for all buffers independently of the core they are allocated to.

**Impact**
Unable to use PCM probe to debug issues with multicore topologies.

**Environment**

1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: n/a
    * SOF: 9d37a33335d4a68e0433f6bca6cb5739b06500d5
2) Name of the topology file
    * Topology: sof-tgl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: Intel TGL
   
**Screenshots or console output**

lgirdwood, 2022-08-05:
@kv2019i xtos or zephy ?

kv2019i, 2022-08-05:
@lgirdwood wrote:
> @kv2019i xtos or zephy ?

Both.

lyakh, 2022-08-12:
probes don't acquire buffers correctly, let me try to cook up a patch

kv2019i, 2022-08-15:
Discussion about how to fix this: https://github.com/thesofproject/sof/pull/6137#discussion_r945478298