thesofproject/sof #6101: [BUG] TimeoutError occurred TestPtpParallelInternal

apakuls, 2022-08-05:
**Describe the bug**
ERROR:Exception TimeoutError occurred at 'C:\\work\\tools\\pyexecnetcache\\cavs_scripts-py\\utilities\\hda_streamer.py':600 Play stream not running for 1 seconds

Bug was found in:
10_01_TestPtpParallelInternal8000Hz16b16b2ch
10_01_TestPtpParallelInternal48000Hz32b32b8ch
10_01_TestPtpParallelInternal48000Hz24b32b4ch
10_01_TestPtpParallelInternal24000Hz32b32b8ch

**Topology**
```
                        pipe_plb
                        +--------------------+
        +----------+    | +------+   +-----+ |   +----+
        | Host plb |----->|Buffer|-->| Dai |---->|SSP0|
        +----------+    | +------+   +-----+ |   +----+
                        +--------------------+    |  |
                                                  |  |
                        pipe_cap                  |  |
                        +--------------------+    |  |
        +----------+    | +------+   +-----+ |   +----+
        | Host cap |<-----|Buffer|<--| Dai |<----|SSP0|
        +----------+    | +------+   +-----+ |   +----+
                        +--------------------+

        (...)
        SSPn += 1

                        pipe_plb
                        +--------------------+
        +----------+    | +------+   +-----+ |   +----+
        | Host plb |----->|Buffer|-->| Dai |---->|SSPn|
        +----------+    | +------+   +-----+ |   +----+
                        +--------------------+    |  |
                                                  |  |
                        pipe_cap                  |  |
                        +--------------------+    |  |
        +----------+    | +------+   +-----+ |   +----+
        | Host cap |<-----|Buffer|<--| Dai |<----|SSPn|
        +----------+    | +------+   +-----+ |   +----+
                        +--------------------+

```
**To Reproduce**
Run tests with:
10_01_TestPtpParallelInternal8000Hz16b16b2ch

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash  
   * SOF BRANCH: main HASH: 2cfefc1103ef73713031a9f076dba953357f0229
2) Name of the platform(s) on which the bug is observed.
    * Platform: TGL RVP

**Logs**
[TestPtpParallelInternal8000Hz16b16b2ch.zip](https://github.com/thesofproject/sof/files/9269182/TestPtpParallelInternal8000Hz16b16b2ch.zip)

apakuls, 2022-08-05:
Also occurs on:
12_01_TestMixerCapture8000Hz16b16b2ch
12_01_TestMixerCapture16000Hz16b16b8ch
12_01_TestMixerCapture24000Hz24b32b4ch
12_01_TestMixerCapture48000Hz32b32b4ch

14_03_TestDemuxSynchronizedPlayback8000Hz24b32b2ch
14_03_TestDemuxSynchronizedPlayback24000Hz24b32b8ch
14_10_TestDemuxMemLeakStaticPipelines

lgirdwood, 2022-08-05:
@apakuls can you confirm xtos/zephyt and whether multicore ?

wszypelt, 2022-08-08:
@lgirdwood Platform issue

closed 2022-08-08 by wszypelt