thesofproject/sof #6102: [BUG] IPC timeout after 5.0 seconds in test 14_05_TestDemuxSynchronizedSimpleStreamOperations8000Hz16b16b4ch

apakuls, 2022-08-05:
**Describe the bug**
ERROR:Exception TimeoutError occurred at 'C:\\work\\tools\\cavs_scripts-py\\utilities\\ipc_driver.py':171 FW status on IPC timeout after 5.0 seconds: 0x00000005 0x00000000 0x00004000

Bug was found in:
14_05_TestDemuxSynchronizedSimpleStreamOperations8000Hz16b16b4ch
14_05_TestDemuxSynchronizedSimpleStreamOperations16000Hz24b32b4ch
14_05_TestDemuxSynchronizedSimpleStreamOperations24000Hz32b32b2ch
14_05_TestDemuxSynchronizedSimpleStreamOperations48000Hz32b32b4ch

**Topology**
```
        pipe_cap
    +-------------------------------+
    | +------+   +------+   +-----+ |
    | | Host |<--| Buff |<--| Dai |<------------------------------------------------------<
    | +------+   +------+   +-----+ |                                                     |
    +-------------------------------+                                                     |
                                                                                          |
         pipe_plb                                                                         |
    +----------------------------------------------------------------------------+        |
    | +------+   +------+   +-----+   +------+   +-------+   +------+   +------+ |     +------+
    | | Host |-->| Buff |-->| Vol |-->| Buff |-->| DeMux |-->| Buff |-->| Dai1 |------>| SSPx |
    | +------+   +------+   +-----+   +------+   +-------+   +------+   +------+ |     +------+
    +-----------------------------------------------|----------------------------+
                                                    |        pipe_plb
                                                    |      +---------------------+
                                                    |      | +------+   +------+ |    +--------+
                                                    +------->| Buff |-->| Dai2 |----->| SSPx+1 |
                                                           | +------+   +------+ |    +--------+
                                                           +---------------------+        |
        pipe_cap                                                                          |
    +-------------------------------+                                                     |
    | +------+   +------+   +-----+ |                                                     |
    | | Host |<--| Buff |<--| Dai |<------------------------------------------------------<
    | +------+   +------+   +-----+ |
    +-------------------------------+

```
**To Reproduce**
Run tests with:
14_05_TestDemuxSynchronizedSimpleStreamOperations8000Hz16b16b4ch

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash  
   * SOF BRANCH: main HASH: 2cfefc1103ef73713031a9f076dba953357f0229
2) Name of the platform(s) on which the bug is observed.
    * Platform: TGL RVP

**Logs**
[TGL_ipc_Timeout.zip](https://github.com/thesofproject/sof/files/9269398/TGL_ipc_Timeout.zip)

lgirdwood, 2022-08-05:
@apakuls xtos or Zephyr ?

wszypelt, 2022-08-08:
@lgirdwood Platform issue

closed 2022-08-08 by wszypelt