thesofproject/sof #6111: [BUG] Dell XPS 13 9315: No sound cards detected

paulmenzel, 2022-08-09:
On the Dell XPS 13 9315 with Debian sid/unstable and Linux 5.19 from the suite *experimental*, no soundcards are found.

    00:1f.3 Multimedia audio controller [0401]: Intel Corporation Device [8086:51cc] (rev 01)

[Output of `/usr/sbin/alsa-info`](https://alsa-project.org/db/?f=debacf544a4340e3f6013a102a235acfe777ca73)

paulmenzel, 2022-08-09:
Added soundwire error messages.

plbossart, 2022-08-09:
know issue duplicate of https://github.com/thesofproject/linux/issues/3777

Solved by upstream commits, see https://lore.kernel.org/alsa-devel/20220621225641.221170-1-pierre-louis.bossart@linux.intel.com/

paulmenzel, 2022-08-09:
Thank you for the prompt reply. I am happy fixes are in Linus’ master branch.

Two questions:

1.  Is that a regression in Linux 5.18?
2.  Did Dell send one of these devices to Intel, so they can be used for testing?

paulmenzel, 2022-08-09:
For the record, the [Linux messages](https://github.com/thesofproject/sof/files/9292818/20220808-dell-xps-13-9315-ubuntu-20.04-5.14.0-1047.54-oem-dmesg.txt) from the Dell OEM Ubuntu 22.04 installation.

plbossart, 2022-08-10:
> * Is that a regression in Linux 5.18?

Unclear. The probe message has been there for multiple cycles, since commit 528be501b7d4a ("soundwire: sdw_slave: add probe_complete structure and new fields")

it's not clear what caused the timeout in practice. Possibly PCI power management?

> * Did Dell send one of these devices to Intel, so they can be used for testing?

Yes, and we didn't see the issue with the Canonical OEM image. It's likely a configuration that causes delays in the probe and timeouts.

paulmenzel, 2022-08-10:
> > * Did Dell send one of these devices to Intel, so they can be used for testing?
> 
> Yes, and we didn't see the issue with the Canonical OEM image. It's likely a configuration that causes delays in the probe and timeouts.

Yes, me neither, but I’d thought the Intel (FLOSS) QA is going to test mainly with the upstream versions.

plbossart, 2022-08-10:
I was the author of that broken solution in commit 528be50 ("soundwire: sdw_slave: add probe_complete structure and new fields") and we never saw any issues in all our tests. It's just luck that we had a solution ready: I was doing some cleanups to make the driver more compliant with the device model, and remove all sorts of timing dependencies to rely instead of the deferred probe mechanism. I had no idea that cleanup would actually solve a real problem, it was only necessary in academic corner cases with some drivers blacklisted and manually inserted after a long delay.

IchikaZou, 2022-11-07:
Hi, I running kernel version 5.19.17 and I encountered the same problem on Dell XPS 9320, does the patch only work above 5.19.17?

plbossart, 2022-11-07:
The recommendation is to update to version 6.0.6 and later.

IchikaZou, 2022-11-07:
> The recommendation is to update to version 6.0.6 and later.

I see, thanks. 

I need to wait the kernel version 6 of lts come out so zfs could bump the Linux compatible.

nileshpatra, 2022-11-09:
Hi, I am on 6.0.7 but the issue seems to still persist.

```
$ uname -a
Linux debian 6.0.0-3-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.0.7-1 (2022-11-05) x86_64 GNU/Linux
```

The version of `firmware-sof-signed` is `2.2.2-1`

And
output of `[/usr/sbin/alsa-info](http://alsa-project.org/db/?f=4475830b7ee10217918453286dd5ef83b51804e6)` shows:

```
[   10.913863] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[   10.914021] sof_sdw sof_sdw: snd_soc_register_card failed -517
[   10.919220] sof_sdw sof_sdw: snd_soc_register_card failed -517
[   15.081027] wlp0s20f3: authenticate with b4:a7:c6:52:b9:50
--
[   15.499648] IPv6: ADDRCONF(NETDEV_CHANGE): wlp0s20f3: link becomes ready
[   21.018708] sof_sdw sof_sdw: snd_soc_register_card failed -517
[   21.018753] platform sof_sdw: deferred probe pending
```

I am quite clueless at the moment as to what should be done. I'd appreciate any inputs.

paulmenzel, 2022-11-09:
@plbossart, see the string at the end: 6.0.7-1.

@nileshpatra, the upstream stable series and Debian does not cherry-pick/carry the patch fixing the issue in contrast to Arch Linux, cf issue https://github.com/thesofproject/linux/issues/3777.

IchikaZou, 2022-12-06:
I see, I would contact gentoo kernel maintainers to see weather they can't help to patch the kernel. 

The patch is in here right?

https://lore.kernel.org/alsa-devel/20220621225641.221170-4-pierre-louis.bossart@linux.intel.com/

plbossart, 2022-12-12:
The backport was picked by GregKH and will be in 6.0.13 (it's not in 6.0.12), so closing this bug.

closed 2022-12-12 by plbossart

IchikaZou, 2022-12-15:
Thanks for the update.

IchikaZou, 2022-12-16:
Hi, I just update to 6.0.13 but my sound card still not detected.

paulmenzel, 2023-01-09:
Does it work with Linux 6.1.x?

IchikaZou, 2023-01-10:
> Does it work with Linux 6.1.x?

Not yet, openzfs haven't add support for kernel 6.1.x yet.

usrhere, 2023-07-10:
Still no audio on  Dell '**XPS-13 9315 2-in-1**', did someone have success with **2-in-1** model?`
```
 $ uname -r
 6.4.1-arch2-1
```

```
$ lspci | grep -i audio
0000:00:1f.3 Audio device: Intel Corporation Alder Lake Smart Sound Technology Audio Controller (rev 01)
```

```
$ sudo dmesg | grep 0000:00:1f.3
[    0.605112] pci 0000:00:1f.3: [8086:51cc] type 00 class 0x040380
[    0.605176] pci 0000:00:1f.3: reg 0x10: [mem 0x6040290000-0x6040293fff 64bit]
[    0.605250] pci 0000:00:1f.3: reg 0x20: [mem 0x6040000000-0x60400fffff 64bit]
[    0.605411] pci 0000:00:1f.3: PME# supported from D3hot D3cold
[    1.234132] pci 0000:00:1f.3: Adding to iommu group 16
[    4.870957] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    4.870980] snd_hda_intel 0000:00:1f.3: SoundWire enabled on CannonLake+ platform, using SOF driver
[    4.997577] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
[    4.997591] sof-audio-pci-intel-tgl 0000:00:1f.3: SoundWire enabled on CannonLake+ platform, using SOF driver
[    4.997606] sof-audio-pci-intel-tgl 0000:00:1f.3: enabling device (0000 -> 0002)
[    4.997715] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[    4.997773] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    5.004264] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    5.026984] sof-audio-pci-intel-tgl 0000:00:1f.3: No SoundWire machine driver found
[    5.026987] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 4
[    5.026988] sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    5.026990] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 0
[    5.031919] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    5.031921] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[    5.031924] sof-audio-pci-intel-tgl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[    5.126944] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    5.126947] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:0
[    5.137192] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:22:1 Kernel ABI 3:23:0
[    5.170252] input: sof-hda-dsp HDMI/DP,pcm=1 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input16
[    5.170299] input: sof-hda-dsp HDMI/DP,pcm=2 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input17
[    5.170332] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input18
```

```
$ lsmod | grep snd_sof_pci_intel_tgl
snd_sof_pci_intel_tgl    12288  0
snd_sof_intel_hda_common   229376  1 snd_sof_pci_intel_tgl
snd_sof_pci            24576  2 snd_sof_intel_hda_common,snd_sof_pci_intel_tgl
snd_soc_acpi_intel_match    77824  2 snd_sof_intel_hda_common,snd_sof_pci_intel_tgl
```

```
$ pactl info
Server String: /run/user/1000/pulse/native
Library Protocol Version: 35
Server Protocol Version: 35
Is Local: yes
Client Index: 197
Tile Size: 65472
User Name: jo
Host Name: 2in1
Server Name: PulseAudio (on PipeWire 0.3.72)
Server Version: 15.0.0
Default Sample Specification: float32le 2ch 48000Hz
Default Channel Map: front-left,front-right
Default Sink: alsa_output.pci-0000_00_1f.3-platform-skl_hda_dsp_generic.pro-output-3
Default Source: alsa_output.pci-0000_00_1f.3-platform-skl_hda_dsp_generic.pro-output-1.monitor
Cookie: 0411:1d93
```

Thanks

paulmenzel, 2023-07-10:
@usrhere, yes the problem is fixed on the Dell XPS 13 9315, and also the 9320 2-in-1 to my knowledge. Please create a new issue.

usrhere, 2023-08-22:
> Well it isn't

Please check https://github.com/thesofproject/linux/issues/4522