thesofproject/sof #6116: [BUG] microphone not detected and not working on 2022 Razer Blade 14

turowicz, 2022-08-10:
**Describe the bug**
I've installed Ubuntu 22.04 on a 2022 Razer Blade 14. Audio playback appears to be working, but the built in microphone, is not being detected, doesn't seem to be recording any sound. I am also not able to detect a connected BT speaker with a mic (Jabra).

[image]

**To Reproduce**
Clean install of Ubuntu 22.04. I have updated the kernel to 5.19.0 and applied ath11k patches to get wifi support.

**Reproduction Rate**
100%

**Expected behavior**
I expect the microphone to be detected and to record when enabled.

**Impact**
I can't use the built in microphone for video conferencing. I have tried bluetooth microphones with no luck.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: `5.19.0-051900-generic`
    * SOF: `firmware-sof-signed/jammy-updates,jammy-updates,now 2.0-1ubuntu3 all [installed,automatic]`
2) Name of the topology file
    * Topology: I'm not sure how to determine this.
3) Name of the platform(s) on which the bug is observed.
    * Platform: x86_64

**Screenshots or console output**
dmesg: [dmesg.zip](https://github.com/thesofproject/sof/files/9300310/dmesg.zip)

plbossart, 2022-08-10:
@turowicz an you please attach the result of 'alsa-info', it's not clear that you are even using the SOF driver?

turowicz, 2022-08-10:
http://alsa-project.org/db/?f=c857e693b115cecaa37910551102054d8c9e1c8c

plbossart, 2022-08-10:
That's an AMD device, that will be for the AMD folks to comment on. I have no idea if this platform is supposed to be handled by SOF.

turowicz, 2022-08-10:
I came here from there: https://github.com/thesofproject/sof/issues/5989

lgirdwood, 2022-08-10:
@thesofproject/amd fyi.

flyingfalling, 2022-08-31:
I have an identical issue (identical laptop). I don't see SOF stuff being loaded at all (which may be the issue?), but the similar laptop (razer blade 15 2022) problem was solved by a change in topology file, and so we were hoping such a simple fix was available for the 14-inch version as well. I was directed here by: https://www.linuxquestions.org/questions/linux-hardware-18/audio-issues-with-razer-blade-15-2022-a-4175708817/page2.html which referenced the solution of SOF project members for the 15-inch model in https://github.com/thesofproject/sof/issues/5989

The difference is of course that the 14-inch uses that AMD ryzen 6000 processor, whereas the 15-inch uses the 12-gen intel one.

I am attaching alsa info (although it is largely identical to the poster above):
http://alsa-project.org/db/?f=896e5ae8fbb2855d67023d2beb9469da52d782bc

You mentioned that the part is AMD card...but your referenced AMD team here @thesofproject/amd does not seem to exist?

lgirdwood, 2022-08-31:
> You mentioned that the part is AMD card...but your referenced AMD team here @thesofproject/amd does not seem to exist?

@balakishorepati fyi

pramodhrachuri, 2023-04-09:
any update on this?

turowicz, 2023-04-12:
I just installed kernel 6.2 and problem disappeared

closed 2023-04-12 by turowicz