thesofproject/sof #6118: [BUG][IPC4] ipc timed out for 0x13020004|0x0 GLB_SET_PIPELINE_STATE on TGLU_UP_HDA_IPC4ZPH

keqiaozhang, 2022-08-10:
**Describe the bug**
This issue happens when testing playback on TGLU_UP_HDA with IPC4 SOF Zephyr configuration. 
The IPC error is:
```
[   60.818680] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 2 state 4
[   60.818682] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13020004|0x0: GLB_SET_PIPELINE_STATE
[   61.320573] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13020004|0x0
[   61.320576] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to set state 4 for pipeline 2
[   61.320579] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[   61.320583] kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -110
```
It seem that this issue only happens with HDA mode. we didn't observer this issue with SDW or I2S mode.

**To Reproduce**
TPLG=/lib/firmware/intel/avs-tplg/cavs-mixin-mixout-hda.tplg MODEL=TGLU_UP_HDA_IPC4ZPH ~/sof-test/test-case/check-playback.sh -d 10 -l 1 -r 1

**Reproduction Rate**
100%

** TPLG **
[image]

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev/[67fc9186384f](https://github.com/thesofproject/linux/commit/67fc9186384f)
    * SOF: main/[a10e118a4d25](https://github.com/thesofproject/sof/commit/a10e118a4d25)
    * Zephyr:[8e55e59c5917](https://github.com/zephyrproject-rtos/zephyr/commit/zephyr-v3.1.0-3042-g8e55e59c5917)
2) Name of the topology file
    * Topology:cavs-mixin-mixout-hda.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGLU_UP_HDA_IPC4ZPH

[dmesg.txt](https://github.com/thesofproject/sof/files/9300633/dmesg.txt)

keqiaozhang, 2022-08-23:
This issue is still reproducible on TGLU_UP_HDA_IPC4ZPH.
Please refer to inner test: http://sof-ci.sh.intel.com/#/result/planresultdetail/14754?model=TGLU_UP_HDA_IPC4ZPH&testcase=check-capture-all-formats

lyakh, 2022-08-23:
> This issue is still reproducible on TGLU_UP_HDA_IPC4ZPH. Please refer to inner test: http://sof-ci.sh.intel.com/#/result/planresultdetail/14754?model=TGLU_UP_HDA_IPC4ZPH&testcase=check-capture-all-formats

@keqiaozhang correct me if I'm wrong, but the original bug description was for playback, whereas this bug, that you linked here, seems to be for capture? It *can* well be the same kind prob

closed 2022-08-26 by lyakh

keqiaozhang, 2022-08-29:
I have filed a new one to track the capture issue:https://github.com/thesofproject/sof/issues/6203

keqiaozhang, 2022-08-29:
> but the original bug description was for playback, whereas this bug, that you linked here, seems to be for capture?

I have filed a new one to track the capture issue:https://github.com/thesofproject/sof/issues/6203

PR #6133 (2022-08-12): mixin-mixout: fix a firmware exception
find_mixout_source_info() can return NULL, always check return value before dereferencing.

Partially fixes #6118 - no more firmware crash, but the error has to be fixed too.