thesofproject/sof #6121: [BUG] script xtensa-build-zephyr.py fails on python 3.6

aborisovich, 2022-08-10:
**Describe the bug**
Running script `./scripts/xtensa-build-zephyr.py tgl` on Python 3.6.X will result in error since `shlex.join`
was introduced in Python 3.8:
[image]

**To Reproduce**
Install Python 3.6.X according to instructions on [Zephyr - Getting Started](https://docs.zephyrproject.org/latest/develop/getting_started/index.html) guide.
Follow build instructions defined in SOF docs [Build with Zephyr](https://thesofproject.github.io/latest/getting_started/build-guide/build-with-zephyr.html) and use xtensa-build-zephyr.py script.

**Reproduction Rate**
Always.

**Expected behavior**
SOF project documentation contains information that Python 3.8.X+ is required or
script is fixed and working with declared by Zephyr Python 3,6.X version.

**Impact**
Annoyance and confusion.

**Environment**
Where bug occurs:
    * SOF: main
OS: Windows, Linux
Bug introduced in commit ce8bc4a3941a5f005427caf93dea32d996161ef2

nashif, 2022-08-10:
> Install Python 3.6.X according to instructions on [Zephyr - Getting Started](https://docs.zephyrproject.org/latest/develop/getting_started/index.html) guide.

that is the minimum required, you do not need to use it, you should use a more recent version if you can.

aborisovich, 2022-08-11:
@lgirdwood @marc-hb can we define Python 3.8.x as minimal requirement for SOF project?
Which Python version CI runs (for both Zephyr and non-Zephyr CI)?

lgirdwood, 2022-08-15:
> @lgirdwood @marc-hb can we define Python 3.8.x as minimal requirement for SOF project? Which Python version CI runs (for both Zephyr and non-Zephyr CI)?

Ack - I think most distros are running 3.10 now.

marc-hb, 2022-08-15:
tl;dr: 3.8 requirement is fine by me.

Python 3.6 is out of support, should never be used: https://www.python.org/downloads

Zephyr requirements are one thing, SOF requirements are another. Maybe Zephyr wants to be extraordinarily nice and not break backward-compatibility with insecure 3.6 but that does not mean SOF has to be as "nice". Zephyr has obviously many more users than SOF.

3.7 is supported for only one more year.

3.8 was released in 2019, I think that's plenty enough time ago.

aborisovich, 2022-08-18:
Ok then, is anyone willing to update SOF project documentation with this information?
I think it should be placed somewhere in the https://thesofproject.github.io/latest/getting_started/index.html#build-sof .

lgirdwood, 2022-08-18:
@marc-hb could you update sof-docs with Python version and the new alsa-lib version (merged) ?

marc-hb, 2022-08-18:
Python >= 3.8 assert submitted in #6162. Assert is better than documentation: documentation gets out of date and people don't read it (gross exaggeration to get the point across).

For alsa-lib sof-docs has always been recommending to build from git (but many people skip that)

PR #6162 (2022-08-19): xtensa-build-zephyr.py: require python 3.8
Fixes #6121
commit 3c6dcdb5 (2022-08-19): xtensa-build-zephyr.py: abort when python version < 3.8

shlex.join() requires 3.8, maybe others too. 3.8 is 3 years old and it's
the default Python version in Ubuntu 20.04

Fixes #6121
closed 2022-08-19 by lgirdwood