thesofproject/sof #6122: [BUG] script xtensa-build-zephyr.py fails on Windows

aborisovich, 2022-08-10:
Script xtensa-build-zephyr.py fails its execution on Windows system.
It looks like west_root_dir variable value returned by west tool has postfix of "/r" (carriage return) character (where on linux os it is a "\n" (newline character).
[image]

**To Reproduce**
Execute command `./sof/scripts/xtensa-build-zephyr.py tgl`

**Reproduction Rate**
100%

**Expected behavior**
Script `./scripts/xtensa-build-zephyr.py` works on Windows OS without errors.

**Impact**
Annoyance - we still can build using west tool directly according to [Build with Zephyr](https://thesofproject.github.io/latest/getting_started/build-guide/build-with-zephyr.html) guide.

**Environment**
1) Branch name and commit hash:
    * SOF: main
2) Operating system:
    * Windows
3) Name of the platform(s) on which the bug is observed.
    * Platform: all

Bug introduced by commit 608833cf27afd0c327e97929f8c0222064c4ee71 .

**Screenshots or console output**
[image]

lgirdwood, 2022-08-15:
> It looks like west_root_dir variable value returned by west tool has postfix of "/r" (carriage return) character (where on linux os it is a "\n" (newline character). [image]

@aborisovich do we need an ```if (WIN32)``` here in the script do use ```\r```?

aborisovich, 2022-08-15:
Well yeah but in bug resolution I just stripped down all trailing whitespaces.
Basically on linux OS west command `west topdir` returns eg.:
_/home/root/projects/sof_workspace/sof\n_
And on Windows OS west topdir returns eg.:
_C:\workspace\sof_workspace\sof/r_
(note trailing visible whitespaces).
So I just removed them all.

closed 2022-08-15 by aborisovich