thesofproject/sof #6127: SOF binary blob release

kuanhsuncheng, 2022-08-11:
Hi partners,

This Allen from MediaTek.

We are going to develop sof with other third party colleagues in mt8186.

If we want to share the SOF binary blob to to them, is it ok we send a PR to linux-firmware.git ?

or could we create a branch in thesofproject/sof-bin?

Please kindly let me know if you have any concern.

Thanks.

kuanhsuncheng, 2022-08-12:
Hi @dbaluta  @plbossart @lgirdwood @marc-hb,

Please let us know, if there is a better way to do, thanks.

lgirdwood, 2022-08-15:
@kuanhsuncheng this is an interesting discussion. There are some complexities around deploying the FW binaries and the topology files and the debug trace files. e.g. trying to keep all in sync when topology will update faster.
There was also a latency in getting code into linux-firmware, but this may be faster now.

Intel decided to use a sof-bin repo to solve the synchronization with topology and allow for fats updates. Both Redhat and Ubuntu pick this up and and make packages. You are free to use sof-bin if required (it has all the scripts for installation etc).

kuanhsuncheng, 2022-08-15:
Hi Liam, 

Thanks for your reply. That will be great if we can use sof-bin.

@tinghan-shen send a pull request to linux firmware.
https://lore.kernel.org/linux-firmware/bee56d689e67b4bb4d46d14d07e93ab212a6bb6b.camel@mediatek.com/T/#u

We will adjust according to PR upload status in linux firmware.

Thank you for the information about my question.
Allen

closed 2022-08-15 by kuanhsuncheng