thesofproject/sof #6136: [BUG] [XTOS/Zephyr] IPC4: any pipeline with MIXIN/MIXOUT module is broken

ujfalusi, 2022-08-12:
**Describe the bug**
Any IPC4 pipeline involving a MIXIN/MIXOUT module does not work, as of now causing a firmware crash with NULL pointer dereference

**To Reproduce**
Boot a machine using cavs-mixin-mixout-hda.tplg (upXtreme-i11 for example)
Try to play audio on analog output:
aplay -i -Dplughw:0,0 <file>

**Reproduction Rate**
100/100

**Expected behavior**
Audio is playing.

**Impact**
Makes any pipeline which have MIXIN/MIXOUT widget non working

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: HEAD
    * SOF: HEAD
2) Name of the topology file
    * Topology: cavs-mixin-mixout-hda.tplg (copied as sof-hda-generic.tplg)
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL (upx-i11)

**Screenshots or console output**
```
[ 5693.565417] snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[ 5693.565968] snd_sof_intel_hda_common:hda_dsp_stream_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x14001e successful
[ 5693.565974] snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 1 state 3
[ 5693.565979] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13010003|0x0: GLB_SET_PIPELINE_STATE
[ 5693.566055] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000000|0x0: GLB_SET_PIPELINE_STATE
[ 5693.566071] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x13010003|0x0: GLB_SET_PIPELINE_STATE
[ 5693.566075] snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 1 state 4
[ 5693.566079] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13010004|0x0: GLB_SET_PIPELINE_STATE
[ 5693.566563] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000000|0x0: GLB_SET_PIPELINE_STATE
[ 5693.566593] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x13010004|0x0: GLB_SET_PIPELINE_STATE
[ 5693.566599] snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 2 state 3
[ 5693.566614] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13020003|0x0: GLB_SET_PIPELINE_STATE
[ 5693.566695] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000000|0x0: GLB_SET_PIPELINE_STATE
[ 5693.566709] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x13020003|0x0: GLB_SET_PIPELINE_STATE
[ 5693.566713] snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 2 state 4
[ 5693.566718] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13020004|0x0: GLB_SET_PIPELINE_STATE
[ 5694.067009] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13020004|0x0
[ 5694.067015] sof-audio-pci-intel-tgl 0000:00:1f.3: failed to set state 4 for pipeline 2
[ 5694.067019] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[ 5694.067024]  HDA Analog: ASoC: trigger FE cmd: 1 failed: -110
```
FW log:
```
:00:00.000,000] <inf> dma_intel_adsp_gpdma: intel_adsp_gpdma_init: dma dma@7d000 initialized
[00:00:00.000,000] <inf> dma_intel_adsp_gpdma: intel_adsp_gpdma_init: dma dma@7c000 initialized
*** Booting Zephyr OS build zephyr-v3.1.0-3042-g8e55e59c5917  ***
[00:00:00.000,000] <inf> main: SOF on intel_adsp_cavs25
544695 zephyr_ll_scheduler_init(): unsupported domain 2
[00:00:00.000,000] <inf> main: SOF initialized
560144 zephyr_ll_scheduler_init(): unsupported domain 2
6951312 failed to find dai comp or sink pipeline not running.
6953813 failed to find dai comp or sink pipeline not running.
[00:00:00.168,000] <err> os:  ** FATAL EXCEPTION
[00:00:00.168,000] <err> os:  ** CPU 0 EXCCAUSE 13 (load/store PIF data error)
[00:00:00.168,000] <err> os:  **  PC 0xbe032fc8 VADDR 0x4
[00:00:00.168,000] <err> os:  **  PS 0x60020
[00:00:00.168,000] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:0 CALLINC:2)
[00:00:00.168,000] <err> os:  **  A0 0xbe024498  SP 0x9e06e2b9  A2 0x9e072680  A3 0x9e0743d0
[00:00:00.168,000] <err> os:  **  A4 0xffffff70  A5 (nil)  A6 0x7fffffff  A7 0xbe072940
[00:00:00.168,000] <err> os:  **  A8 0xbe074780  A9 0x300 A10 (nil) A11 0x20000000
[00:00:00.168,000] <err> os:  ** A12 0xffffffff A13 (nil) A14 0x9e074810 A15 0xbe024420
[00:00:00.168,000] <err> os:  ** LBEG 0xbe032f7c LEND 0xbe032f90 LCOUNT (nil)
[00:00:00.168,000] <err> os:  ** SAR 0x20

Backtrace:0xBE032FC5:0xBE05D610 0xBE024495:0xBE05D670 0xBE0244F0:0xBE05D690 0xBE02401B:0xBE05D710 0xBE0205B5:0xBE05D760 0xBE024E3D:0xBE05D7A0 0xBE014103:0xBE05D7F0 

[00:00:00.168,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.168,000] <err> os: Current thread: 0x9e070938 (unknown)
[00:00:00.169,000] <err> os: Halting system
```
The crash happens in `src/audio/mixin_mixout.c`  `mixin_copy()` because `src_info` is **NULL**

**Note**
https://github.com/thesofproject/sof/pull/6133 fixes the crash but audio is still not working.

ujfalusi, 2022-08-12:
FYI @serhiy-katsyuba-intel, @lyakh

lyakh, 2022-08-12:
#6118 is also an instance of this bug

lyakh, 2022-08-15:
should be fixed by #6138

ujfalusi, 2022-08-16:
With https://github.com/thesofproject/sof/pull/6138 merged, the issue is resolved.
Thanks again @lyakh !

closed 2022-08-16 by ujfalusi