thesofproject/sof #6154: [FEATURE]: Add support to SOF mixer to overcome saturation possibility

jairaj-arava, 2022-08-17:
Adding multiple PCM streams (pipelines) to the mixer component may result in possibility of Saturation in the output. Hence, need mixer enhancements to overcome this.

jairaj-arava, 2022-08-17:
@sathya-nujella

plbossart, 2022-08-18:
This is a decade-old problem that's simply fixed by adjusting the gains on all paths prior to mixing to make sure there's no saturation added. In other words, the problem is solved PRIOR to mixing.

If you want a more elaborate solution, the mixer will have to include a dynamic range control unit which is considerably more expensive in terms of processing power, latency if a look-ahead buffer is added, and tuning becomes a lot more complicated.

singalsu, 2023-09-28:
This could be achieved now in modular way in topology with attenuation in each mixer input (gain), mixin/mixout, and DRC with target gain and limiter to restore the level to e.g. gain of 1 when single stream is present.

singalsu, 2025-02-13:
Mixin-mixout supports the mix gain, and we have DRC available. This is possible with current SOF with suitable topology.

closed 2025-02-13 by singalsu