thesofproject/sof #6185: [BUG][ADL] ipc tx timed out for 0x60050000| TRIG_STOP when testing multiple pipeline on ADLP-NOCODEC

keqiaozhang, 2022-08-25:
**Describe the bug**
Observed this issue in our CI daily test.  There's a ipc tx timed out for 0x60050000| TRIG_STOP when testing `multiple-pipeline-playback-50.sh`. There's a dai xrun error in etrace. 

**dmesg**
```
[ 4713.598286] kernel: snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 4715.300735] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[ 4715.300783] kernel: snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 4715.316684] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 3 dir 0 cmd 0
[ 4715.804568] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x60050000 (msg/reply size: 12/12)
[ 4715.804594] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[ 4715.804602] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 4715.804618] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000781 rirb 00
[ 4715.804629] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 4715.804644] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 4715.804653] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[ 4715.804658] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[ 4715.804663] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[ 4715.804669] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[ 4715.804682] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x00000005: module: ROM, state: FW_ENTERED, running
[ 4715.804825] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: unexpected fault 0x00000000 trace 0x00004000
[ 4715.804834] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
```
**etrace**
```
[   124817824.363102] (     2660270.250000) c2 dma-trace              src/drivers/dw/dma.c:1128 ERROR dw_dma_get_data_size(): xrun detected
[   124817851.029768] (          26.666666) c2 dma-trace            src/audio/dai-legacy.c:905  ERROR dai_report_xrun(): underrun due to no data available
[   124817872.696433] (          21.666666) c2 dma-trace            src/audio/dai-legacy.c:787  ERROR comp_underrun(): dev->comp.id = 2, source->avail = 192, copy_bytes = 0
[   137303672.044042] (    12485799.000000) c0 dma-trace              src/drivers/dw/dma.c:1128 ERROR dw_dma_get_data_size(): xrun detected
[   137303702.356541] (          30.312498) c0 dma-trace            src/audio/dai-legacy.c:905  ERROR dai_report_xrun(): underrun due to no data available
[   137303729.075290] (          26.718748) c0 dma-trace            src/audio/dai-legacy.c:787  ERROR comp_underrun(): dev->comp.id = 26, source->avail = 0, copy_bytes = 0
[   137303809.908620] (          80.833328) c0 dma-trace          ....../pipeline-stream.c:143  ERROR pipeline_copy(): ret = -61, start->comp.id = 26, dir = 1
[   137303838.137786] (          28.229166) c0 dma-trace          ..../pipeline-schedule.c:208  ERROR pipeline_task(): xrun recovery failed! pipeline is stopped.
[   138311675.389404] (     1007837.250000) c3 dma-trace              src/drivers/dw/dma.c:1128 ERROR dw_dma_get_data_size(): xrun detected
[   138311700.753987] (          25.364582) c3 dma-trace            src/audio/dai-legacy.c:905  ERROR dai_report_xrun(): underrun due to no data available
[   138311722.316486] (          21.562500) c3 dma-trace            src/audio/dai-legacy.c:787  ERROR comp_underrun(): dev->comp.id = 14, source->avail = 0, copy_bytes = 0
```
**To Reproduce**
TPLG=/lib/firmware/intel/sof-tplg/sof-adl-nocodec.tplg MODEL=ADLP_RVP_NOCODEC ~/sof-test/test-case/multiple-pipeline.sh -f p -l 50

**Reproduction Rate**
Only observed this issue once in CI, no more reproductions so far.

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 
[29675885dc03](https://github.com/thesofproject/linux/commit/29675885dc03)

SOF Branch: main
SOF Commit: 
[141efbd7b0e9](https://github.com/thesofproject/sof/commit/141efbd7b0e9)
 
[image]

[dmesg.txt](https://github.com/thesofproject/sof/files/9420717/dmesg.txt)
[logger.txt](https://github.com/thesofproject/sof/files/9420718/logger.txt)

mengdonglin, 2022-08-25:
Closing as this issue is observed with XTOS building. Will focus on Zephyr building on main branch for TGL/ADL platforms.

plbossart, 2022-08-25:
we have to fix this, otherwise changes for PR #https://github.com/thesofproject/sof/pull/6131 will not work either.

keqiaozhang, 2022-08-29:
Observed this issue again in CI stress test: http://sof-ci.sh.intel.com/#/result/planresultdetail/14917?model=ADLP_RVP_NOCODEC&testcase=multiple-pipeline-playback-200

closed 2022-09-06 by mengdonglin

sathyap-chrome, 2022-10-25:
@mengdonglin if we are still seeing this issue, can we re open.
Also - we have another issue on ADL -N with ipc failures - not sure both are same though (linked above by @kv2019i