thesofproject/sof #6192: [BUG][IPC4] IPC4 firmware build on main branch is broken

keqiaozhang, 2022-08-26:
**Describe the bug**
The IPC4+Zephyr firmware build is broken on main branch.
This is a regression and caused by:
https://github.com/thesofproject/sof/commit/a0b832686cfdacb5b2d61f7002aa41b210d215db

** error **
```
06:14:05 [255/288] Building C object 'modules/sof/CMakeFiles/modules_sof.dir/srv/home/jenkins/workspace/sof_generic_build@2/sof/src/audio/copier/copier_hifi.c.obj'
06:14:05 FAILED: modules/sof/CMakeFiles/modules_sof.dir/srv/home/jenkins/workspace/sof_generic_build@2/sof/src/audio/copier/copier_hifi.c.obj 
06:14:05 ccache /srv/home/jenkins/xcc/install/tools/RG-2017.8-linux/XtensaTools/bin/xt-xcc -DCC_OPTIMIZE_FLAGS=\"-O2\" -DKERNEL -DXCC_TOOLS_VERSION=\"xcc\" -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -DRELATIVE_FILE=\"../sof/src/audio/copier/copier_hifi.c\" -I/srv/home/jenkins/workspace/sof_generic_build@2/zephyr/include -I/srv/home/jenkins/workspace/sof_generic_build@2/build-tgl/zephyr/include/generated -I/srv/home/jenkins/workspace/sof_generic_build@2/zephyr/soc/xtensa/intel_adsp/cavs -I/srv/home/jenkins/workspace/sof_generic_build@2/zephyr/soc/xtensa/intel_adsp/cavs/include -I/srv/home/jenkins/workspace/sof_generic_build@2/zephyr/soc/xtensa/intel_adsp/cavs/include/intel_tgl_adsp -I/srv/home/jenkins/workspace/sof_generic_build@2/zephyr/soc/xtensa/intel_adsp/common/include -I/srv/home/jenkins/workspace/sof_generic_build@2/sof/zephyr/include -I/srv/home/jenkins/workspace/sof_generic_build@2/sof/zephyr/../src/platform/intel/cavs/include -I/srv/home/jenkins/workspace/sof_generic_build@2/sof/zephyr/../src/platform/tigerlake/include -I/srv/home/jenkins/workspace/sof_generic_build@2/sof/zephyr/../rimage/src/include -I/srv/home/jenkins/workspace/sof_generic_build@2/sof/zephyr/../zephyr/include -I/srv/home/jenkins/workspace/sof_generic_build@2/sof/zephyr/../src/include -I/srv/home/jenkins/workspace/sof_generic_build@2/sof/zephyr/../src/arch/xtensa/include -isystem /srv/home/jenkins/workspace/sof_generic_build@2/zephyr/lib/libc/minimal/include -O2 -imacros /srv/home/jenkins/workspace/sof_generic_build@2/build-tgl/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -fms-extensions -imacros/srv/home/jenkins/workspace/sof_generic_build@2/zephyr/include/zephyr/toolchain/xcc_missing_defs.h -imacros /srv/home/jenkins/workspace/sof_generic_build@2/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -fgnu89-inline -Werror=implicit-int -fno-pic -fstrict-overflow -ffunction-sections -fdata-sections -mlongcalls -std=c99 -std=gnu99 -fno-inline-functions -MD -MT 'modules/sof/CMakeFiles/modules_sof.dir/srv/home/jenkins/workspace/sof_generic_build@2/sof/src/audio/copier/copier_hifi.c.obj' -MF modules/sof/CMakeFiles/modules_sof.dir/srv/home/jenkins/workspace/sof_generic_build@2/sof/src/audio/copier/copier_hifi.c.obj.d -o 'modules/sof/CMakeFiles/modules_sof.dir/srv/home/jenkins/workspace/sof_generic_build@2/sof/src/audio/copier/copier_hifi.c.obj' -c '/srv/home/jenkins/workspace/sof_generic_build@2/sof/src/audio/copier/copier_hifi.c'
06:14:05 /srv/home/jenkins/workspace/sof_generic_build@2/sof/src/audio/copier/copier_hifi.c:23: error: redefinition of ‘log_const_copier_hifi’
06:14:05 /srv/home/jenkins/workspace/sof_generic_build@2/sof/src/audio/copier/copier_hifi.c:10: error: previous definition of ‘log_const_copier_hifi’ was here
06:14:05 /srv/home/jenkins/workspace/sof_generic_build@2/sof/src/audio/copier/copier_hifi.c:23: error: redefinition of ‘__log_current_const_data’
06:14:05 /srv/home/jenkins/workspace/sof_generic_build@2/sof/src/audio/copier/copier_hifi.c:10: error: previous definition of ‘__log_current_const_data’ was here
06:14:05 /srv/home/jenkins/workspace/sof_generic_build@2/sof/src/audio/copier/copier_hifi.c:23: error: redefinition of ‘__log_level’
06:14:05 /srv/home/jenkins/workspace/sof_generic_build@2/sof/src/audio/copier/copier_hifi.c:10: error: previous definition of ‘__log_level’ was here
06:14:05 [256/288] Building C object zephyr/kernel/CMakeFiles/kernel.dir/queue.c.obj
06:14:05 [257/288] Building C object 'modules/sof/CMakeFiles/modules_sof.dir/srv/home/jenkins/workspace/sof_generic_build@2/sof/src/trace/dma-trace.c.obj'
06:14:05 [258/288] Building C object 'modules/sof/CMakeFiles/modules_sof.dir/srv/home/jenkins/workspace/sof_generic_build@2/sof/src/platform/intel/cavs/platform.c.obj'
06:14:05 [259/288] Building C object 'modules/sof/CMakeFiles/modules_sof.dir/srv/home/jenkins/workspace/sof_generic_build@2/sof/src/lib/dai.c.obj'
06:14:05 [260/288] Building C object 'modules/sof/CMakeFiles/modules_sof.dir/srv/home/jenkins/workspace/sof_generic_build@2/sof/src/ipc/ipc4/helper.c.obj'
06:14:05 [261/288] Building C object 'modules/sof/CMakeFiles/modules_sof.dir/srv/home/jenkins/workspace/sof_generic_build@2/sof/src/audio/base_fw.c.obj'
06:14:05 [262/288] Building C object 'modules/sof/CMakeFiles/modules_sof.dir/srv/home/jenkins/workspace/sof_generic_build@2/sof/src/audio/module_adapter/module_adapter.c.obj'
06:14:05 [263/288] Building C object 'modules/sof/CMakeFiles/modules_sof.dir/srv/home/jenkins/workspace/sof_generic_build@2/sof/src/audio/module_adapter/module/volume/volume.c.obj'
06:14:05 [264/288] Building C object 'modules/sof/CMakeFiles/modules_sof.dir/srv/home/jenkins/workspace/sof_generic_build@2/sof/src/ipc/ipc4/handler.c.obj'
06:14:05 ninja: build stopped: subcommand failed.
06:14:05 FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /srv/home/jenkins/workspace/sof_generic_build@2/build-tgl
06:14:05 Traceback (most recent call last):
06:14:05   File "/srv/home/jenkins/workspace/sof_generic_build@2/sof/scripts/xtensa-build-zephyr.py", line 614, in <module>
06:14:05     main()
06:14:05   File "/srv/home/jenkins/workspace/sof_generic_build@2/sof/scripts/xtensa-build-zephyr.py", line 610, in main
06:14:05     build_platforms()
06:14:05   File "/srv/home/jenkins/workspace/sof_generic_build@2/sof/scripts/xtensa-build-zephyr.py", line 511, in build_platforms
06:14:05     raise cpe
06:14:05   File "/srv/home/jenkins/workspace/sof_generic_build@2/sof/scripts/xtensa-build-zephyr.py", line 505, in build_platforms
06:14:05     execute_command(build_cmd, cwd=west_top)
06:14:05   File "/srv/home/jenkins/workspace/sof_generic_build@2/sof/scripts/xtensa-build-zephyr.py", line 265, in execute_command
06:14:05     return subprocess.run(*run_args, **run_kwargs)
06:14:05   File "/usr/lib/python3.8/subprocess.py", line 516, in run
06:14:05     raise CalledProcessError(retcode, process.args,
06:14:05 subprocess.CalledProcessError: Command '['west', 'build', '--build-dir', 'build-tgl', '--board', 'intel_adsp_cavs25', '/srv/home/jenkins/workspace/sof_generic_build@2/sof/app', '--', '-DOVERLAY_CONFIG=/srv/home/jenkins/workspace/sof_generic_build@2/sof/app/debug_overlay.conf;/srv/home/jenkins/workspace/sof_generic_build@2/sof/app/overlays/tgl/ipc4_overlay.conf']' returned non-zero exit status 1.
06:14:05 build tgl zph fw with options: --use-platform-subdir -i IPC4 --key-type-subdir dbgkey -k /srv/home/jenkins/xcc/meu_tgl/private.pem --fw-naming AVS
06:14:09 tgl build fail
```

juimonen, 2022-08-26:
fix here: https://github.com/thesofproject/sof/pull/6195

lgirdwood, 2022-08-26:
> fix here: #6195

Merged and closing.

closed 2022-08-26 by lgirdwood