thesofproject/sof #6193: [BUG] IPC error for 0x80010000 (CONFIG) when testing multiple pipelines on TGLU_RVP_NOCODEC

miRoox, 2022-08-26:
**Describe the bug**
CI observes an IPC error for 0x80010000 (CONFIG) when testing multiple pipelines on TGLU_RVP_NOCODEC 

**To Reproduce**
```
TPLG=/lib/firmware/intel/sof-tplg/sof-tgl-nocodec.tplg MODEL=TGLU_RVP_NOCODEC ~/sof-test/test-case/multiple-pipeline.sh -f p -c 20 -l 50
```

**Reproduction Rate**
Around 1/10

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/4e6ca75f7357
    * SOF: https://github.com/thesofproject/sof/commit/8d9c5485c403
2) Name of the topology file
    * Topology: sof-tgl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGLU_RVP_NOCODEC 

**Screenshots or console output**

**dmesg**

```
[ 5958.895651] kernel: snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30010000: GLB_TPLG_MSG: COMP_NEW
[ 5958.895891] kernel: snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 5958.906542] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x80010000 (msg/reply size: 216/12): -62
[ 5958.906545] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to set dai config for NoCodec-3
[ 5958.906553] kernel: snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30020000: GLB_TPLG_MSG: COMP_FREE
[ 5958.906796] kernel: snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x30110000: GLB_TPLG_MSG: PIPE_FREE
[ 5958.907574] kernel: snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[ 5958.907787] kernel: snd_sof:snd_sof_dsp_core_put: sof-audio-pci-intel-tgl 0000:00:1f.3: Core 1 powered down
[ 5958.907788] kernel: snd_sof:sof_widget_free: sof-audio-pci-intel-tgl 0000:00:1f.3: widget PIPELINE.20.DMIC0.IN freed
[ 5958.907790] kernel: snd_sof:sof_widget_free: sof-audio-pci-intel-tgl 0000:00:1f.3: widget DMIC0.IN freed
[ 5958.907793] kernel: snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x40070000: GLB_PM_MSG: CORE_ENABLE
[ 5958.907906] kernel: snd_sof:snd_sof_dsp_core_put: sof-audio-pci-intel-tgl 0000:00:1f.3: Core 1 powered down
[ 5958.907907] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to set up connected widgets
[ 5958.907911] kernel: snd_sof:sof_ipc3_pcm_hw_params: sof-audio-pci-intel-tgl 0000:00:1f.3: stream_tag 2
[ 5958.907928] kernel: snd_sof:ipc3_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 5958.908042] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60010000 (msg/reply size: 108/20): -19
[ 5958.908044] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: HW params ipc failed for stream 2
[ 5958.908045] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -19
[ 5958.908064] kernel:  DMIC: ASoC: __soc_pcm_hw_params() failed (-19)
[ 5958.908074] kernel:  DMIC: ASoC: dpcm_fe_dai_hw_params failed (-19)
[ 5958.908078] kernel: snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 10 dir 1
```

[console_log.txt](https://github.com/thesofproject/sof/files/9429502/console_log.txt)
[full_dmesg.txt](https://github.com/thesofproject/sof/files/9429501/full_dmesg.txt)

lyakh, 2022-08-30:
@miRoox can we get a CI reference - a result number or a link? Or at least a firmware log?

miRoox, 2022-08-30:
Sorry, I just edit the description with the result ID. The firmware log is attached:

[full_slogger.txt](https://github.com/thesofproject/sof/files/9451107/full_slogger.txt)

lgirdwood, 2022-08-30:
Looks like secondary core gets stuck
```
[   452797833.986597] (         557.916626) c0 dma-trace                 src/trace/trace.c:141  INFO Suppressed 1 similar messages:    Number of Blocks: total %d used %d free %d
[   452797849.507430] (          15.520833) c0 dma-trace                 src/trace/trace.c:141  INFO Suppressed 1 similar messages:  %d Bytes blocks ID:%d base 0x%x
[   452797869.976179] (          20.468750) c0 ipc                  src/ipc/ipc3/handler.c:1605 INFO ipc: new cmd 0x60010000
[   452797924.142844] (          54.166664) c0 ipc                  src/ipc/ipc3/handler.c:207  ERROR ipc: comp 64 not found
Skipped 0 bytes after the last statement.
[   452793892.320087] (       20307.500000) c0 idc                           src/idc/idc.c:113  ERROR idc_wait_in_blocking_mode() error: timeout, target_core 2
[   452793918.361753] (          26.041666) c0 idc                ......./intel/cavs/idc.c:144  ERROR idc_send_msg(), blocking msg 0x4000000 failed for core 2
[   452797924.142844] (          54.166664) c0 ipc                  src/ipc/ipc3/handler.c:207  ERROR ipc: comp 64 not found
```

lyakh, 2022-08-30:
@lgirdwood that seems to be a result of earlier problems, the first (unusual) firmware ERROR is this:
```
[   452645775.190556] (        1904.895752) c2 mixer        1.0      src/audio/component.c:66   INFO comp_set_state(), state already set to 5
[   452645873.367636] (          98.177078) c2 dai          1.2     src/audio/dai-legacy.c:976  WARN dai_copy(): Copy_bytes 0 + avail bytes 304 < period bytes 384, possible glitch
[   452645891.544718] (          18.177082) c2 dai          1.2     src/audio/dai-legacy.c:984  WARN dai_copy(): nothing to copy
[   452647894.252972] (        2002.708252) c2 dw-dma                 src/drivers/dw/dma.c:1128 ERROR dw_dma_get_data_size(): xrun detected
[   452647917.690471] (          23.437500) c2 dai          1.2     src/audio/dai-legacy.c:905  ERROR dai_report_xrun(): underrun due to no data available
[   452647939.565470] (          21.875000) c2 dai          1.2     src/audio/dai-legacy.c:787  ERROR comp_underrun(): dev->comp.id = 2, source->avail = 192, copy_bytes = 0
[   452647956.961303] (          17.395832) c2 pipe         1.3   ....../pipeline-stream.c:270  INFO pipe trigger cmd 6
[   452716011.177349] (       68054.218750) c0 ipc                  src/ipc/ipc3/handler.c:1605 INFO ipc: new cmd 0x40070000
[   452716038.208598] (          27.031248) c0 ipc                  src/ipc/ipc3/handler.c:716  INFO ipc: pm core mask 0xf -> enable
[   452716234.146090] (         195.937485) c1 edf-schedule       /schedule/edf_schedule.c:259  INFO edf_scheduler_init()
[   452716346.073169] (         111.927078) c1 idc                           src/idc/idc.c:351  INFO idc_init()
```

fredoh9, 2022-10-12:
For past +one month, I don't this error

closed 2022-10-12 by fredoh9