thesofproject/sof #6204: [BUG][IPC4] ipc timed out for 0x13050003|GLB_SET_PIPELINE_STATE when testing multiple-pipeline-all

keqiaozhang, 2022-08-29:
**Describe the bug**
We detected this issue in our CI, this issue happens on TGLU_RVP_NOCODEC_IPC4ZPH when testing multiple-pipeline-all.
From the dmesg, there's a IPC timed out and failed to pause pipeline 5:
```
[ 3947.711217] kernel: snd_sof:sof_widget_free: sof-audio-pci-intel-tgl 0000:00:1f.3: widget pipeline.14 freed
[ 3947.711219] kernel: snd_sof:sof_widget_free: sof-audio-pci-intel-tgl 0000:00:1f.3: widget copier.DMIC.14.1 freed
[ 3947.711226] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13050003|0x0: GLB_SET_PIPELINE_STATE
[ 3948.005457] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 2 state 3
[ 3948.009443] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 4 state 3
[ 3948.217830] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13050003|0x0
[ 3948.217850] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to pause pipeline 5
[ 3948.217878] kernel: snd_sof_intel_hda_common:hda_dsp_stream_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x1a0]=0x20340000 successful
[ 3948.217901] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[ 3948.217907] kernel:  Port2: ASoC: trigger FE cmd: 0 failed: -110
```

**To Reproduce**
TPLG=/lib/firmware/intel/avs-tplg/cavs-tgl-nocodec.tplg MODEL=TGLU_RVP_NOCODEC_IPC4ZPH ~/sof-test/test-case/multiple-pipeline.sh -f p -c 20 -l 50

**Reproduction Rate**
almost 100%

**TPLG**
[image]

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
Kernel Branch: topic/sof-dev
Kernel Commit: [4e6ca75f7357](https://github.com/thesofproject/linux/commit/4e6ca75f7357)

SOF Branch: main
SOF Commit: [130ea515be76](https://github.com/thesofproject/sof/commit/130ea515be76)

Zephyr Commit: 
[zephyr-v3.1.0-3042-g8e55e59c5917](https://github.com/zephyrproject-rtos/zephyr/commit/zephyr-v3.1.0-3042-g8e55e59c5917)
2) Name of the topology file
    * Topology: cavs-tgl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGLU_RVP_NOCODEC_IPC4ZPH

[dmesg.txt](https://github.com/thesofproject/sof/files/9441882/dmesg.txt)

keqiaozhang, 2022-08-31:
In today's inner test, The IPC ID is changed to `0x13060003`:
```
[ 1781.434986] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 6 state 3
[ 1781.435020] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13060003|0x0: GLB_SET_PIPELINE_STATE
[ 1781.941167] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x13060003|0x0
[ 1781.941176] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to pause pipeline 6
[ 1781.941180] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[ 1781.941186] kernel:  Port2: ASoC: trigger FE cmd: 1 failed: -110
```

https://sof-ci.sh.intel.com/#/result/planresultdetail/15002?model=TGLU_RVP_NOCODEC_IPC4ZPH&testcase=check-playback-all-formats
https://sof-ci.sh.intel.com/#/result/planresultdetail/15002?model=TGLU_RVP_NOCODEC_IPC4ZPH&testcase=check-playback-50rounds

kv2019i, 2022-09-15:
This is fixed now and not seen in the daily results for these platforms, closing.

closed 2022-09-15 by kv2019i