thesofproject/sof #6214: [BUG][IPC4] Failed to resume arecord stream after S3 on TGLU_UP_HDA_IPC4ZPH

keqiaozhang, 2022-08-30:
**Describe the bug**
This issue happens on TGLU_UP_HDA_IPC4ZPH when testing `check-suspend-resume-with-capture-5.sh`. We only observed this issue with capture, no such issue with playback. The console log shows that there's I/O error happened after resuming from S3, but there's no errors or warnings in dmesg. 

**Screenshots or console output**
```
rtcwake: assuming RTC uses UTC ...
rtcwake: wakeup from "mem" using /dev/rtc0 at Tue Aug 30 02:28:16 2022
Suspended. Trying resume. Failed. Restarting stream. Done.
arecord: pcm_read:2178: read error: Input/output error
```

**To Reproduce**
1. run arecord on pcm 0 first: arecord -Dhw:0,0 -r 48000 -c 2 -f S16_LE /dev/null &
2. do s3: sudo rtcwake -m mem -s 5
3. after the system resumed from S3, you will see the i/o error in console output.

**Reproduction Rate**
100%

**tplg**
[image]

**Environment**
**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
  Kernel Branch: topic/sof-dev
  Kernel Commit: [4e6ca75f7357](https://github.com/thesofproject/linux/commit/4e6ca75f7357)
  SOF Branch: main
  SOF Commit: [1044e004e4ec](https://github.com/thesofproject/sof/commit/1044e004e4ec)
  Zephyr Commit: 
[zephyr-v3.1.0-3042-g8e55e59c5917](https://github.com/zephyrproject-rtos/zephyr/commit/zephyr-v3.1.0-3042-g8e55e59c5917)
2) Name of the topology file
    * Topology: cavs-mixin-mixout-hda.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGLU_UP_HDA_IPC4ZPH

[dmesg.txt](https://github.com/thesofproject/sof/files/9449112/dmesg.txt)

lgirdwood, 2022-08-30:
@keqiaozhang able to bisect on SOF commit ?

kv2019i, 2022-09-15:
This seems like a duplicate of #5827  . This problem does still occur and visible in today's daily results on these platforms.

keqiaozhang, 2022-09-20:
Confirmed that this issue is not reproducible. Closing this bug.

closed 2022-09-20 by keqiaozhang