thesofproject/sof #6222: [BUG][IPC4] ipc error for msg 0x11010007|0x0 GLB_CREATE_PIPELINE

keqiaozhang, 2022-08-31:
**Describe the bug**
Observed this issue on TGLU_UP_HDA_IPC4ZPH when testing `check-playback-100times.sh`.  But hard to reproduce it again manually.
Inner test ID:http://sof-ci.sh.intel.com/#/result/planresultdetail/15002
error:
```
[ 1124.101110] kernel: snd_sof:sof_ipc4_widget_setup: sof-audio-pci-intel-tgl 0000:00:1f.3: Create widget pipeline.1 instance 0 - pipe 1 - core 0
[ 1124.101114] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x11010007|0x0: GLB_CREATE_PIPELINE
[ 1124.101232] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x31000009|0x0: GLB_CREATE_PIPELINE
[ 1124.101235] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 9 - Specified resource not found
[ 1124.101545] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0x11010007|0x0
[ 1124.101548] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to create module pipeline.1
[ 1124.101550] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to set up connected widgets
[ 1124.102336] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[ 1124.102862] kernel: snd_sof_intel_hda_common:hda_dsp_stream_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x14001e successful
```

**To Reproduce**
TPLG=/lib/firmware/intel/avs-tplg/cavs-mixin-mixout-hda.tplg MODEL=TGLU_UP_HDA_IPC4ZPH ~/sof-test/test-case/check-playback.sh -d 1 -l 100 -r 1

**Reproduction Rate**
hard to tell, only observed this issue once in CI

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
Kernel Branch: topic/sof-dev [e37ac35bc502](https://github.com/thesofproject/linux/commit/e37ac35bc502)
SOF Branch: main [42aec0a1867b](https://github.com/thesofproject/sof/commit/42aec0a1867b)
Zephyr Commit: [zephyr-v3.1.0-3042-g8e55e59c5917](https://github.com/zephyrproject-rtos/zephyr/commit/zephyr-v3.1.0-3042-g8e55e59c5917)
Name of the topology file
Topology: cavs-mixin-mixout-hda.tplg
Name of the platform(s) on which the bug is observed.
Platform: TGLU_UP_HDA_IPC4ZPH

[dmesg.txt](https://github.com/thesofproject/sof/files/9458890/dmesg.txt)

kv2019i, 2022-09-05:
I can get this even with all the recent IPC timeout fixes in place. FW log shows:
--cut--
[00:03:04.174,000] <err> ipc: ipc: current status 5
[00:03:04.174,000] <err> ipc: failed to process msg 19 status 7
[00:03:04.174,000] <err> ipc: ipc4: 0 failed err 7
[00:03:04.175,000] <inf> pipe: pipe trigger cmd 0
[00:03:04.175,000] <inf> pipe: pipe reset
[00:03:04.175,000] <inf> copier: copier_config(): Component is in active state. Ignore resetting
[00:03:04.175,000] <err> copier: failed to find dai comp or sink pipeline not running.
--cut--

kv2019i, 2022-09-15:
Let's track this "[ 1124.101235] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 9 - Specified resource not found"
case in #6213 . Closing this one.

closed 2022-09-15 by kv2019i