thesofproject/sof #6228: [BUG] Use of Q_MULTS(R)_32X32 macro without (int64_t) typecast

singalsu, 2022-08-31:
**Describe the bug**
The fractional multiply macro for 32 bit integers needs int64_t first argument or a typecast to ensure the product does not overflow in multiplication.

The susceptible  code lines found are 

nclude/sof/drivers/dmic.h:    time_ms = Q_MULTSR_32X32((int32_t)rate, LOGRAMP_TIME_COEF_Q15, 0, 15, 0) +
audio/tdfb/tdfb.c:    cd->max_frames = Q_MULTSR_32X32(dev->frames, TDFB_MAX_FRAMES_MULT_Q14, 0, 14, 0);
audio/tdfb/tdfb_direction.c:    cd->direction.max_lag = Q_MULTSR_32X32(fs, t_max, 0, 15, 0) + 1;
audio/tdfb/tdfb_direction.c:    thr = sat_int32(Q_MULTS_32X32(ambient >> 32, POWER_THRESHOLD, 31, 10, 31));
audio/tdfb/tdfb_direction.c:    src_x = Q_MULTSR_32X32((int32_t)cos_az, SOURCE_DISTANCE, 15, 12, 12);
audio/tdfb/tdfb_direction.c:    src_y = Q_MULTSR_32X32((int32_t)sin_az, SOURCE_DISTANCE, 15, 12, 12);
audio/tdfb/tdfb_direction.c:    az_slow = Q_MULTSR_32X32((int32_t)az, SLOW_AZ_C1, 12, 15, 12) +
audio/tdfb/tdfb_direction.c:        Q_MULTSR_32X32((int32_t)cd->direction.az_slow, SLOW_AZ_C2, 12, 15, 12);
audio/tdfb/tdfb_direction.c:    deg = Q_MULTS_32X32((int32_t)cd->direction.az_slow, COEF_RAD_TO_DEG, 12, 9, 0) -
audio/module_adapter/module/volume/volume.c:    cd->initial_ramp = Q_MULTSR_32X32(curve_duration,

The last line already has a fix PR. It looks based on last three arguments for macro (Q-format for product and output) like most of above should be implemented with a new Q_MULTS(R)_16X16 macro for 16 bits multiplication with (int32_t)type cast to avoid confusion.

**To Reproduce**
N/A

**Reproduction Rate**
N/A

**Expected behavior**
N/A

**Impact**
Risk for severe computation error if the int32_t * int32_t -> int32_t range is exceeded.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: N/A
    * SOF: 7b1906e7fabe0c24733dddbbd75ca4fc184f2318
2) Name of the topology file
    * Topology: N/A
3) Name of the platform(s) on which the bug is observed.
    * Platform: All

**Screenshots or console output**
N/A

Thanks to @ArsenEloglian for finding and reporting the misuse of multiplication macro!

singalsu, 2022-08-31:
There are two fix possibilities, 1) add typecast to macro arguments or 2) keep macro as-is and add/correct the type casts in susceptible code lines.

I'd prefer myself 2) to make most efficient code without type casts overhead when not needed. Plus all of above (except one?) will likely become Q_MULTS(R)_16X16 to avoid the confusion and show the real word lengths used.

lgirdwood, 2022-08-31:
> I'd prefer myself 2) to make most efficient code without type casts overhead when not needed. Plus all of above (except one?) will likely become Q_MULTS(R)_16X16 to avoid the confusion and show the real word lengths used.

Please do 2 and add 16x16 version.

singalsu, 2022-09-06:
All fixes are now, merged! Can close this issue now.

closed 2022-09-06 by singalsu