thesofproject/sof #6235: [BUG][IPC4] ipc timed out for 0x44000000|0x31400008 when system boot on TGLU_RVP_NOCODEC

miRoox, 2022-09-01:
**Describe the bug**
CI observe an ipc timed out just after firmware boot.
internal report ID: 15039

**To Reproduce**
`~/sof-test/test-case/verify-kernel-boot-log.sh`

**Reproduction Rate**
Only once in CI.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/e37ac35bc502
    * SOF: https://github.com/thesofproject/sof/commit/0209c464e6f5
2) Name of the topology file
    * Topology: cavs-tgl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGLU_RVP_NOCODEC

**Screenshots or console output**

**dmesg**
```
[    3.490554] kernel: snd_sof:snd_sof_run_firmware: sof-audio-pci-intel-tgl 0000:00:1f.3: firmware boot complete
[    3.490556] kernel: snd_sof:sof_set_fw_state: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state change: 5 -> 6
[    3.490600] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x43000000|0x20701000: MOD_LARGE_CONFIG_GET
[    3.490720] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x63000000|0x30700104: MOD_LARGE_CONFIG_GET
[    3.494934] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x43000000|0x20701000: MOD_LARGE_CONFIG_GET
[    3.494937] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Booted firmware version: 2.0.0.1
[    3.494944] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x44000000|0x31400008: MOD_LARGE_CONFIG_SET [data size: 8]
[    3.519684] kernel: [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    3.524187] kernel: ACPI: video: Video Device [GFX0] (multi-head: yes  rom: no  post: no)
[    3.525474] kernel: input: Video Bus as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:00/input/input5
[    3.539736] kernel: fbcon: i915drmfb (fb0) is primary device
[    3.573123] kernel: Asix Electronics AX88772A usb-003:004:10: attached PHY driver (mii_bus:phy_addr=usb-003:004:10, irq=POLL)
[    3.573400] kernel: asix 3-8.1:1.0 eth0: register 'asix' at usb-0000:00:14.0-8.1, ASIX AX88772 USB 2.0 Ethernet, 00:0e:c6:b7:2c:17
[    3.573444] kernel: usbcore: registered new interface driver asix
[    4.000200] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x44000000|0x31400008
[    4.000202] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: sof_ipc4_set_get_data: large config set failed at offset 0: -110
[    4.000206] kernel: snd_sof:ipc4_mtrace_init: sof-audio-pci-intel-tgl 0000:00:1f.3: initialization failed, fw tracing is disabled
[    4.000303] kernel: snd_sof:sof_machine_register: sof-audio-pci-intel-tgl 0000:00:1f.3: created machine sof-nocodec
[    4.000414] kernel: snd_sof_pci:sof_pci_probe_complete: sof-audio-pci-intel-tgl 0000:00:1f.3: Completing SOF PCI probe
[    4.695359] kernel: Console: switching to colour frame buffer device 480x135
[    4.719109] kernel: i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[    4.750013] kernel: snd_sof:snd_sof_load_topology: sof-audio-pci-intel-tgl 0000:00:1f.3: loading topology:intel/avs-tplg/sof-tgl-nocodec.tplg
[    4.750538] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:24:0 Kernel ABI 3:23:0
```

[full_dmesg.txt](https://github.com/thesofproject/sof/files/9467139/full_dmesg.txt)

kv2019i, 2022-09-05:
I'll assign this. Like to be fixed with https://github.com/thesofproject/sof/pull/6232 and https://github.com/thesofproject/linux/pull/3843

kv2019i, 2022-09-15:
This should be fixed now and not seen in latest daily on these platforms.

closed 2022-09-15 by kv2019i