thesofproject/sof #6237: [BUG][IPC4] error “Unknown error while processing the request“ happens when testing multiple pipeline playback on TGLU_RVP_SDW

miRoox, 2022-09-01:
**Describe the bug**
WIP

CI report ID: 15039

**To Reproduce**
```
TPLG=/lib/firmware/intel/avs-tplg/cavs-sdw.tplg MODEL=TGLU_RVP_SDW_IPC4ZPH ~/sof-test/test-case/multiple-pipeline.sh -f p -l 50
```

**Reproduction Rate**
Once in CI.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: cavs-sdw.tplg 
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**

**dmesg**
```
[ 1181.064093] kernel: snd_sof:sof_pcm_close: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: close stream 1 dir 1
[ 1181.066048] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 2 state 3
[ 1181.066052] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13020003|0x0: GLB_SET_PIPELINE_STATE
[ 1181.076664] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000006|0x0: GLB_SET_PIPELINE_STATE
[ 1181.076669] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 6 - Unknown error while processing the request
[ 1181.076720] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0x13020003|0x0
[ 1181.077924] kernel: snd_soc_rt711:rt711_sdw_write: rt711 sdw:0:025d:0711:00: [rt711_sdw_write] 00e0 <= 0003
[ 1181.078045] kernel: snd_soc_rt711:rt711_sdw_write: rt711 sdw:0:025d:0711:00: [rt711_sdw_write] 00f0 <= 0003
[ 1181.078048] kernel: snd_soc_rt711:rt711_clock_config: rt711 sdw:0:025d:0711:00: rt711_clock_config complete, clk_freq=4800000
```

kv2019i, 2022-09-05:
This issue (and similar) where there is no IPC timeout, are good to track separately.

kv2019i, 2022-09-15:
Let's track these "error 6" on SET_PIPELINE_STATE errors in https://github.com/thesofproject/sof/issues/6220 . Closing this one,

closed 2022-09-15 by kv2019i