thesofproject/sof #6249: Tplg / Alsa widgets/controls on SKL

EMLommers, 2022-09-04:
Currently only sound through speakers is possible.
In my opinion this has to do with asla widget/controls  media* and codec*, which are available in the closed source DSP AUDIO, which is related to the Pin number of the audio chip. 
Could you add those controls in order to change headphones to speaker and vice versa?

lgirdwood, 2022-09-06:
@crojewsk may be able to advise here, you could be missing a UCM config.

EMLommers, 2022-09-10:
Like to control Speaker and Headphone...

Disable all:
media0_out mo codec0_in mi Switch' off
media0_out mo dmic01_hifi_in mi Switch' off
codec1_out mo media0_in mi Switch' off
codec0_out mo media0_in mi Switch' off

Enable Speaker:   codec0_out mo media0_in mi Switch' on
Enable Headset: codec1_out mo media0_in mi Switch' on

Disable Headphone: codec1_out mo media0_in mi Switch' off
Disable Speaker: codec0_out mo media0_in mi Switch' off

lgirdwood, 2022-09-12:
@EMLommers can you provide the dmesg for audio, the Intel AVS driver report no errors.

plbossart, 2022-10-04:
@EMLommers these mixers are not created by the SOF driver/topology, but the Skylake driver/topology. Different solution, different support channel.

closed 2022-10-06 by EMLommers