thesofproject/sof #6256: [BUG] update ABI version in fw_reg in memory windows

RanderWang, 2022-09-05:
The new fw_reg definition is introduced.  abi_ver should be set to FW_REGS_ABI_VER in memory windows, but it is not done now. It is checked on kernel side and result a error.

````C
struct ipc4_fw_registers {
	/* Current ROM / FW status(at 0x0) */
	struct ipc4_fw_status_reg fsr;

	/* Last ROM / FW error code(at 0x4). */
	ipc4_last_error lec;

	/* Current DSP clock status(at 0x8). */
	struct ipc4_fw_pwr_status fps;

	/* Last Native Error Code(from external library) (at 0xC). */
	uint32_t lnec;

	/* Copy of LTRC HW register value(FW only) (at 0x10). */
	uint32_t ltr;

	uint32_t rsvd0;

	/* ROM info(at 0x18). */
	union ipc4_rom_info rom_info;

	**/* Version of the layout, set to the current FW_REGS_ABI_VER. */**
	uint32_t abi_ver;
````

RanderWang, 2022-09-05:
@mmaka1   @serhiy-katsyuba-intel  could you help to update abi_ver in fw_reg in memory windows ? thanks

mwasko, 2022-09-06:
@RanderWang I assume that this error is not blocking any functional testing on kernel side, so I mark this as P3. If it require prioritization then please let me know.

RanderWang, 2022-09-06:
it blocks deep buffer feature on MTL. This feature works with close source fw.

lgirdwood, 2022-09-06:
> it blocks deep buffer feature on MTL. This feature works with close source fw.

@RanderWang ok, are we using this version to determine programming flows for features ? and does the upstream kernel use this value now ?

RanderWang, 2022-09-07:
> > it blocks deep buffer feature on MTL. This feature works with close source fw.
> 
> @RanderWang ok, are we using this version to determine programming flows for features ? and does the upstream kernel use this value now ?

sure. Since FW uses updated FW_reg version, we need to check it to avoid panic with old version.  I have a PR for it and it is not merged.

dnikodem, 2022-09-07:
@RanderWang do I understand correctly that you use sof repo to build MTL?

RanderWang, 2022-09-07:
I tested ADL since it also uses the fw_reg.

RanderWang, 2022-10-18:
fixed

closed 2022-10-18 by RanderWang