thesofproject/sof #6266: [BUG][IPC4] failed to prepare widget gain.5.1 when playback with S32_LE format on HDA/Soundwire mode

miRoox, 2022-09-07:
**Describe the bug**
When playback with S32_LE format audio on Deep Buffer, it will failed to install hw params
```
aplay: set_params:1416: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: 32
FRAME_BITS: 64
CHANNELS: 2
RATE: 48000
PERIOD_TIME: (42666 42667)
PERIOD_SIZE: 2048
PERIOD_BYTES: 16384
PERIODS: 4
BUFFER_TIME: (170666 170667)
BUFFER_SIZE: 8192
BUFFER_BYTES: 65536
TICK_TIME: 0
```
and the dmesg shows it's "failed to prepare widget gain.5.1"

CI report ID: 15173

**To Reproduce**
For HDA mode
```
aplay   -Dhw:0,1 -r 48000 -c 2 -f S32_LE -d 3 /dev/zero -v -q
```

For Soundwire mode
```
aplay   -Dhw:0,2 -r 48000 -c 2 -f S32_LE -d 3 /dev/zero -v -q
```

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commit/91a177700173
    * FIrmware: IPC4 tester
2) Name of the topology file
    * Topology:
cavs-mixin-mixout-hda.tplg
[image]
cavs-sdw.tplg 
[image]
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGLU_RVP_HDA/ADLP_RVP_SDW

**Screenshots or console output**

**dmesg**
```
[  159.606496] kernel: snd_sof:sof_ipc4_init_audio_fmt: sof-audio-pci-intel-tgl 0000:00:1f.3: matching audio format index for 48000Hz, 32bit, 2 channels: 2
[  159.606499] kernel: snd_sof:sof_ipc4_init_audio_fmt: sof-audio-pci-intel-tgl 0000:00:1f.3: Init input audio formats for copier.host.5.1
[  159.606501] kernel: snd_sof:sof_ipc4_dbg_audio_format: sof-audio-pci-intel-tgl 0000:00:1f.3:  #0: 48000KHz, 32bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x2002)
[  159.606505] kernel: snd_sof:sof_ipc4_init_audio_fmt: sof-audio-pci-intel-tgl 0000:00:1f.3: Init output audio formats for copier.host.5.1
[  159.606507] kernel: snd_sof:sof_ipc4_dbg_audio_format: sof-audio-pci-intel-tgl 0000:00:1f.3:  #0: 48000KHz, 32bit (ch_map 0xffffff10 ch_cfg 1 interleaving_style 0 fmt_cfg 0x2002)
[  159.606509] kernel: snd_sof:sof_ipc4_prepare_copier_module: sof-audio-pci-intel-tgl 0000:00:1f.3: copier copier.host.5.1, IPC size is 84
[  159.606512] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: sof_ipc4_init_audio_fmt: Unsupported audio format: 48000Hz, 32bit, 2 channels
[  159.606514] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to prepare widget gain.5.1
[  159.606515] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to prepare connected widgets
[  159.606517] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed widget list set up for pcm 1 dir 0
[  159.606519] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -22
[  159.606522] kernel:  DeepBuffer: ASoC: __soc_pcm_hw_params() failed (-22)
[  159.606549] kernel:  DeepBuffer: ASoC: dpcm_fe_dai_hw_params failed (-22)
[  159.606552] kernel: snd_sof:sof_pcm_hw_free: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: free stream 1 dir 0
[  159.607566] kernel: snd_sof:sof_pcm_close: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: close stream 1 dir 0
```

[full_dmesg.txt](https://github.com/thesofproject/linux/files/9502032/full_dmesg.txt)

lgirdwood, 2022-09-07:
@serhiy-katsyuba-intel fyi

marc-hb, 2022-09-07:
Stupid github auto-close, let's wait for tests to pass

miRoox, 2022-09-08:
Confirmed with tests