thesofproject/sof #6282: [BUG] failed to get stream_start_offset

RanderWang, 2022-09-09:
Stream_start_offset is update by host copier->copier_comp_trigger. The problem is if dai pipeline is not ready, the stream_start_offset will be set to zero. And no chance to update it to correct value,  then kernel driver will always get zero of stream_start_offset

````c
static int copier_comp_trigger(struct comp_dev *dev, int cmd)
{
	/* dai is in another pipeline and it is not prepared or active */
	if (dai_copier->state <= COMP_STATE_READY ||
	    dai_cd->endpoint[IPC4_COPIER_GATEWAY_PIN]->state <= COMP_STATE_READY) {
		struct ipc4_pipeline_registers pipe_reg;

		comp_warn(dev, "dai is not ready");

		pipe_reg.stream_start_offset = 0;
		pipe_reg.stream_end_offset = 0;
		mailbox_sw_regs_write(cd->pipeline_reg_offset, &pipe_reg, sizeof(pipe_reg));

		return 0;
	}

      // updated stream_start_offset
}
````

RanderWang, 2022-09-09:
@sathya-nujella  could you help to address this issue ?

RanderWang, 2022-09-09:
sorry add @mwasko

lgirdwood, 2022-09-09:
Adding @marcinszkudlinski too.

plbossart, 2022-09-12:
How is this possible @RanderWang to do a trigger on a path that is not fully prepared? In theory we should prepare all the paths and only then do a trigger.

@ranj063 FYI.

RanderWang, 2022-09-13:
> 

@plbossart  we have two or more pipelines for a single pcm stream. And we trigger start one by one, from host pipeline to dai pipeline

plbossart, 2022-09-13:
I understand the configuration, I just don't understand how the dai pipeline could be not ready when either of the two PCM host pipelines is triggered. That's not possible with the ALSA programming model. Please help me understand the flow and sequence that leads to the dai pipeline not being ready.

RanderWang, 2022-09-14:
> I understand the configuration, I just don't understand how the dai pipeline could be not ready when either of the two PCM host pipelines is triggered. That's not possible with the ALSA programming [...]

Please check the pasted code. I meant dai is not ready to feed back hw position and dai is not active at query time.

````c
static int copier_comp_trigger(struct comp_dev *dev, int cmd)
{
	/* dai is in another pipeline and it is not prepared or active */
	if (dai_copier->state <= COMP_STATE_READY ||
	    dai_cd->endpoint[IPC4_COPIER_GATEWAY_PIN]->state <= COMP_STATE_READY) {
		struct ipc4_pipeline_registers pipe_reg;

		comp_warn(dev, "dai is not ready");

		pipe_reg.stream_start_offset = 0;
		pipe_reg.stream_end_offset = 0;
		mailbox_sw_regs_write(cd->pipeline_reg_offset, &pipe_reg, sizeof(pipe_reg));

		return 0;
	}

      // updated stream_start_offset
}
````

plbossart, 2022-09-14:
Sorry @RanderWang you are just stating facts but there's no explanation as to why the DAI is not ready. This is not a normal programming sequence, there is no reason why a different pipeline connected to the PCM has not been prepared by the time we do a trigger. 

I wonder is there is something in the kernel where the BE is not prepared somehow.

@ranj063 please chime in.

RanderWang, 2022-09-14:
> 

@plbossart  DAI is ready to feed back data when it is started. Now HOST is started before DAI, so DAI is not ready when HOST is started

RanderWang, 2022-09-14:
@plbossart  DAI can't get validate hw info when DMA is not started by DAI start, at this time DAI is ready but not active. DAI is ready to get hw info when DAI is started and in active status.  If DAI is only in ready status,  HOST can't get HW info from DAI.

RanderWang, 2022-09-14:
in FW , it should be
````c
dai_copier->state == COMP_STATE_ACTIVE
````

plbossart, 2022-09-14:
@RanderWang you're missing the point. The ALSA framework will have two separate stages for prepare and trigger. The DAI pipeline has to be READY when we trigger the host. It makes no sense that the DAI is not ready when the host is triggered, that's an invalid sequence.

RanderWang, 2022-09-14:
> @RanderWang you're missing the point. The ALSA framework will have two separate stages for prepare and trigger. The DAI pipeline has to be READY when we trigger the host. It makes no sense that the DAI is not ready when the host is triggered, that's an invalid sequence.

@plbossart I agree with you. My idea is that DAI pipeline is in ready state when we trigger the host, but DAI pipeline is not triggered start. At this time HOST can't get DAI position since DAI is not active and can't get valid hw dma info. Here we need DAI active, not just ready, so the check is
````c
if (dai_copier->state <= COMP_STATE_READY ||
	    dai_cd->endpoint[IPC4_COPIER_GATEWAY_PIN]->state <= COMP_STATE_READY) {
		struct ipc4_pipeline_registers pipe_reg;

		comp_warn(dev, "dai is not ready");
````

plbossart, 2022-09-14:
I still don't understand the issue @RanderWang. If the stream has not started, the link counters should report zero. that's not a bug that's a feature.

RanderWang, 2022-09-14:
> I still don't understand the issue @RanderWang. If the stream has not started, the link counters should report zero. that's not a bug that's a feature.

sure, then stream_start_offset should be zero. if so I can close this bug.

closed 2022-09-15 by RanderWang