thesofproject/sof #6285: [FEATURE] Generic cache management for`module_set_large_config()`

lyakh, 2022-09-12:
**Is your feature request related to a problem? Please describe.**
A problem with cache management has recently been fixed by #6274 for the volume module. However, that fix has to be made generic, not volume-specific.

**Describe the solution you'd like**
Cache invalidation for module configuration has to be dome in module core, not in individual module drivers.

**Additional context**
See the aforementioned PR and particularly comment https://github.com/thesofproject/sof/pull/6274#issuecomment-1241610167 in it.

lgirdwood, 2022-10-18:
May need locking too.

kv2019i, 2023-02-17:
@lyakh  Is this still open? If yes, ok to queue for v2.5?

kv2019i, 2023-03-06:
@lyakh Still on track for V2.5 -- we aim to branch this week?

kv2019i, 2023-03-09:
No response, moving to v2.6.

kv2019i, 2023-11-22:
@lyakh This has been pushed onwards since v2.5, let's try to cover this in v2.9 planning, so we have actual time to do this.

PR #8534 (2023-12-01): cache handling in volume - address #6285
clarify IPC-version specific processing paths and remove a redundant cache invalidation. This should resolve #6285