thesofproject/sof #6289: [BUG][JSL] Chuwi Hi10 Go - No sound output/input

ethylmorphine, 2022-09-12:
**Describe the bug**
> Debian 11
> Linux 5.19.0
> essx8336 codec

Kernel loads firmware, and there are output/input devices listed in pavucontrol, but the is no sound (even no hissing) from speakers no matter what volume is set.

**To Reproduce**
Install Debian on JSL device, run speaker-test to no result.

**Reproduction Rate**
I've heard nothing at all from the speakers since the day I installed Linux.

**Expected behavior**
Well, to get a working audio.

**Impact**
Youtube is kinda meh without sound.

**Environment**
> sof-bin-2.2
> sof-tplg-2.2.1
> JSL platform

**Screenshots or console output**
Firmware is loaded properly, and the topology seems to be corresponding to what I see in UEFI (disabled DMICs, and SSP0 enabled).
```
[   17.396602] sof-audio-pci-intel-icl 0000:00:1f.3: enabling device (0000 -> 0002)
[   17.396903] sof-audio-pci-intel-icl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   17.397033] sof-audio-pci-intel-icl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   17.403841] sof-audio-pci-intel-icl 0000:00:1f.3: use msi interrupt mode
[   17.481316] sof-audio-pci-intel-icl 0000:00:1f.3: NHLT_DEVICE_I2S detected, ssp_mask 0x1
[   17.481325] sof-audio-pci-intel-icl 0000:00:1f.3: hda codecs found, mask 4
[   17.518805] sof-audio-pci-intel-icl 0000:00:1f.3: firmware: direct-loading firmware intel/sof/intel-signed/sof-jsl.ri
[   17.518899] sof-audio-pci-intel-icl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[   17.518901] sof-audio-pci-intel-icl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:21:0
[   17.518904] sof-audio-pci-intel-icl 0000:00:1f.3: FW ABI is more recent than kernel
[   17.518913] sof-audio-pci-intel-icl 0000:00:1f.3: unknown sof_ext_man header type 3 size 0x30
[   17.621120] sof-audio-pci-intel-icl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[   17.621126] sof-audio-pci-intel-icl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:21:0
[   17.621128] sof-audio-pci-intel-icl 0000:00:1f.3: FW ABI is more recent than kernel
[   17.640533] sof-essx8336 sof-essx8336: quirk mask 0x0
[   17.640539] sof-essx8336 sof-essx8336: quirk SSP0
[   17.651492] sof-audio-pci-intel-icl 0000:00:1f.3: firmware: direct-loading firmware intel/sof-tplg/sof-jsl-es8336-ssp0.tplg
[   17.651515] sof-audio-pci-intel-icl 0000:00:1f.3: Topology: ABI 3:22:1 Kernel ABI 3:21:0
[   17.651519] sof-audio-pci-intel-icl 0000:00:1f.3: warn: topology ABI is more recent than kernel
[   17.651764] sof-essx8336 sof-essx8336: ASoC: Parent card not yet available, widget card binding deferred
[   17.670994] zram: Added device: zram0
[   17.673113] es8316 i2c-ESSX8336:00: assuming static mclk
[   17.744125] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   17.744132] Bluetooth: BNEP filters: protocol multicast
[   17.744138] Bluetooth: BNEP socket layer initialized
[   17.747206] sof-essx8336 sof-essx8336: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 3
[   17.747210] sof-essx8336 sof-essx8336: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 4
[   17.763649] zram0: detected capacity change from 0 to 524288
[   17.775370] input: sof-essx8336 Headset as /devices/pci0000:00/0000:00:1f.3/sof-essx8336/sound/card0/input30
[   17.776440] input: sof-essx8336 HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/sof-essx8336/sound/card0/input31
[   17.776575] input: sof-essx8336 HDMI/DP,pcm=6 as /devices/pci0000:00/0000:00:1f.3/sof-essx8336/sound/card0/input32
[   17.776799] input: sof-essx8336 HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/sof-essx8336/sound/card0/input33
```

aplay shows that both analog and digital outputs are there.
```
☁  ~  aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofessx8336 [sof-essx8336], device 0: ES8336 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofessx8336 [sof-essx8336], device 5: HDMI 1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofessx8336 [sof-essx8336], device 6: HDMI 2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofessx8336 [sof-essx8336], device 7: HDMI 3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

speaker-test detects the card as well but during the test there is absolutely no sound. Generally, when speakers get powered on, they hiss, and in this case, they are completely silent. It's not a hardware failure though, because sound worked on Windows.
```
☁  ~  speaker-test -c 2                             

speaker-test 1.2.4

Playback device is default
Stream parameters are 48000Hz, S16_LE, 2 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 96 to 1048576
Period size range from 32 to 349526
Using max buffer size 1048576
Periods = 4
was set period_size = 262144
was set buffer_size = 1048576
 0 - Front Left
 1 - Front Right
Time per period = 10.956371
 0 - Front Left
^C 1 - Front Right
Time per period = 5.472374
```

Also, I've tried topologies and fw binaries from 1.7 up to the latest one with the same result: either I get snd_hda_generic module which is unable to detect analog audio (i.e. HDMI only) or I get no sound. Go figure.

plbossart, 2022-09-12:
https://github.com/thesofproject/linux/wiki/ES8336-support

ethylmorphine, 2022-09-14:
Okay, that was quite a quest. I have compiled a kernel from the tag and finally got the real reason why there is no sound.
`sof-audio-pci-intel-icl 0000:00:1f.3: Overriding topology with MCLK mask 0x1 from NHLT`

> In rare cases, some platforms use the MCLK1 signal instead of MCLK0. As of July 2022, there is no turn-key solution for those platforms.

Rather interesting is that I hear hissing from the left speaker when I switch from SSP0 to HDA link in UEFI. At least one of them gets powered on, yay.

plbossart, 2022-09-14:
> Okay, that was quite a quest. I have compiled a kernel from the tag and finally got the real reason why there is no sound. `sof-audio-pci-intel-icl 0000:00:1f.3: Overriding topology with MCLK mask [...]

That statement  in the documentation is obsolete, the selection of MCLK is now supported. But in this case the default was already MCLK0, mask 0x1 means MCLK0...

> 
> Rather interesting is that I hear hissing from the left speaker when I switch from SSP0 to HDA link in UEFI. At least one of them gets powered on, yay.

I don't understand what you are referring to, this device can only support SSP, not HDaudio.

ethylmorphine, 2022-09-14:
My bad. I meant that the only thing I was able to get from speakers is hissing when I switched SSP0 to HDA link in UEFI. I do realize that HD audio is not supported by the platform.

At this point, I'm out of ides as to why there is no sound. `alsaucm reload` does not report any issues with configuration as well.