thesofproject/sof #6316: [FEATURE] add get_total_data_processed in mixin &mixout for new pipeline latency algorithm

RanderWang, 2022-09-20:
In patch https://github.com/thesofproject/sof/pull/5985 update the algorithm of pipeline latency calculation. It is based on the number of buffered data blocks between the first and last component. It add ops get_total_data_processed in copier module, but didn't add it in mixin & mixout module.  The algorithm can't work for multiple-pipeline stream with mixin & mixout.

RanderWang, 2022-09-20:
@softwarecki

softwarecki, 2022-09-20:
@RanderWang yeah, that's right! I didn't do it right away because another colleague was expanding a functionality od the mixin/mixout at that time. At the moment I'm on the turbo critical path related to the upstream mtl platform so adding that functionality got a lower priority. I have this functionality on my list and will add it in the future.