thesofproject/sof #6321: [BUG][TPLG] topology2 fails to build

keqiaozhang, 2022-09-22:
**Describe the bug**
This issue is caused by the change in `tools/topology/topology2/CMakeLists.txt`. 

Commit ID: https://github.com/thesofproject/sof/commit/784bce763c0e2951746b62ade1e521c1f1ebd3ed.
[image]

It seems that `CONFIG_CAVS` and `CONFIG_ACE` flags are not, then cavs/ace subdirectories can not be created.

**To Reproduce**
`./scripts/build-tools.sh -T`

mengdonglin, 2022-09-22:
@softwarecki @mwasko Can you help on this issue?

@plbossart @ranj063 FYI. Topology binaries for ACE and cAVS platforms are going to be splitted.

marc-hb, 2022-09-23:
Fix: `git show 784bce763c0e -- tools/ | git apply -R`

ujfalusi, 2022-09-23:
Submitted a fix: https://github.com/thesofproject/sof/pull/6325

ujfalusi, 2022-09-23:
> @softwarecki @mwasko Can you help on this issue?
> 
> @plbossart @ranj063 FYI. Topology binaries for ACE and cAVS platforms are going to be splitted.

It does not mean that the splitting _must_ break existing builds.
Looks to me that 784bce763c0e2951746b62ade1e521c1f1ebd3ed has not even compile tested (for topologies at least) and I do wonder why CI let that patch slip through. I know that we only test IPC3/topology1 in CI, but the build system should have complained at least to block it.

It is one thing to have separate topology2 files for CAVS/ACE, but I don't really see the reason why they have to be exclusive.

keqiaozhang, 2022-09-25:
Fixed by [PR#6325](https://github.com/thesofproject/sof/pull/6325)

closed 2022-09-25 by keqiaozhang