thesofproject/sof #6330: [BUG] [IPC4] IPC timed out for ssp loopback in multi stream capture

bardliao, 2022-09-25:
**Describe the bug**

IPC timed out when I tested https://github.com/thesofproject/sof/pull/6329 with SSP loopback mode.

**To Reproduce**
sudo ./sof/tools/mtrace/mtrace-reader.py
aplay -Dhw:0,0 -f s32_le -r 48000 -c 2 -i -vvv ~/California_Dreaming_48k_16bit.wav
arecord -Dhw:0,0 -f s32_le -c 2 -r 48000 /tmp/port0.wav -vvv
arecord -Dhw:0,12 -f s32_le -c 2 -r 48000 /tmp/port12.wav -vvv
stop aplay
stop arecord

**Reproduction Rate**
100%

**Expected behavior**
No error

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: daily test kernel version
    * SOF: daily test zephyr IPC4 version 
2) Name of the topology file
    * Topology: sof-tgl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGLU

**Screenshots or console output**
```
[  203.246522] snd_sof:sof_ipc4_route_free: sof-audio-pci-intel-tgl 0000:00:1f.3: unbind modules copier.SSP.8.1:0 -> gain.7.1:0
[  203.246528] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x46020004|0x20006: MOD_UNBIND
[  203.246694] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x66000000|0x20006: MOD_UNBIND
[  203.246738] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x46020004|0x20006: MOD_UNBIND
[  203.246747] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x12080000|0x0: GLB_DELETE_PIPELINE
[  203.753044] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x12080000|0x0
[  203.753080] sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[  203.753085] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  203.753121] sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[  203.753131] sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[  203.753165] sof-audio-pci-intel-tgl 0000:00:1f.3: Host IPC initiator: 0x92080000|0x0|0x0, target: 0x0|0x0|0x0, ctl: 0x3
[  203.753173] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  203.753178] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  203.753182] sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[  203.753187] sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[  203.753211] sof-audio-pci-intel-tgl 0000:00:1f.3: 0x00000005: module: ROM, state: FW_ENTERED, running
[  203.753273] sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0x5 0x0 0x4000 0x0 0x0 0x0 0x0 0x4000
[  203.753278] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  203.753284] sof-audio-pci-intel-tgl 0000:00:1f.3: failed to free pipeline widget pipeline.8
[  203.753294] snd_sof:sof_ipc4_route_free: sof-audio-pci-intel-tgl 0000:00:1f.3: unbind modules gain.7.1:0 -> copier.host.7.1:0
[  203.753302] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x46020006|0x30004: MOD_UNBIND
[  204.256974] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x46020006|0x30004
[  204.256989] sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[  204.256994] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  204.257035] sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[  204.257041] sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[  204.257088] sof-audio-pci-intel-tgl 0000:00:1f.3: Host IPC initiator: 0x92080000|0x0|0x0, target: 0x0|0x0|0x0, ctl: 0x3
[  204.257095] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  204.257099] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  204.257103] sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[  204.257108] sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[  204.257135] sof-audio-pci-intel-tgl 0000:00:1f.3: 0x00000005: module: ROM, state: FW_ENTERED, running
[  204.257171] sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0x5 0x0 0x4000 0x0 0x0 0x0 0x0 0x4000
[  204.257176] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  204.257182] sof-audio-pci-intel-tgl 0000:00:1f.3: failed to unbind modules gain.7.1 -> copier.host.7.1
[  204.257194] snd_sof:sof_widget_free: sof-audio-pci-intel-tgl 0000:00:1f.3: widget gain.7.1 freed
[  204.257203] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x12070000|0x0: GLB_DELETE_PIPELINE
[  204.760982] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc timed out for 0x12070000|0x0
[  204.760998] sof-audio-pci-intel-tgl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[  204.761003] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  204.761043] sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[  204.761050] sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
[  204.761097] sof-audio-pci-intel-tgl 0000:00:1f.3: Host IPC initiator: 0x92080000|0x0|0x0, target: 0x0|0x0|0x0, ctl: 0x3
[  204.761105] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  204.761109] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  204.761114] sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
[  204.761119] sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[  204.761146] sof-audio-pci-intel-tgl 0000:00:1f.3: 0x00000005: module: ROM, state: FW_ENTERED, running
[  204.761183] sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0x5 0x0 0x4000 0x0 0x0 0x0 0x0 0x4000
[  204.761187] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  204.761194] sof-audio-pci-intel-tgl 0000:00:1f.3: failed to free pipeline widget pipeline.7
[  204.761203] sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to free connected widgets
[  204.761216] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_free on 0000:00:1f.3: -110
```
[dmesg_scenario1.txt](https://github.com/thesofproject/sof/files/9640186/dmesg_scenario1.txt)

```
<err> copier: failed to find dai comp or sink pipeline not running.
```

[mtrace_scenario1.txt](https://github.com/thesofproject/sof/files/9640187/mtrace_scenario1.txt)

bardliao, 2022-10-31:
This will be fixed by https://github.com/thesofproject/linux/pull/3957

keqiaozhang, 2022-11-14:
This issue has been fixed by https://github.com/thesofproject/linux/pull/3957

closed 2022-11-14 by keqiaozhang