thesofproject/sof #6346: [FEATURE] Chromebook SoundWire topologies need to be re-aligned with others

plbossart, 2022-09-28:
**Is your feature request related to a problem? 
The Chromebook topology1 files are not aligned with the others, specifically they do not rely on the mixer pipelines and cannot support additional streams in preparation of the 'deep buffer' support. This will prevent testing of this capability on N-1 platforms for CRAS evolutions.

**Describe the solution you'd like**
Re-align sof-tgl-sdw-rt5682-max98373, and addition of deep buffer device for headset.

@naveen-manohar @sathya-nujella This needs to be done by the Chrome team pretty please.

lgirdwood, 2022-09-28:
@RDharageswari fyi.

RDharageswari, 2022-10-04:
@plbossart : Sure. We will take a look at it