thesofproject/sof #6351: [BUG][MTL] ipc timed out for 0x13010004 when playing

keqiaozhang, 2022-09-29:
**Describe the bug**
This is the first time we check SOF MTL-IPC4 firmware on MTL-RVP HDA device.  Firmware and tplg can be loaded successfully.
But when doing playback on PCM 0, it will fail with I/O error and some IPC error in kernel logs:
```
[22919.570262] snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc4 set pipeline 0 state 3
[22919.570266] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13000003|0x0: GLB_SET_PIPELINE_STATE
[22919.570557] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x33000000|0x0: GLB_SET_PIPELINE_STATE
[22919.570616] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x13000003|0x0: GLB_SET_PIPELINE_STATE
[22919.570621] snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc4 set pipeline 0 state 4
[22919.570625] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13000004|0x0: GLB_SET_PIPELINE_STATE
[22919.571346] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x33000000|0x0: GLB_SET_PIPELINE_STATE
[22919.571368] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x13000004|0x0: GLB_SET_PIPELINE_STATE
[22919.571373] snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc4 set pipeline 1 state 3
[22919.571377] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13010003|0x0: GLB_SET_PIPELINE_STATE
[22919.571488] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx reply: 0x33000000|0x0: GLB_SET_PIPELINE_STATE
[22919.571505] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx done : 0x13010003|0x0: GLB_SET_PIPELINE_STATE
[22919.571509] snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc4 set pipeline 1 state 4
[22919.571513] snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-mtl 0000:00:1f.3: ipc tx      : 0x13010004|0x0: GLB_SET_PIPELINE_STATE
[22920.079421] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x13010004|0x0
[22920.079516] sof-audio-pci-intel-mtl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[22920.079528] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[22920.079598] sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0x13010004|0x0|0x0, target: 0x1b060000|0x0|0x0, ctl: 0x3
[22920.079740] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[22920.079798] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[22920.079853] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[22920.079896] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[22920.079968] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x5, ROM error: 0x0
[22920.080021] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[22920.080084] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[22920.080130] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[22920.080190] sof-audio-pci-intel-mtl 0000:00:1f.3: failed to set state 4 for pipeline 2
[22920.080249] sof-audio-pci-intel-mtl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
[22920.080323]  HDA Analog: ASoC: trigger FE cmd: 1 failed: -110
[22920.080735] snd_sof:sof_pcm_trigger: sof-audio-pci-intel-mtl 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
```

**To Reproduce**
aplay -Dhw:0,0 -f dat -c 2 /dev/zero -vvv

**Reproduction Rate**
100%

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev/(https://github.com/thesofproject/linux/commit/e684a226dc872968f2d7fdaed0cd16e5cd5af111)
    * SOF: mtl-001-drop-stable(https://github.com/thesofproject/sof/commit/e683f55db037ec3498a2a4165eff0d5b319a684e)
2) Name of the topology file
    * Topology:sof-hda-generic-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: MTL-RVP HDA

[dmesg.txt](https://github.com/thesofproject/sof/files/9673216/dmesg.txt)

lgirdwood, 2022-10-03:
@lyakh does this look like the issue you fixed recently in main ?
@marcinszkudlinski fyi, could be a fix in main if confirmed by @lyakh

keqiaozhang, 2022-10-11:
Checked with tip sof/main, same result:

```
[  116.147054] soundwire_cadence:cdns_fill_msg_resp: soundwire_intel soundwire_intel.link.0: Msg ignored for Slave 0
[  116.147097] soundwire_bus:sdw_program_device_num: soundwire sdw-master-0: No more devices to enumerate
[  116.147254] soundwire_cadence:cdns_update_slave_status_work: soundwire_intel soundwire_intel.link.0: Slave status change: 0x20001
[  116.147356] soundwire_bus:sdw_modify_slave_status: rt711-sdca sdw:0:025d:0711:01: signaling enumeration completion for Slave 4
[  116.147902] soundwire_bus:sdw_slave_set_frequency: rt711-sdca sdw:0:025d:0711:01: Configured bus base 1, scale 3, mclk 19200000, curr_freq 4800000
[  116.148353] rt711-sdca sdw:0:025d:0711:01: runtime PM trying to activate child device sdw:0:025d:0711:01 but parent (sdw-master-0) is not active
[  116.384566] snd_soc_rt711_sdca:rt711_sdca_calibration: rt711-sdca sdw:0:025d:0711:01: rt711_sdca_calibration calibration complete, ret=0
[  116.385114] snd_soc_rt711_sdca:rt711_sdca_io_init: rt711-sdca sdw:0:025d:0711:01: rt711_sdca_io_init hw_init complete
[  116.385129] soundwire_bus:sdw_handle_slave_status: rt711-sdca sdw:0:025d:0711:01: signaling initialization completion for Slave 4
[  116.651781] sof-audio-pci-intel-mtl 0000:00:1f.3: ipc timed out for 0x43000000|0x20701000
[  116.651894] sof-audio-pci-intel-mtl 0000:00:1f.3: preventing DSP entering D3 state to preserve context
[  116.651904] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump start ]------------
[  116.652010] sof-audio-pci-intel-mtl 0000:00:1f.3: Host IPC initiator: 0xc3000000|0x20701000|0x0, target: 0x1b080000|0x0|0x0, ctl: 0x3
[  116.652169] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ IPC dump end ]------------
[  116.652226] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump start ]------------
[  116.652281] sof-audio-pci-intel-mtl 0000:00:1f.3: IPC timeout
[  116.652322] sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (6)
[  116.652403] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM status: 0x5, ROM error: 0x0
[  116.652457] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM debug status: 0x50000005, ROM debug error: 0x0
[  116.652521] sof-audio-pci-intel-mtl 0000:00:1f.3: ROM feature bit enabled
[  116.652567] sof-audio-pci-intel-mtl 0000:00:1f.3: ------------[ DSP dump end ]------------
[  116.652623] sof-audio-pci-intel-mtl 0000:00:1f.3: sof_ipc4_set_get_data: large config get failed at offset 0: -110
[  116.652702] sof-audio-pci-intel-mtl 0000:00:1f.3: error: failed to boot DSP firmware -110
[  116.652763] snd_sof:sof_set_fw_state: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state change: 6 -> 3
[  116.670526] snd_sof_intel_hda_common:mtl_dsp_core_power_down: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x178d04]=0x2000000 successful
[  116.673160] snd_sof_intel_hda_common:mtl_power_down_dsp: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1000]=0x0 successful
[  116.673937] snd_sof:sof_set_fw_state: sof-audio-pci-intel-mtl 0000:00:1f.3: fw_state change: 3 -> 0
[  116.673950] sof-audio-pci-intel-mtl 0000:00:1f.3: error: sof_probe_work failed err: -110
```

mengdonglin, 2022-10-19:
mtl-001-drop-stable is deprecated. Will check SOF main branch when baseline MTL support is upstreamed to SOF main.

closed 2022-10-19 by mengdonglin