thesofproject/sof #6367: [BUG] WORK IN PROGRESS

greg-intel, 2022-10-04:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
Please also read https://thesofproject.github.io/latest/contribute/process/bug-tracking.html for further information on submitting bugs.

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

greg-intel, 2022-10-04:
Resolution verified.

1. git revert
1. `./scripts/docker-run.sh ./scripts/build-tools.sh -T` (on build/local machine)
1. Transferred `cavs-tgl-nocodec.tplg` to `/lib/firmware/intel/avs-tplg`
2. In `/home/ubuntu/sof-test/tools/kmod` ran `sof_remove.sh` followed by `sof_insert.sh`
3. Ran the above reproduction case
4. Passed

ranj063, 2022-10-04:
@greg-intel lets not close this issue because the revert is only a temporary solution. @RanderWang FYI. can you please look into this?

RanderWang, 2022-10-10:
Sure.  Some potential bugs would be detected with my PR as I expected

RanderWang, 2022-10-11:
The problem is that for capture, gain module should be in the same pipeline of dai copier else FW can't find sink. The root reason is that gain module doesn't support get_sink() ops. (closed source FW, not cSOF)

RanderWang, 2022-10-13:
@greg-intel  can we close it since we find root reason

marc-hb, 2022-10-13:
We reverted a regression. @ranj063 asked to file this as a reminder to resubmit the same but fixed "feature", without the regression. Will you resubmit a fixed version or are you giving up for some reason? (because it can't be done?)

ranj063, 2022-10-13:
@marc-hb here's the resubmit (merged but with a revised version of the original) and a fix on top of it pending merge https://github.com/thesofproject/sof/pull/6417

marc-hb, 2022-10-13:
Thanks @ranj063 ! So let's close this once #6417 is merged and the next daily run is passing.

lgirdwood, 2022-10-19:
> Thanks @ranj063 ! So let's close this once #6417 is merged and the next daily run is passing.

Now merged, can this be closed ?

closed 2022-10-19 by marc-hb