thesofproject/sof #6371: [BUG] TGL Regression 07_02_TestKdSsp16000Hz24b32b2ch err -22

antonirx, 2022-10-05:
**Describe the bug**
 Test 07_02_TestKdSsp16000Hz24b32b2ch
 FW replied with error. Error code: -22 0x00000005 0x00000000 0x00004000

**Topology**
[image]

**Reproduction Rate**
100 %

3) Name of the platform(s) on which the bug is observed.
    TGL

*Logs*
[ERR-22.zip](https://github.com/thesofproject/sof/files/9713593/ERR-22.zip)

lgirdwood, 2022-10-05:
Looks like KPB is out of memory. Can you reproduce with IPC4 and Zephyr config ?

antonirx, 2022-10-12:
Cannot reproduce KD test - blocked by higher prio issue.

antonirx, 2022-10-21:
Found also on:
c807d6844b535ff4ad90ac6429a2fef46f398cb4

antonirx, 2023-01-05:
on TGL issue does not occur
Tested FW hash: 4943aaa

lgirdwood, 2023-01-06:
@antonirx main branch for TGL should use IPC4/Zephyr config, only the TGL production branch uses IPC3/xtos config. Can you test with main branch config. If unsure, pls chat with @mwasko .

closed 2025-07-24 by github-actions[bot]