thesofproject/sof #6372: [BUG][CHROME] DSP Panic in 17_x test group

antonirx, 2022-10-05:
**Describe the bug**
Tests affected:
17_01_TestKdSsp16000Hz24b32b2ch
17_06_TestMultiCoreCompSrcPlb48000Hz32b32b8ch
17_08_TestMulticoreKdCore1PlaybackD0ix16000Hz16b16b
17_09_TestMulticoreKdWithResetCore0PlaybackD0ixCoreSwitch16000Hz24b32b
**Topology**
[image]

**Reproduction Rate**
100%

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
    SOF: BRANCH: main
    PLATFORM: TGL
    BUILD: CHROME

**Screenshots or console output**
[image]
[image]

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html
[17_9_DSP_PANIC.zip](https://github.com/thesofproject/sof/files/9713720/17_9_DSP_PANIC.zip)

lgirdwood, 2022-10-05:
@antonirx can this be duplicated with IPC4 pipeline and Zephyr RTOS ?

antonirx, 2022-10-12:
At the moment we have no scenarios for multicore and I2S, we can create them if needed for this bug.

antonirx, 2022-10-21:
Reproduced on TGL:
c807d6844b535ff4ad90ac6429a2fef46f398cb4
new failures:
17_00_TestMultiCoreSameKernelSsp16000Hz32b32b4ch
17_05_TestMultiCoreCompVolCap8000Hz16b16b1ch
17_05_TestMultiCoreCompVolCap48000Hz32b32b8ch
17_07_TestMultiCoreSrcCap8000Hz16b16b1ch
17_07_TestMultiCoreSrcCap16000Hz24b32b1ch
22_05_TestDsmMulticoreSsp48000Hz24b32b2ch

lgirdwood, 2022-10-24:
> At the moment we have no scenarios for multicore and I2S, we can create them if needed for this bug.

Yes please, we need to create these for IPC4/Zephyr config. @mwasko fyi.

antonirx, 2023-01-24:
Not reproduced on TGL: 8bfcf59980b7302bc3621dc3b889ac279ad81ae4

closed 2023-01-24 by wszypelt