thesofproject/sof #6379: [BUG][ADL]: maxim 98373 speaker pop sound when switching audio output device

macchian, 2022-10-07:
**Describe the bug**
98373 speaker pop sound occurred on upstream adl-004-drop-stable  and main branch as well.

experiment:
  1. bypass all smart amp DSM, lib module and binary.
  2. moving FE SSP stop sequence before BE amp shutdown.

both still have pop sound. 

**To Reproduce**
1. plug in headset 
2. playing music on Youtube.com.
3. switching audio devices by CRAS UI
4. speaker pop suddenly when switching audio routing path back to speaker.

**Reproduction Rate**
100%

**Expected behavior**
no pop sound.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.10
    * SOF: upstream adl-004-drop-stable, main branch
 
2) Name of the topology file
    * Topology: sof-adl-max98373-nau8825.tplg

3) Name of the platform(s) on which the bug is observed.
    * Platform: ADL

**Screenshots or console output**
attachment. kernel logs. sof-logger trace logs. 
1. 300Hz pop noise capture
2. trace logs
3. dmesg
4. pop noise.wav

macchian, 2022-10-07:
[dmesg_pop.txt](https://github.com/thesofproject/sof/files/9730800/dmesg_pop.txt)
[image]
[sof_trace_trace.log](https://github.com/thesofproject/sof/files/9730802/sof_trace_trace.log)

macchian, 2022-10-07:
[Noise_without_DSM.zip](https://github.com/thesofproject/sof/files/9730825/Noise_without_DSM.zip)

macchian, 2022-10-07:
[RanderWang](https://github.com/RanderWang), need your advices.

macchian, 2022-10-11:
@lgirdwood, @mwasko, pop also be reproduced on main branch, any comments?

plbossart, 2022-10-12:
@macchian your dmesg log seems wrong, it contains major errors:

````
[    5.034902] sof-audio-pci-intel-tgl 0000:00:1f.3: error: ipc error for 0x30010000 size 20
[    5.034907] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load widget CODEC_ADAPTER1.0
[    5.035022] sof-audio-pci-intel-tgl 0000:00:1f.3: error: tplg component load failed -22
[    5.035031] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to load DSP topology -22
[    5.035034] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_component_probe on 0000:00:1f.3: -22
[    5.035060] sof_nau8825 adl_max98373_nau8825: ASoC: failed to instantiate card -22
[    5.035140] sof_nau8825: probe of adl_max98373_nau8825 failed with error -22
````

plbossart, 2022-10-12:
@macchian we need to experiment with simpler setups at the hw: device level, i.e. without CRAS in the picture

Suggestions:
a) play on headset, stop 10s, play on amp, and check if there is any pop.
b) play on the amp while playing on the headset
c) try to play on the headset, wait 500ms and play on the amp with a script to simulate a switch.

macchian, 2022-10-13:
> a) play on headset, stop 10s, play on amp, and check if there is any pop.

my bad... wrong file. reattached the correct one.
[dmesg_pop.txt](https://github.com/thesofproject/sof/files/9770836/dmesg_pop.txt)

macchian, 2022-10-13:
> @macchian we need to experiment with simpler setups at the hw: device level, i.e. without CRAS in the picture
>
> Suggestions: a) play on headset, stop 10s, play on amp, and check if there is any pop. [...]

@plbossart Thanks for your advices! As you mentioned, I write a script spk_switch_on_off.sh and gen a 1k_sine_48k_16bits_60s.wav
[1k_sine_48k_16bits_60s.zip](https://github.com/thesofproject/sof/files/9775824/1k_sine_48k_16bits_60s.zip)

[spk_switch_on_off.zip](https://github.com/thesofproject/sof/files/9775817/spk_switch_on_off.zip)

The script is playing 1k sine wav in background and every 500ms switching speaker(-D hw:0,0) to headphone (-D hw:0,1) repeatedly.

Summarize test result below:

a) play on headset, stop 10s, play on amp, and check if there is any pop.
- Speaker still has pop but not every time.

b) play on the amp while playing on the headset
c) try to play on the headset, wait 500ms and play on the amp with a script to simulate a switch.
- Speaker and headphone have pop during every on/off switching.

mengdonglin, 2022-10-14:
@macchian It's not likely to be SOF FW or driver issue, as the firmware and driver are not really involved in switching endpoints. It's more likely to be a codec or machine driver or UCM issue.

macchian, 2022-10-17:
Hi @mengdonglin ,
  From Maxim's debugging results, it looks like not relevant to codec or DSM issue. Because the experiments below still hear the pop sound. There is 300Hz signal producing. (see attachment file pop picture)
Is it a possible way by eq-iir reducing it?

Currently it reproduces on TGL and ADL/RPL, I believe MTL should as well. Could you suggest further method to fix it?

Maxim's test:
1. bypass DSM algo, the pop still occurs.
2. Keep the codec amp pin power always On , the pop exists.

3. I also modified the FE SSP stop sequence before BE amp shutdown.
    The sequence ensures the i2s data stop before codec amp shutdown. In this case, pop can hear.

plbossart, 2022-10-18:
If you have a pop on the headset, then DSM is not involved. It's likely the codec driver that doesn't do the right thing. And sure enough, it's the nau8825 that's used. That's a new variant that hasn't been used before....

lgirdwood, 2022-10-19:
@macchian do you still see the pops when switching speakers / headphones when playing silence ?
```aplay -f dat /dev/zero```

macchian, 2022-10-20:
> @macchian do you still see the pops when switching speakers / headphones when playing silence ? `aplay -f dat /dev/zero`

@lgirdwood 
I test the silence playback. The silence result is no pop sound on 98373 amp when switching headphone to speaker repeatedly.
  Furthermore I test another ADL device with 98390 amp. It’s no pop either playing 1k or silence.

plbossart, 2022-10-20:
@macchian interesting result.  Is there any difference in the DAI configuration (clocks, TDM, etc) and/or support for DSM between the device w/ 98373 (pops) and the device w/ 98390 (no pops). Can you please point us to the topologies?

macchian, 2022-10-21:
> @macchian interesting result. Is there any difference in the DAI configuration (clocks, TDM, etc) and/or support for DSM between the device w/ 98373 (pops) and the device w/ 98390 (no pops). Can you please point us to the topologies?

Share the dai diffs between information.

98373: 8ch tdm with 2ch playback(stereo speakers) + 8ch capture(echo ref+dsm slots)

98390: 4ch tdm with 2ch playback (stereo speakers). no DSM.

Note:
1. pop reproduced by dsm_pass.
2. enabled SSP_CC_BCLK_ES on 98373, still has pop.

98373 ->
	SSP_CONFIG(DSP_B, SSP_CLOCK(mclk, SSP_MCLK, codec_mclk_in),
		      SSP_CLOCK(bclk, 12288000, codec_slave),
		      SSP_CLOCK(fsync, 48000, codec_slave),
		      SSP_TDM(8, 32, 15, 255),
		      SSP_CONFIG_DATA(SSP, SMART_SSP_INDEX, 32, 0, SMART_SSP_QUIRK)))

https://github.com/thesofproject/sof/blob/73a413a38fc24a9763ac55df9ebb528219c5d139/tools/topology/topology1/sof-smart-amplifier.m4#L240

98390 ->
	CODEC, `MAX98390', `
	SSP_CONFIG(DSP_B, SSP_CLOCK(mclk, 19200000, codec_mclk_in),
		SSP_CLOCK(bclk, 6144000, codec_slave),
		SSP_CLOCK(fsync, 48000, codec_slave),
		SSP_TDM(4, 32, 3, 15),
		SSP_CONFIG_DATA(SSP, SPK_SSP_INDEX, 32)))',

https://github.com/thesofproject/sof/blob/73a413a38fc24a9763ac55df9ebb528219c5d139/tools/topology/topology1/sof-tgl-max98357a-rt5682.m4#L442

plbossart, 2022-10-21:
Humm, maybe a red-herring but the use of 12.288 MHz could be a problem, depending on the hardware layout. It's been a problem before and that could explain difference between hardware skus.

@macchian can you remind me why we are using 8 slots for the capture? we've got two amplifiers, so do we really need 4 slots for each?

lgirdwood, 2022-10-24:
> [dmesg_pop.txt](https://github.com/thesofproject/sof/files/9730800/dmesg_pop.txt) [image] [sof_trace_trace.log](https://github.com/thesofproject/sof/files/9730802/sof_trace_trace.log)

@macchian @plbossart one other thing here is the 40ms gap between pops and the magnitude of the pops.

@macchian some more things to check.

1) Can you check the CRAS buffer/period size. 40ms is way too long for FW inject pops as it will only have a < 10ms internally.

2) What is the size of the pop in mV ? The scale of the pops is a lot bigger than the signal i.e. invalid/stale data being replayed would be at the same volume.

3) Can you capture the pop L & R channels individually with different scope probes. Do the pops align in time domain and magnitude ?

4) Does the pop magnitude vary with HW codec or Amp volume ? i.e. do you see the spike playing silence at max HW volume. Use the scope for this as a small mV pop may not be audible.

ujfalusi, 2022-10-24:
ASoC have simple infrastructure to test and debug pop noise generated by codecs along the path when the DAPM widget got changed.
It is injecting delays between register writes and one can watch `dmesg -w` and listen for pop noise to narrow down it's source.

`echo 1000 > /sys/kernel/debug/asoc/`name-of-the-sound-card`/dapm_pop_time`
will inject 1 second delay between register writes.

macchian, 2022-10-26:
> Humm, maybe a red-herring but the use of 12.288 MHz could be a problem, depending on the hardware layout. It's been a problem before and that could explain difference between hardware skus. [...]

@plbossart , 
> @macchian can you remind me why we are using 8 slots for the capture? 
- The VI data capture 8 slots set for maximum 4 x speakers.

we've got two amplifiers, so do we really need 4 slots for each?
- Because the identical sof-smart-amplifier.m4 topology supports 2 or 4 speakers SKUs. Therefore the configuration here has been set maximum 4 speakers settings.

Besides, the patch history mentioned:
commit 97377c3757e9889597fb4c43d2fa7aef05a0f06e

    This commit changes SSP bclk rate from 9600000 to
    12288000 ind ssp container to 32 bits order to fix
    glitch issue.

macchian, 2022-10-26:
> ASoC have simple infrastructure to test and debug pop noise generated by codecs along the path when the DAPM widget got changed. It is injecting delays between register writes and one can watch [...]

commands I ran on device:
`echo 1000 > /sys/kernel/debug/asoc/`name-of-the-sound-card`/dapm_pop_time`
play -n synth sine 

pop looks like happen at

hda-stream

lgirdwood, 2022-10-26:
@macchian any update on tests 1, 2, 3 or 4 above ? Results from any of them would be useful info to consider now whilst others are in progress.

lgirdwood, 2022-10-26:
@macchian and the DAPM pop test is easy to do when time permits :)

lgirdwood, 2022-10-26:
Thanks @macchian - difficult to tell which part is popping from the video. 

I've looked at the pops on audacity, we always have 2 pops on and the gap is variable 30 - 40 ms. The important thing is that the audio is correct between pops and the sine wave peaks still align before and after pops.

@macchian this looks like the pop is introduced by the AMP powering ON/OFF otherwise the sine wave would come out of alignment after the first pop. Please check the with the AMP provider the correct ON/OFF programming sequence with any delays. We may be misaligned here,

bardliao, 2022-10-27:
@lgirdwood Can we enable loopback mode? 
https://github.com/thesofproject/sof/blob/73a413a38fc24a9763ac55df9ebb528219c5d139/tools/topology/topology1/sof-smart-amplifier.m4#L244
Set SMART_SSP_QUIRK in the topology can enable loopback mode, am I right?

bardliao, 2022-10-27:
It seems smart amp topology doesn't support LBM.
So, I edit nocodec topology to fit the product to test loopback mode.
https://github.com/bardliao/sof/commit/f1ea06a847ac3e8a8549402b64a9d07adaac70f2
The loopback wav looks pretty good to me.
[loop.zip](https://github.com/thesofproject/sof/files/9878101/loop.zip)

Note that the I2S format between the smart amp topology and the nocodec topology are different.
Nocodec:
```
DAI_CONFIG(SSP, SSP1_IDX, 7, SSP1-Codec,
           SSP_CONFIG(I2S, SSP_CLOCK(mclk, 38400000, codec_mclk_in),
                      SSP_CLOCK(bclk, 2400000, codec_slave),
                      SSP_CLOCK(fsync, 48000, codec_slave),
                      SSP_TDM(2, 25, 3, 3),
                      SSP_CONFIG_DATA(SSP, SSP1_IDX, 24, 0, SSP_QUIRK_LBM, 0,
                                      eval(SSP_CC_MCLK_ES | SSP_CC_BCLK_ES))))
```
Smart amp:
```
DAI_CONFIG(SSP, SMART_SSP_INDEX, SMART_BE_ID, SMART_SSP_NAME,
        SSP_CONFIG(DSP_B, SSP_CLOCK(mclk, SSP_MCLK, codec_mclk_in),
                      SSP_CLOCK(bclk, 12288000, codec_slave),
                      SSP_CLOCK(fsync, 48000, codec_slave),
                      SSP_TDM(8, 32, 15, 255),
                      SSP_CONFIG_DATA(SSP, SMART_SSP_INDEX, 32, 0, SMART_SSP_QUIRK)))
```
I tried to use the same I2S format, but I will get IPC error when I start the capture PCM.

bardliao, 2022-10-27:
BTW, If someone can probe the I2S pins physically with a logic analyzer to get the audio data from I2S pins directly, it would be  
a better way to clarify where is the pop from.

bardliao, 2022-10-27:
I can hear pop noise when speaker playback starts **AND** stops. And I also hear obvious pop noise when headphone playback stops.
The test wav is [1K_0dB.zip](https://github.com/thesofproject/sof/files/9878325/1K_0dB.zip)

ujfalusi, 2022-10-27:
@bardliao, can you isolate the pop with dapm_pop_time of 2 seconds?
Does the codec have Mute switch on headset, speaker? Does toggling it produces pop noise as well?

bardliao, 2022-10-28:
@ujfalusi @lgirdwood @macchian I am pretty sure the pop is from the codec itself after doing some experiment.
There are two speaker codecs (left and right) in the device.
And a pop is generated when a speaker is started and stopped. That's why we can hear two pops when speaker playback starts and two pops when speaker playback stops.
If I set dapm_pop_time  = 2sec, there are some "da" "da" sound in the transition stage. 
Below is the dmesg tested with  dapm_pop_time  = 2000, I added some empty line when I hear pop.
```
[Apr22 14:02] sof_nau8825 adl_max98373_nau8825: DAPM sequencing finished, waiting 2000ms
[  +2.028752] sof_nau8825 adl_max98373_nau8825: DAPM sequencing finished, waiting 2000ms
[  +2.050928] max98373 i2c-MX98373:00: pop test : Queue Right HiFi Playback: reg=0xffffffff, 0x0/0x0
[  +0.000046] sof-audio-pci-intel-tgl 0000:00:1f.3: pop test : Queue Smart Amplifier Playback 0: reg=0xffffffff, 0x0/0x0
[  +0.000023] sof-audio-pci-intel-tgl 0000:00:1f.3: pop test : Queue PCM0P: reg=0xffffffff, 0x1/0x0
[  +0.000020] sof-audio-pci-intel-tgl 0000:00:1f.3: pop test : Queue SSP1.OUT: reg=0xffffffff, 0x1/0x0
[  +0.000022] max98373 i2c-MX98373:00: pop test : Queue Right BE_OUT: reg=0xffffffff, 0x0/0x0
[  +0.000022] max98373 i2c-MX98373:00: pop test : Queue Right DAI Sel Mux: reg=0xffffffff, 0x1/0x1
[  +0.000020] max98373 i2c-MX98373:00: pop test : Queue Right Amp Enable: reg=0x202b, 0x1/0x1
[  +0.000021] max98373 i2c-MX98373:00: pop test : Applying 0x1/0x1 to 202b in 2000ms
[  +2.057783] max98373 i2c-MX98373:00: pop test : Right Amp Enable POST_PMU
[  +0.032544] sof-audio-pci-intel-tgl 0000:00:1f.3: pop test : Queue BUF1.0: reg=0xffffffff, 0x1/0x0
[  +0.000025] sof-audio-pci-intel-tgl 0000:00:1f.3: pop test : Queue SMART_AMP1.0: reg=0xffffffff, 0x1/0x0
[  +0.000010] sof-audio-pci-intel-tgl 0000:00:1f.3: pop test : Queue BUF1.1: reg=0xffffffff, 0x1/0x0
[  +0.000015] sof_nau8825 adl_max98373_nau8825: pop test : Queue Right Spk: reg=0xffffffff, 0x0/0x0
[  +0.000203] sof_nau8825 adl_max98373_nau8825: DAPM sequencing finished, waiting 2000ms

[  +2.002763] max98373 i2c-MX98373:01: pop test : Queue Left HiFi Playback: reg=0xffffffff, 0x0/0x0
[  +0.000027] max98373 i2c-MX98373:01: pop test : Queue Left BE_OUT: reg=0xffffffff, 0x0/0x0
[  +0.000011] max98373 i2c-MX98373:01: pop test : Queue Left DAI Sel Mux: reg=0xffffffff, 0x1/0x1
[  +0.000012] max98373 i2c-MX98373:01: pop test : Queue Left Amp Enable: reg=0x202b, 0x1/0x1
[  +0.000012] max98373 i2c-MX98373:01: pop test : Applying 0x1/0x1 to 202b in 2000ms
[  +2.060351] max98373 i2c-MX98373:01: pop test : Left Amp Enable POST_PMU
[  +0.032063] sof_nau8825 adl_max98373_nau8825: pop test : Queue Left Spk: reg=0xffffffff, 0x0/0x0
[  +0.000335] sof_nau8825 adl_max98373_nau8825: DAPM sequencing finished, waiting 2000ms

[  +5.758428] sof_nau8825 adl_max98373_nau8825: pop test : Queue Right Spk: reg=0xffffffff, 0x0/0x0
[  +0.000030] max98373 i2c-MX98373:00: pop test : Queue Right Amp Enable: reg=0x202b, 0x0/0x1
[  +0.000012] max98373 i2c-MX98373:00: pop test : Applying 0x0/0x1 to 202b in 2000ms

[  +2.017122] max98373 i2c-MX98373:00: pop test : Right Amp Enable POST_PMD
[  +0.032060] max98373 i2c-MX98373:00: pop test : Queue Right BE_OUT: reg=0xffffffff, 0x0/0x0
[  +0.000027] max98373 i2c-MX98373:00: pop test : Queue Right DAI Sel Mux: reg=0xffffffff, 0x0/0x1
[  +0.000010] max98373 i2c-MX98373:00: pop test : Queue Right HiFi Playback: reg=0xffffffff, 0x0/0x0
[  +0.000280] sof_nau8825 adl_max98373_nau8825: DAPM sequencing finished, waiting 2000ms
[  +2.004029] sof_nau8825 adl_max98373_nau8825: pop test : Queue Left Spk: reg=0xffffffff, 0x0/0x0
[  +0.000028] sof-audio-pci-intel-tgl 0000:00:1f.3: pop test : Queue BUF1.0: reg=0xffffffff, 0x0/0x0
[  +0.000009] sof-audio-pci-intel-tgl 0000:00:1f.3: pop test : Queue SMART_AMP1.0: reg=0xffffffff, 0x0/0x0
[  +0.000009] sof-audio-pci-intel-tgl 0000:00:1f.3: pop test : Queue BUF1.1: reg=0xffffffff, 0x0/0x0
[  +0.000014] max98373 i2c-MX98373:01: pop test : Queue Left Amp Enable: reg=0x202b, 0x0/0x1
[  +0.000010] max98373 i2c-MX98373:01: pop test : Applying 0x0/0x1 to 202b in 2000ms
[  +2.059528] max98373 i2c-MX98373:01: pop test : Left Amp Enable POST_PMD
[  +0.032118] max98373 i2c-MX98373:01: pop test : Queue Left BE_OUT: reg=0xffffffff, 0x0/0x0
[  +0.000025] max98373 i2c-MX98373:01: pop test : Queue Left DAI Sel Mux: reg=0xffffffff, 0x0/0x1
[  +0.000010] max98373 i2c-MX98373:01: pop test : Queue Left HiFi Playback: reg=0xffffffff, 0x0/0x0
[  +0.000012] sof-audio-pci-intel-tgl 0000:00:1f.3: pop test : Queue Smart Amplifier Playback 0: reg=0xffffffff, 0x0/0x0
[  +0.000008] sof-audio-pci-intel-tgl 0000:00:1f.3: pop test : Queue PCM0P: reg=0xffffffff, 0x0/0x0
[  +0.000008] sof-audio-pci-intel-tgl 0000:00:1f.3: pop test : Queue SSP1.OUT: reg=0xffffffff, 0x0/0x0
[  +0.000267] sof_nau8825 adl_max98373_nau8825: DAPM sequencing finished, waiting 2000ms

[  +2.006744] sof_nau8825 adl_max98373_nau8825: DAPM sequencing finished, waiting 2000ms
```

There is no single mute switch for speaker, but we have "Spk Switch" that can enable/disable Speakers.
[controls.txt](https://github.com/thesofproject/sof/files/9884565/controls.txt)
And I can hear the pop when I set `amixer -c0 cset name='Left Spk Switch' 0` and `amixer -c0 cset name='Left Spk Switch' 1` during playback is running.
However, there is no "da" "da" noise when I switch "Spk Switch" on or off.
Below is the dmesg when I switch 'Left Spk Switch'.
```
[... cut ...]
[  +0.000017] sof-audio-pci-intel-tgl 0000:00:1f.3: pop test : Queue PCM0P: reg=0xffffffff, 0x0/0x0
[  +0.000017] sof-audio-pci-intel-tgl 0000:00:1f.3: pop test : Queue SSP1.OUT: reg=0xffffffff, 0x0/0x0
[  +0.000267] sof_nau8825 adl_max98373_nau8825: DAPM sequencing finished, waiting 1000ms
```

Intel Audio DSP doesn't know user is switching the "Spk Switch", so, the audio data is consistent.
It implies that the pop is generated when the codec is doing the enabling/disabling process.
Like
Queue Left Amp Enable: reg=0x202b, 0x1/0x1
Queue Left Amp Enable: reg=0x202b, 0x0/0x1

bardliao, 2022-10-28:
> Hi @mengdonglin , From Maxim's debugging results, it looks like not relevant to codec or DSM issue. Because the experiments below still hear the pop sound. There is 300Hz signal producing. (see [...]

@macchian Do you know if they do anything to prevent the codec driver from accessing the codec registers?
For example, let regcache_cache_only be always true, or use the dummy codec driver instead of max98373 codec driver.
If they didn't do that, the codec driver will still act as usual, and it can't prove the pop is not from the enabling/disabling process.

> 3. I also modified the FE SSP stop sequence before BE amp shutdown.
>    The sequence ensures the i2s data stop before codec amp shutdown. In this case, pop can hear.

macchian, 2022-11-02:
Hi all,
  As per working with @bardliao and @ryans-lee @Maxim, I summarize the pop test results below:

1)	Make ‘Digital Volume’ mute and test if it generates the same amount of noise.
-	Pop sound level is following by ‘Digital Volume’ boost/attenuate [Confirmed]
-	I2S signal scope need ODM rework I2S. Jonathan. [TBD]

2)	Manually turn on/off 0x20FF(Global en) register while playing silent music and check the noise level. This amount of noise is how much we can expect from the amp.
-  During playing by aplay -D hw:0,0 -f dat 1k.wav,
	a. turn off left Spk global_en first
    #echo 0 0x20FF > /sys/kernel/debug/regmap/i2c-MX98373:01 (One pop sound)
	b. then turn off right Spk global_en
    #echo 0 0x20ff > /sys/kernel/debug/regmap/i2c-MX98373:00 (One pop sound)
  c. turn on right spk global_en again 
    #echo 1 0x20ff > /sys/kernel/debug/regmap/i2c-MX98373:00 (speaker sound not resume sound when set global_en on)
    So in this case, I only observe pop when running right spk global_en off.

3)	Make the 0x202B always ON and do the same test in #2.
-	Applied the codes from Bard.
Pop occurs:
1 pop occurs when aplay starting, 2 pop occur when aplay stopped. [Confirmed]

static const struct snd_soc_dapm_widget max98373_dapm_widgets[] = {
SND_SOC_DAPM_DAC_E("Amp Enable", "HiFi Playback",
-       MAX98373_R202B_PCM_RX_EN, 0, 0, max98373_dac_event,
+       SND_SOC_NOPM, 0, 0, max98373_dac_event,
        SND_SOC_DAPM_POST_PMU | SND_SOC_DAPM_POST_PMD |
        SND_SOC_DAPM_PRE_PMU | SND_SOC_DAPM_PRE_PMD),
SND_SOC_DAPM_MUX("DAI Sel Mux", SND_SOC_NOPM, 0, 0,
@@ -443,6 +443,8 @@ static int max98373_probe(struct snd_soc_component *component)
                MAX98373_R2043_AMP_EN,
                MAX98373_SPK_EN_MASK, 1);
-	
-	+       regmap_write(max98373->regmap, MAX98373_R202B_PCM_RX_EN, 0x1);

4)	Switch ‘Digital Volume’during playing 1k sine
-	During aplay -D hw:0,0 -f dat 1k.wav, switch Speaker Digital Volume from 0 to 100 repeatedly. No pop. Confirmed by Bard and Mac.

5)	Switch ‘Spk Switch’ during playing 1k sine
-	During aplay -D hw:0,0 -f dat 1k.wav, switch ‘Left/Right Spk Switch’ repeatedly, 1 pop occurs from aplay starting and 2 pops from aplay stopped. Confirmed by Bard and Mac.

6)	Play silence.wav and stop
-	No pop from aplay starting and stopped. [Confirmed]

plbossart, 2022-12-13:
@bardliao can you provide a summary? My understanding is that the issue is at the codec driver level?

bardliao, 2022-12-13:
@plbossart There are 2 issues.
a. When we run aplay with a 1k tone and stop during playback, we will hear a noise due to the discontinuous audio.
b. 
1. Start playback with Chrom UI on speaker
2. Insert a headphone -> audio output will switch to the headphone
3. remove the headphone -> ideally, audio should stop. But it will start speaker and stop. We will hear a pop when the speaker is powered on.

We measured the I2S signal from I2S pins with a logic analyzer and the captured signal looks good. So, I don't think it is a SOF issue.

macchian, 2023-02-08:
refer to #7048 and #4179

johnylin76, 2023-02-08:
FYI, the production build for ADL-P devices have been migrated from adl-004 to rpl-001, which means the fix commit should be merged back to rpl-001-drop-stable instead.

lgirdwood, 2023-02-21:
@macchian can we close ?

macchian, 2023-02-22:
> @macchian can we close ?

Yes, the remaining fix PR #4179 has approved and to be merged. We can close it. Thanks!

closed 2023-02-22 by macchian