thesofproject/sof #6397: Tiger Lake SOF Digital Mic stutters when sharing flowgraph w/ USB devices.

GITLauriette, 2022-10-11:
**Describe the bug**
The digital microphone on a Tiger Lake chipset works properly only when its flowgraph is not shared with other asynchronous USB audio input devices.  The screenshots show normal samples when the Digital Mic is the sole input to Reaper (DAW used for recording).  However, as soon as an external USB audio interface is attached to other Reaper inputs, the Digital Mic begins dropping sections of data.  This experiment captures before/after the USB source is connected, whilst I was singing "Ahhhhh" into the Mics just to create a visible signal.

[image]
[image]
[image]
[image]

**To Reproduce**
1) Start with completely disconnected devices (using qjackctl)
2) Connect laptop internal Digital Mic to something -- works fine
3) Connect some other USB input to anything that shares the flowgraph with the Digital Mic, and the Digital Mic will begin stuttering.

**Reproduction Rate**
All the time -- 100% repeatable failure

**Expected behavior**
The Digital Mic should provide correct samples regardless of which other devices are sharing its timing/fowgraph.

**Impact**
This has been a serious problem, as the laptop we intended to use for group presentations was unable to employ it's own microphone.  In desperation, I wrote a hack JACK client that forks into two separate processes for Input and Output, and communicates between the two via mmap().  The Digital Mic hooks only to this client's input, so that the Mic never "sees" any asynchronous clock domain directly.  This separation of clocks can be checked in pw-top.  So, we have this hack workaround for the moment...

**Environment**
Vanilla Fedora 36 configuration on HP pavilion laptop, model 15-eg0165st

[alan@metta ~]$ rpm -qa | grep sof | grep firm
alsa-sof-firmware-2.2.2-1.fc36.noarch

[alan@metta ~]$ uname -a
Linux metta 5.19.14-200.fc36.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Oct 5 21:31:17 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

[alan@metta ~]$ inxi -A
Audio:
  Device-1: Intel Tiger Lake-LP Smart Sound Audio
    driver: sof-audio-pci-intel-tgl
  Device-2: Universal Audio Volt 2 type: USB
    driver: hid-generic,snd-usb-audio,usbhid
  Sound Server-1: ALSA v: k5.19.14-200.fc36.x86_64 running: yes
  Sound Server-2: PipeWire v: 0.3.59 running: yes

**Screenshots or console output**
Screenshots are attached, showing working and non-working qjackctl connections and samples recorded using Reaper.

I'm not aware of this failure being "logged" anywhere.  If more internal information is needed please let me know.

Thank you for your hard work, and attention to this matter.
  -alan

plbossart, 2022-10-11:
There is no relationship or dependency at the hardware or firmware level between DMIC and USB. There's nothing we can do in kernel or firmware to solve your problem @DoOrDaiii 

I believe the issue is that you are trying to record in the same session from two interfaces which have their own master clock. That can't be good, one of the two is going to be faster or slower than then other, and that means underflows and overflows in the data stream. When ALSA drivers detect this, they stop, prepare and re-trigger the interface, so you will have LONG periods with no data as a consequence. That seems to match your waveforms.

Some sort of asynchronous sample rate conversion is needed in userspace IMHO.

DoOrDaiii, 2022-10-11:
Thank you for your reply, plbossart .

JACK clock domain crossings have traditionally being done either with zita_a2j (my personal favorite), or alsa_in.  I assumed that the JACK interfaces presented by pipewire handled hardware I/O in some similar way that allowed multiple async clocks to be interconnected.  Is that a misconception on my part?

On this same laptop I've tried connecting as many as four different USB audio sources/sinks simultaneously using pipewire/JACK, and they all work flawlessly.  The only exception has been when trying to include the laptop internal digital Mic in such a mix.  Given that pipewire works smoothly with all other independent USB streams, why should it be that only this one particular component (the internal Digital Mic) is having an async problem?

Also, on another older laptop, also running FC36 but with non-SOF internal interface, 

[alan@janeycat ~]$ inxi -A
Audio:
  Device-1: Intel Ice Lake-LP Smart Sound Audio driver: snd_hda_intel
  Device-2: Universal Audio Volt 2 type: USB driver: hid-generic,snd-usb-audio,usbhid
  Sound Server-1: ALSA v: k5.19.11-200.fc36.x86_64 running: yes
  Sound Server-2: PipeWire v: 0.3.58 running: yes

Everything works fine.  I'm able to combine the laptop internal Mic with external USB devices and they're all happy.  This is why I thought the problem was specific to the SOF firmware.

> Some sort of asynchronous sample rate conversion is needed in userspace IMHO.

You're right about that, and I wrote a hack JACK client to separate the Digital Mic into its own private clock domain.  But the SOF microphone is the ONLY one of many devices that seems to need this special treatment.

plbossart, 2022-10-11:
There are quite a few USB devices which 'lie' about their asynchronous behavior @DoOrDaiii. Most headsets are synchronous but their descriptors mention an ASYNC capability to work-around legacy bugs with obscure versions of Windows. That would explain why multiple USB devices behave in the same manner, since they all lock on the SOF timing.

Even with true async devices, all devices would receive the same amount of data based on the SOF timing and the adjustments would only very slowly. You would need to wait for a while before seeing differences between devices.

What you should do is test at the hw: device level. Try recording with arecord -Dhw: from a USB device and a DMIC device in parallel, without even using all your fancy setup.

If all recorded files are fine, then the issue is in your application when combining different streams captured at different rates.

DoOrDaiii, 2022-10-11:
Thank you, once again, @plbossart .  I had not realized that supposedly asynchronous USB devices may be operating differently at a lower level.

> What you should do is test at the hw: device level. Try recording with arecord -Dhw: from a USB device and a DMIC device in parallel, without even using all your fancy setup.

I did this by simultaneously executing two commands, letting them run for a few minutes:
$ arecord -r 48000 -c 2 -D hw:1,6 -f S32_LE sof.wav
$ arecord -r 48000 -c 2 -D hw:0 -f S32_LE usb.wav

As you predicted, both files look perfectly fine!

So, my apologies, it looks like I've posted this "bug" in the wrong place.  The problem seems to lie within the pipewire/JACK mechanism (which I believe is supposed to handle differing clock rates), NOT within the SOF interface.

I will close this ticket as RESOLVED, and with gratitude for the time you've spent helping me understand what's going on.  Best Regards,  -alan

closed 2022-10-11 by DoOrDaiii

plbossart, 2022-10-11:
I would guess the problem is that the userspace configuration is based on a nominal frequency value and when there is more than one clock source things go South quickly. Usually there needs to be some sort of explicit tagging that one or more inputs are not synchronous to the master.

It's fine to report such issues, it also gives us an idea of the complicated setups used on top of our firmware/driver. Best of luck @DoOrDaiii  - nice username btw.