thesofproject/sof #6403: [BUG] SSP slave mode stopped working with v2.0 release on APL-Up2

plbossart, 2022-10-12:
**Describe the bug**
The sof-wm8804 machine driver is used to put together an APL-UP2 board with an HifiBerry Digi+ HAT board.
This worked fine for several years.

With firmware updates, this setup is broken, playback xruns without ever starting. 

Release 1.8 and 1.9 work fine.
Release 2.0 does not work at all.

The key point is that this setup relies on the SSP slave mode, aka codec_provider. We've clearly dropped the ball on validation of this mode.

**To Reproduce**
Use an Up2 board with a Digi+ and configure the ACPI scripts to use 

**Reproduction Rate**
100%

**Expected behavior**
no xruns with SSP slave supported.

**Impact**
annoyance, it's a regression but downgrading firmware solves the issue.

**Screenshots or console output**

````
root@plb-UP-APL01:~# speaker-test -Dhw:0,0 -c2 -r48000

speaker-test 1.2.6

Playback device is hw:0,0
Stream parameters are 48000Hz, S16_LE, 2 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 96 to 16384
Period size range from 48 to 4096
Using max buffer size 16384
Periods = 4
was set period_size = 4096
was set buffer_size = 16384
 0 - Front Left
Write error: -5,Input/output error
xrun_recovery failed: -5,Input/output error
Transfer failed: Input/output error
````

plbossart, 2022-10-12:
I'll be happy to bisect when someone tells me how this is supposed to be done, the use of git submodules is a complete PITA, even more so when there are dependencies between parts.

marc-hb, 2022-10-15:
Unless the coupling between the main repo and the sub repo is very loose, bisecting requires running an extra `git submodule update --recursive --init` to keep everything in sync at every step. Only then will `git status` report a clean repo.

However: building rimage commits before 02abc5d342a3 fails with openssl3 because it was using deprecated RSA functions `error: ‘RSA_many_functions’ is deprecated: Since OpenSSL 3.0 -Werror=deprecated-declarations`

Workarounds:
- Compile on an old Linux distribution still shipping openssl1.1 (tested)
- Install openssl1 manually (untested)
- Try this barely tested rimage hack:
```diff
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -27,7 +27,7 @@ add_executable(rimage
 set_property(TARGET rimage PROPERTY C_STANDARD 99)
 
 target_compile_options(rimage PRIVATE
-       -Wall -Werror -Wmissing-prototypes -Wimplicit-fallthrough
+       -Wall         -Wmissing-prototypes -Wimplicit-fallthrough
 )
```

Warning: openssl3 + rimage was broken for a while. I don't think it affects APL but maybe best to avoid the rimage commit range between 0f64a20183a8 and 02abc5d342a3? @juimonen ?

closed 2025-07-24 by github-actions[bot]