thesofproject/sof #6405: [BUG][IPC4] enabling volume and mixer in cavs-nocodec tplg causes some IPC errors

keqiaozhang, 2022-10-12:
**Describe the bug**
Yesterday, we enabled the volume and mixer in cavs-nocodec tplg:https://github.com/thesofproject/sof/commit/5594ad3512830feaa478677718036ffabf51a454

This will cause some IPC errors on IPC4-NOCODEC platforms. Inner test ID:16105
Both IPC4 tester and SOF IPC4-Zephyr firmware have same issues. It can be triggered with some different cases like:
`check-playback-50rounds.sh`,`check-signal-stop-start-playback-50.sh`,`multiple-pause-resume-50.sh` .etc.

**DMESG**
```
[ 1662.311935] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 0 state 3
[ 1662.311938] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13000003|0x0: GLB_SET_PIPELINE_STATE
[ 1662.312140] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000000|0x0: GLB_SET_PIPELINE_STATE
[ 1662.312203] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x13000003|0x0: GLB_SET_PIPELINE_STATE
[ 1662.312209] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 0 state 4
[ 1662.312212] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13000004|0x0: GLB_SET_PIPELINE_STATE
[ 1662.312558] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000007|0x0: GLB_SET_PIPELINE_STATE
[ 1662.312562] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 7 - Unsupported operation requested
[ 1662.312596] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0x13000004|0x0
[ 1662.312603] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to set state 4 for pipeline 1
[ 1662.312606] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[ 1662.312610] kernel:  Port0: ASoC: trigger FE cmd: 1 failed: -22
[ 1662.312614] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[ 1662.312636] kernel: snd_sof_intel_hda_common:hda_dsp_stream_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x140000 successful
[ 1662.312658] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 0 state 2
```
```
[ 2171.879716] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[ 2171.879734] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 0 state 3
[ 2171.879743] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13000003|0x0: GLB_SET_PIPELINE_STATE
[ 2171.890255] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000006|0x0: GLB_SET_PIPELINE_STATE
[ 2171.890261] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 6 - Unknown error while processing the request
[ 2171.890298] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0x13000003|0x0
[ 2171.890309] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to pause pipeline 1
[ 2171.890335] kernel: snd_sof_intel_hda_common:hda_dsp_stream_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: FW Poll Status: reg[0x160]=0x20140000 successful
[ 2171.890362] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[ 2171.890369] kernel:  Port0: ASoC: trigger FE cmd: 0 failed: -22
```

**MTRACE**
```
[    0.000000] <inf> dma_intel_adsp_gpdma: intel_adsp_gpdma_init: dma dma@7c000 initialized
[    0.000000] <inf> dma_intel_adsp_gpdma: intel_adsp_gpdma_init: dma dma@7d000 initialized
*** Booting Zephyr OS build v3.2.0-rc1-2-g0956647aaf6b  ***
[    0.000000] <inf> main: SOF on intel_adsp_cavs25
[    0.000000] <inf> power: pm_runtime_enable_dsp dsp_d0_sref 1
[    0.000000] <inf> ll_schedule: dma_single_chan_domain_init(): num_dma 2, clk 4
[    0.000000] <wrn> ll_schedule: zephyr_ll_scheduler_init(): unsupported domain 2
[    0.000000] <inf> ipc: ipc_init()
[    0.000000] <inf> idc: idc_init()
[    0.000000] <inf> dma_trace: dma_trace_init_complete()
[    0.000000] <inf> hda_dma: hda-dmac :4 -> probe
[    0.000000] <inf> dma: dma_get() ID 4 sref = 1 busy channels 0
[    0.000000] <inf> main: SOF initialized
[    0.000000] <wrn> ll_schedule: zephyr_ll_scheduler_init(): unsupported domain 2
[    0.000000] <inf> idc: idc_init()
[    0.001000] <inf> ipc: rx	: 0xc4000000|0x31400008
[    0.001000] <wrn> basefw: returning success for Set SYSTEM_TIME without handling it
[    0.001000] <inf> ipc: rx	: 0xc4000000|0x3060004c
[    0.002000] <inf> ipc: rx	: 0x91000007|0x0
[    0.002000] <inf> pipe: pipeline new pipe_id 0 priority 0
[    0.002000] <inf> ipc: rx	: 0xc0000004|0x15
[    0.002000] <inf> ipc: buffer new size 0x180 id 0.0 flags 0x0
[    0.002000] <inf> hda_dma: hda-dmac :5 -> probe
[    0.002000] <inf> dma: dma_get() ID 5 sref = 1 busy channels 0
[    0.002000] <inf> pipe: connect buffer 0 as sink
[    0.003000] <inf> ipc: rx	: 0xc0000006|0x10
[    0.003000] <inf> module_adapter: module_init() start
[    0.003000] <inf> module_adapter: module_init() done
[    0.003000] <inf> ipc: rx	: 0xc4000006|0x30000018
[    0.003000] <inf> ipc: rx	: 0xc0000002|0xa
[... 154 lines cut ...]
[    6.708000] <inf> ipc: rx	: 0x93000003|0x0
[    6.708000] <inf> pipe: pipe trigger cmd 2
[    6.719000] <err> ipc: failed to wait schedule thread
[    6.719000] <err> ipc: ipc4: failed to send delayed replTerminate
```

**To Reproduce**
TPLG=/lib/firmware/intel/avs-tplg/cavs-tgl-nocodec.tplg MODEL=TGLU_RVP_NOCODEC_IPC4ZPH ~/sof-test/test-case/check-playback.sh -d 1 -l 1 -r 50

**Reproduction Rate**
almost 100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 
[f048bf5646cc](https://github.com/thesofproject/linux/commit/f048bf5646cc)

SOF Branch: main
SOF Commit: 
[5594ad351283](https://github.com/thesofproject/sof/commit/5594ad351283)

Zephyr Commit: 
[v3.2.0-rc1-2-g0956647aaf6b](https://github.com/zephyrproject-rtos/zephyr/commit/v3.2.0-rc1-2-g0956647aaf6b)

TPLG: cavs-tgl-nocodec.tplg

RanderWang, 2022-10-13:
fixed by https://github.com/thesofproject/sof/pull/6412

kv2019i, 2022-10-13:
@RanderWang I think it's something more complicated. To me any pipeline with mixer cannot handle a Ctrl-Z suspend and check-signal-stop-start.sh fails 100%. We've debugged this with @lyakh and this is completely reproducible.

We can see driver sends SOF_IPC4_PIPE_RESET and the mixin_mixout component erases state in its mixout_reset. 

When driver resumes the pipeline with SOF_IPC4_PIPE_PAUSED followed by SOF_IPC4_PIPE_RUNNING, processing will fail as there at code to reinitialize is only in mixout_bind().

I cannot see how the mixin_mixout has ever handled this case. @ranj063 @ujfalusi given driver works with our reference testbench, the driver flow is assumed to be correct here (for IPC4)?

ranj063, 2022-10-13:
@kv2019i what do you mean reference testbench?

lyakh, 2022-10-14:
> We can see driver sends SOF_IPC4_PIPE_RESET and the mixin_mixout component erases state in its mixout_reset.

@kv2019i https://github.com/thesofproject/sof/pull/6414/commits/eb521650fdfe838b0c92014a11d9834b93b69dd0 should fix it, waiting for CI...

serhiy-katsyuba-intel, 2022-10-14:
> @kv2019i [eb52165](https://github.com/thesofproject/sof/commit/eb521650fdfe838b0c92014a11d9834b93b69dd0) should fix it, waiting for CI...

@lyakh , that added memset() inside mixout_bind() will create problems: mixout can be connected to multiple mixins so every new bind will erase source info created for previous mixin bind. It's just enough to remove clearing from _reset().

Actually, I see there is one more problem in mixout_unbind() in 
/* mixout -> new sink */
	if (dev->ipc_config.id == src_id) {
		mixed_data_info->mixed_frames = 0;
		memset(mixed_data_info->source_info, 0, sizeof(mixed_data_info->source_info));
	} else { /* new mixin -> mixout */
...

If mixout and component on its sink are unbinded and then binded again, source info will be erased on unbind and never re-created as there was no new mixin<->mixout bind. This could be fixed by just clearing mixed_data_info->mixed_frames = 0; and resetting consumed_yet_not_produced_frames = 0; for every mixin in source info but preserving mixin pointers in source info table.

lyakh, 2022-10-17:
> @lyakh , that added memset() inside mixout_bind() will create problems: mixout can be connected to multiple mixins so every new bind will erase source info created for previous mixin bind. It's just [...]

@serhiy-katsyuba-intel Right, thanks, I've removed that.

> Actually, I see there is one more problem in mixout_unbind() in
>
> ```
> /* mixout -> new sink */
> 	if (dev->ipc_config.id == src_id) {
> 		mixed_data_info->mixed_frames = 0; [...]

Could you make a PR? I'm not sure I have a good way to test this

serhiy-katsyuba-intel, 2022-10-17:
> Could you make a PR? I'm not sure I have a good way to test this
OK. For both problems (reset and unbind) or just for second (unbind)? I do not see your pull-request for reset fix (that commit you mentioned seems does not belong to any pull-request).

serhiy-katsyuba-intel, 2022-10-17:
The PR submitted: https://github.com/thesofproject/sof/pull/6433

lgirdwood, 2022-10-19:
@keqiaozhang can we close, fix is merged ?