thesofproject/sof #6420: [BUG] [IPC4] FW reported error: 7 when testing single pipeline on all IPC4 CAVS platforms

keqiaozhang, 2022-10-14:
**Describe the bug**
We observed this weird regression in today's Daily_IPC4 testing(16232). This issue happens on all our CAVS IPC4 platforms, but the symptoms are not the same on each platform. 

On TGLU_RVP_NOCODEC_IPC4ZPH, it can be reproduced easily with playback on PCM0,0. even at the first round of test. But this issue cannot be reproduced with capture.

On TGLU_RVP_SDW_IPC4ZPH, this issue also happens with playback, but it takes several rounds of continuous testing to appear. No reproductions with capture so far.

On TGLU_UP_HDA_IPC4ZPH, this issue only happens with capture. If you run arecord on PCM0,0 twice in a row and there's no delay in the middle, you will hit the issue 100%.
e.g.
```
arecord   -Dhw:0,0 -r 48000 -c 2 -f S16_LE -d 3 /dev/null ; arecord   -Dhw:0,0 -r 48000 -c 2 -f S16_LE -d 3 /dev/null
```
But if you run the tests with a delay, this issue cannot be reprodcued.
```
arecord   -Dhw:0,0 -r 48000 -c 2 -f S16_LE -d 3 /dev/null; sleep 3; arecord   -Dhw:0,0 -r 48000 -c 2 -f S16_LE -d 3 /dev/null
```

There're only 3 new commits compared with yesterday. 

The first one only added some comments, so we can overlook it. 

Then I reset the commit to https://github.com/thesofproject/sof/commit/57029398360454cde6f96a3bae18bb4de44e6136 and tested it on TGLU_RVP_SDW_IPC4ZPH, there're no much difference form the tip commit.

But if I test it on TGLU_RVP_NOCODEC_IPC4ZPH and TGLU_UP_HDA_IPC4ZPH, the playback/capture always fails at the first round of test. So I assumed that https://github.com/thesofproject/sof/commit/57029398360454cde6f96a3bae18bb4de44e6136 and https://github.com/thesofproject/sof/commit/6fc6940e1c07c99e6cbd00c3a3a5b80de27b8b5c depend on each other to some extent. 
So hard to tell which commit caused this regression, need further investigation.

**dmesg**

```
[  175.785666] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13000003|0x0: GLB_SET_PIPELINE_STATE
[  175.785764] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000000|0x0: GLB_SET_PIPELINE_STATE
[  175.785776] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx done : 0x13000003|0x0: GLB_SET_PIPELINE_STATE
[  175.785779] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 0 state 4
[  175.785786] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx      : 0x13000004|0x0: GLB_SET_PIPELINE_STATE
[  175.786072] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx reply: 0x33000007|0x0: GLB_SET_PIPELINE_STATE
[  175.786076] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: FW reported error: 7 - Unsupported operation requested
[  175.786190] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc error for msg 0x13000004|0x0
[  175.786197] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: failed to set state 4 for pipeline 4
[  175.786201] kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -22
[  175.786206] kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -22
[  175.786212] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-tgl 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 0
[  175.786215] kernel: snd_sof:sof_ipc4_set_pipeline_state: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc4 set pipeline 1 state 3
```
[dmesg.txt](https://github.com/thesofproject/sof/files/9782203/dmesg.txt)

**Reproduction Rate**
Almost 100%

**Environment**
Kernel Branch: topic/sof-dev
Kernel Commit: 
[82c540b4ecd9](https://github.com/thesofproject/linux/commit/82c540b4ecd9)

SOF Branch: main
SOF Commit: 
[8fd351ea9aa6](https://github.com/thesofproject/sof/commit/8fd351ea9aa6)

Zephyr Commit: 
[zephyr-v3.2.0-441-ged661a6c6909](https://github.com/zephyrproject-rtos/zephyr/commit/zephyr-v3.2.0-441-ged661a6c6909)

kv2019i, 2022-10-14:
Looking at this now. This has to be related to the Zephyr baseline update. This passed automated IPC3 tests and my local IPC4 tests. Debugging now the delta between CI IPC4 tests and mine.

kv2019i, 2022-10-14:
The error7 seems to be related at least partly to this https://github.com/thesofproject/sof/pull/6421 , let's run through testing with that.

keqiaozhang, 2022-10-24:
This issue is not reproducible on our IPC4 platforms. Closing it.

closed 2022-10-24 by keqiaozhang