thesofproject/sof #6428: [BUG] topology2: cavs-passthrough-hda.conf does not exist

yongzhi1, 2022-10-15:
**Describe the bug**

Fail to build tplg2 with HDA_CONFIG set to "passthrough" option.

**To Reproduce**
Update the make file to pass HDA_CONFIG=passthrough and build cavs-mixin-mixout-hda-4ch-cnl

diff --git a/tools/topology/topology2/cavs/CMakeLists.txt b/tools/topology/topology2/cavs/CMakeLists.txt
index 61ae8379b..fa80acd6c 100644
--- a/tools/topology/topology2/cavs/CMakeLists.txt
+++ b/tools/topology/topology2/cavs/CMakeLists.txt
@@ -12,7 +12,7 @@ set(TPLGS
 "cavs-passthrough-hdmi\;cavs-mixin-mixout-hda-2ch-cnl\;\
 HDA_CONFIG=mix,NUM_DMICS=2,PLATFORM=cnl,PREPROCESS_PLUGINS=nhlt,NHLT_BIN=nhlt.bin"
 "cavs-passthrough-hdmi\;cavs-mixin-mixout-hda-4ch-cnl\;\
-HDA_CONFIG=mix,NUM_DMICS=4,PLATFORM=cnl,PDM1_MIC_A_ENABLE=1,PDM1_MIC_B_ENABLE=1,\
+HDA_CONFIG=passthrough,NUM_DMICS=4,PLATFORM=cnl,PDM1_MIC_A_ENABLE=1,PDM1_MIC_B_ENABLE=1,\

And observe:

Unable to open included conf file /sof/tools/topology/topology2/cavs/cavs-passthrough-hda.conf
Unable to process include file 
Failed to process includes
Failed to process conditional includes in input config
make[3]: *** [topology/topology2/cavs/CMakeFiles/topology2_cavs-mixin-mixout-hda-4ch-cnl.dir/build.make:75: topology/topology2/cavs/cavs-mixin-mixout-hda-4ch-cnl.tplg] Error 1
make[2]: *** [CMakeFiles/Makefile2:10180: topology/topology2/cavs/CMakeFiles/topology2_cavs-mixin-mixout-hda-4ch-cnl.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:10187: topology/topology2/cavs/CMakeFiles/topology2_cavs-mixin-mixout-hda-4ch-cnl.dir/rule] Error 2
make: *** [Makefile:4745: topology2_cavs-mixin-mixout-hda-4ch-cnl] Error 2

ranj063, 2022-10-17:
@yongzhi1 passthrough is not an option for HDA. Looks like it was left behind when we remove cavs-passthrough-hda.conf

yongzhi1, 2022-10-17:
Got it, thanks for the explanation @ranj063 , I was confused by:
https://github.com/thesofproject/sof/blob/main/tools/topology/topology2/cavs/cavs-passthrough-hdmi.conf#L47

closed 2022-10-24 by yongzhi1