thesofproject/sof #6431: [BUG] "No source info" in check-signal-stop-start-playback-50

kv2019i, 2022-10-17:
**Describe the bug**
This is one specific issue we have been tracking as part of https://github.com/thesofproject/sof/issues/6220

Issues has been rootcaused to incorrect clearing of mixer module state in reset and identified by following log in FW log:

[    2.019000] <err> mixer: No source info
[    2.019000] <err> pipe: pipeline_copy(): ret = -22, start->comp.id = 262145, dir = 1
[    2.019000] <err> pipe: pipeline_task(): xrun recovery failed! pipeline is stopped.

**Reproduction Rate**
100%

**Expected behavior**
No error from firmware.

**Impact**
Showstopper

**Environment**
SOF main as tested in 17th Oct daily tests.

kv2019i, 2022-10-17:
FYI @serhiy-katsyuba-intel

serhiy-katsyuba-intel, 2022-10-17:
The PR has been submitted: https://github.com/thesofproject/sof/pull/6433

lgirdwood, 2022-10-19:
@kv2019i can we close as PR merged ?

kv2019i, 2022-10-19:
Yes, let's close.

closed 2022-10-19 by kv2019i